Arduino-IRremote/Arduino-IRremote #283: Sanyo, suspicions of faulty code

adamlhumphreys, 2016-02-22:
**Board:** N/A
**Library Version:** 2.1.0
**Protocol:** Mitsubishi

**Code Block:**
```c
13 // Looks like Sony except for timings, 48 chars of data and time/space different 
...
16 #define SANYO_HDR_MARK	           3500  // seen range 3500 
17 #define SANYO_HDR_SPACE	            950  // seen 950 
18 #define SANYO_ONE_MARK	           2400  // seen 2400 
19 #define SANYO_ZERO_MARK             700  // seen 700 
...
50 	// Initial mark 
51 	if (!MATCH_MARK(results->rawbuf[offset++], SANYO_HDR_MARK))  return false ; 
52 
53 	// Skip Second Mark 
54 	if (!MATCH_MARK(results->rawbuf[offset++], SANYO_HDR_MARK))  return false ; 
55 
56 	while (offset + 1 < irparams.rawlen) { 
57 		if (!MATCH_SPACE(results->rawbuf[offset++], SANYO_HDR_SPACE))  break ; 
58 
59 		if      (MATCH_MARK(results->rawbuf[offset], SANYO_ONE_MARK))   data = (data << 1) | 1 ; 
60 		else if (MATCH_MARK(results->rawbuf[offset], SANYO_ZERO_MARK))  data = (data << 1) | 0 ; 
61 		else                                                            return false ; 
62 		offset++; 
63 	} 
64 
65 	// Success 
66 	results->bits = (offset - 1) / 2; 

```
Unless I'm missing something, this also doesn't look like Sony.
It looks like the header mark/space, being the same value, are both read as marks and then the marks are read as spaces and the spaces read as marks in the while() loop (much like Mitsubishi).

Unless I'm missing something, this is highly suspect.

z3t0, 2016-02-22:
Again, we need official documentation
https://pagure.io/linux/raw/master/f/drivers/media/rc/ir-sanyo-decoder.c 
This should help

adamlhumphreys, 2016-02-22:
I have to say, that definitely looks a lot different. :O Official documentation would be awesome, though I doubt that information is available to the public. (Could be wrong.) Nevertheless, that link may well be accurate, not to mention recent. :)

z3t0, 2016-07-11:
#333

closed 2016-07-11 by z3t0