Arduino-IRremote/Arduino-IRremote #289: Add Features for Arduino to Arduino Communication

z3t0, 2016-02-26:
We could take a look at existing protocols and create helper methods for arduino to arduino communcation

ElectricRCAircraftGuy, 2016-02-26:
I recommend modding the softwareSerial library to allow an IR Led and receiver to wirelessly communicate via software serial. All it requires is PWMing the high signal outputs at the proper IR freq of 38kHz or so, and using low baud rates so the IR receiver error doesn't become a problem, as well as correcting for Mark_Excess on the receiving end of course. The same can be done for the core shiftOut, shiftIn, etc. functions, so that they can be used for IR communication too.

AnalysIR, 2016-02-27:
It is effectively available already within the library...just use NEC to send 4 bytes at a time (32 bits) or even one byte at a time. 

```
sendNEC(byteVal,8); // send one byte
sendNEC(integerVal,16); // send 2 bytes
sendNEC(longVal,32); // send 4 bytes....could use one byte as a checksum
```

Notes:
1. Some IR receivers are designed to turn off when 'continuous signals' are detected (assuming it is noise)
2. Can only be half duplex - so both ends should have unequal timeouts and logic for collision detection, errors & retransmission.
3. NEC also uses inverted bytes as a sort of  basic error correction. I dont believe this is implemented in the library, as there are also non-compliant NEC style systems in the marketplace.

ElectricRCAircraftGuy, 2016-02-27:
sendSony works too. I'm using it to send analog joystick commands, as digital data, with error checking and a parity bit, to control a robot.

z3t0, 2016-02-27:
I see, in that case maybe an example Sketch that shows users how to send data in this manner?

pcoughlin, 2016-02-28:
Would this be a Master-Slave setup or Partners without a master (which would allow for multiple senders and receivers going to/from each other in a multi-cast environment).

ElectricRCAircraftGuy, 2016-02-28:
Non master slave. Adhoc basically. Anyone to anyone. Simultaneous receivers allowed. Up to individual implementation....Ex the example could show addressing where the listeners only respond to data sent to them...or not. Up to the user. Any examples, whether complicated or simple, could be useful.

ElectricRCAircraftGuy, 2016-02-28:
I say create a few basic sender to receiver type examples and call it good. Have one say "hello" to another or something.

pandacriss, 2016-03-01:
we could create a method to ease the transmission. detect the long of the message, separate the bytes and send them in burst separated by a silence to avoid the receiver to confuse it with noise.

closed 2020-10-17 by ArminJo