Arduino-IRremote/Arduino-IRremote #290: sending IR commands from a long distance

pandacriss, 2016-02-27:
im not sure where else to ask this. i cant find if this project have a forum. excuse me if this is not the right place.
i want to send the IR commands from a long distance. i think i could use a transistor to drive multiple leds in parallel, but i dont know about transistors. dont know which one to use and what resistor should go in the gate pin.
could anyone user help me with that please.

bengtmartensson, 2016-02-27:
Processors like AtMega328 (Uno, Nano, etc) cap provide 40mA on the GPIO pins; that is more than enough for quite a high-power sender. Connecting two LEDs in series is a better idea than parallel, higher efficiency, and saves you one resistor. See [my hardware guide](http://www.harctoolbox.org/arduino_nano.html) for an example. If you absolutely want a transistor, I recommend a FET switching transistor like [BS 170 (in the US 2N7000)](https://en.wikipedia.org/wiki/2N7000). It saves you the base/gate resistor, it can be connected directly to a GPIO pin.

AnalysIR, 2016-02-27:
@pandacriss 
How far is 'Long Distance' ?

see: [https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/](https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/)

pandacriss, 2016-02-27:
the stereo is like 10 meter away, by a window and the light would need to bounce in a wall.
i know is far but i could make an array of leds or something

ElectricRCAircraftGuy, 2016-02-27:
That shouldn't be a problem. Use a transistor with and resistor sized to allow up to 500mAh or so through a couple IR LEDs like this (https://www.adafruit.com/products/387), and you'll be good. It can take 100mA max continuous, or 1A max pulsed. Want to know which transistor and resistor could work? Check this product out: https://www.adafruit.com/product/73 - it has a range of 150+ ft. Parts list here: https://learn.adafruit.com/tv-b-gone-kit?view=all#parts-list

pandacriss, 2016-02-28:
thanks all. thanks aircraftguy, this seem good. https://raw.githubusercontent.com/adafruit/TV-B-Gone-kit/master/pcb/tvbgone3sch.png

will try with that led array configuration.
just one last thing, how to use this codes with irremote? how can i convert this format to the hexadecimal used in the lib? http://www.remotecentral.com/cgi-bin/codes/samsung/tv_functions/Text

z3t0, 2016-03-23:
#263 might help

z3t0, 2017-08-22:
Closing due to inactivity.

closed 2017-08-22 by z3t0