Arduino-IRremote/Arduino-IRremote #291: Please Add Support For the ATmega(32/16/8)u2

commanderjsoc, 2016-02-29:
**Board:** ATmega(32/16/8)u2
**Library Version:** 2.1.0
**Protocol:** All

Message: 
"I would like to be able to use the ATmega(32/16/8)u2 with your library. The microchip is somewhat similar to the atmega32u4. I would do it myself but, I am confused by the irSend pwm setups."

**Thanks**
-**Commanderjsoc**

mhdfasilwyd, 2016-03-02:
Me too pls add support fir atmega 16 and 32

z3t0, 2016-03-23:
I personally do not have access to either chip, anyone else?

mhdfasilwyd, 2016-03-23:
Thank you for ur great work i was working for more than months ...... excited with ur replyyyy thanks

z3t0, 2016-03-23:
@mhdfasilwyd haha no problem! thank @mcudude hes the one doing the real work

mhdfasilwyd, 2016-03-24:
Can you help me in my project 

I want to link PIR and nrf2401 for wieless motion sensor i searched internet but i didnt got a perfwct one

MCUdude, 2016-03-24:
If you want help with your project, head over to the [Arduino Forum](http://forum.arduino.cc)

q2dg, 2016-06-16:
Well, this issue could be close when https://github.com/z3t0/Arduino-IRremote/pull/293 gets merged

MCUdude, 2016-06-26:
The ATmega16u2 and the ATmega16 is not the same chip. the ATmega16u2 have built in USB functionality, and is relatively "new" compared to the ATmega16 (which have been on the market for more than 12 years now). You'll need to add some special routines in the IRremote library to add support for the ATmega8u2, ATmega16u2 and ATmega32u2

christw92, 2018-02-13:
@MCUdude Is there any plan for supporting the mentioned models (especially ATmega16u2) in near future ? It seems like they are quite wide spread now. Or at least I got two of them now and would really appreciate if I could use them with IRremote.

MCUdude, 2018-02-13:
I tend to add Arduino support for different AVR "families". [MightyCore](https://github.com/MCUdude/MightyCore) supports ATmega16, ATmega32, ATmega164, ATmega324, ATmega644 and ATmega1284. [MiniCore](https://github.com/MCUdude/MiniCore) supports ATmega8, ATmega48, ATmega88, ATmega168 and ATmega328. All members in each "family" are pin compatible, and drop-in replacable.

Even though **ATmega8**u2 and **ATmega16**u2 pretty much sounds like varuants of **ATmega8** and **ATmega16** they definitly are not. Not the same pheripherals, not the same pinout, not the same pin mapping, not the same packaging.

I have no plans adding support for the ATmega*u2 family, which actually also include the ATmega32u2

DaleMitchell, 2020-01-19:
From an AVR appnote (530):
> The ATmega16U2/8U2 is a functionally identical, drop-in replacement for the AT90USB162/82. 

From `boardsdef.h`:
https://github.com/z3t0/Arduino-IRremote/blob/2dc97b29f2da74e342a308f6e99f153e132ce06a/boarddefs.h#L90

I don't see why there is not support already! So, I propose this simple change to `boardsdef.h`, which I have already made in pull request #672:
```
// ATmega8u2, ATmega16U2, ATmega32U2
#elif defined(__AVR_ATmega8U2__) || defined(__AVR_ATmega16U2__)  || defined(__AVR_ATmega32U2__)
	#define IR_USE_TIMER1     // tx = pin C6
```

commit 3c70d870 (2020-10-12): Support For the ATmega(32/16/8)u2 - closes #291
closed 2020-10-12 by ArminJo