Arduino-IRremote/Arduino-IRremote #294: First interval

lior22lior22, 2016-03-08:
Hello,

i found on internet example to dump ir command recived, 

`#include <IRremote.h>

int RECV_PIN = 11;
IRrecv irrecv(RECV_PIN);
decode_results results;

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
}

int c = 1;

void dump(decode_results *results) {
  int count = results->rawlen;
  Serial.println(c);
  c++;
  Serial.println("For IR Scope: ");
  for (int i = 1; i < count; i++) {
   
	if ((i % 2) == 1) {
  	Serial.print("+");
  	Serial.print(results->rawbuf[i]*USECPERTICK, DEC);
	}
	else {
  	Serial.print(-(int)results->rawbuf[i]*USECPERTICK, DEC);
	}
	Serial.print(" ");
  }
  Serial.println("");
  Serial.println("For Arduino sketch: ");
  Serial.print("unsigned int raw[");
  Serial.print(count, DEC);
  Serial.print("] = {");
  for (int i = 1; i < count; i++) {
   
	if ((i % 2) == 1) {
  	Serial.print(results->rawbuf[i]*USECPERTICK, DEC);
	}
	else {
  	Serial.print((int)results->rawbuf[i]*USECPERTICK, DEC);
	}
	Serial.print(",");
  }
  Serial.print("};");
  Serial.println("");
  Serial.print("irsend.sendRaw(raw,");
  Serial.print(count, DEC);
  Serial.print(",38);");
  Serial.println("");
  Serial.println("");
}

void loop() {
  if (irrecv.decode(&results)) {
	dump(&results);
	irrecv.resume(); // Receive the next value
  }
}`

and i used him to send with sendraw array that outputed from dump function but, i dont understand why in the for loop he begins with results->rawbuf[1] instead of [0], i able to turn on my t.v with the array that comes out, but not the air conditioner, it is possible that its because he doesnt include the first interval?

AnalysIR, 2016-03-08:
Likely because the first position is a space and all IR signals should begin with a mark. Having a space in the first position makes no real sense.

ps: I haven't reviewed the code you posted, which should be formatted using the insert code icon  in the toolbar.

lior22lior22, 2016-03-08:
thanks @AnalysIR , and about the code, i know i put it all inside the ' that show when i press on insert code but it doesnt work

bengtmartensson, 2016-03-09:
IRremote has the peculiarity that the first sample `rawbuf[0]` is the time since listening started until the first mark arrived -- a truly useless piece of information. Likewise, the last sample `rawbuf[rawlen-1]` is the length of the last captured mark, i.e. the length of time the program listened to determine that the signal was considered ended (an often _useful_ piece of information) is disposed. This is bad design. IRLib shares the same problem, Infrared4Arduino does not.

z3t0, 2017-08-22:
Closing due to inactivity.

closed 2017-08-22 by z3t0