Arduino-IRremote/Arduino-IRremote #295: Errors compiling examples - Teensy 3.1

Garywoo, 2016-03-10:
**Board:** Teensy 3.1
**Library Version:** 2.0.1

Please excuse my inexperience here, but I am trying to compile IRrecord.ino, or any of the examples (error occurs for all).
I have previously successfully compiled and ran the examples in library version 0.1 downloaded from [PJRC](https://www.pjrc.com/teensy/td_libs_IRremote.html) with only a few warnings.

**Error:**
```c

Arduino: 1.6.7 (Mac OS X), TD: 1.27, Board: "Teensy 3.2 / 3.1, Serial, 96 MHz optimized (overclock), United Kingdom"

/Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/examples/IRrecord/IRrecord.ino: In function 'void storeCode(decode_results*)':
/Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/examples/IRrecord/IRrecord.ino:48:7: warning: unused variable 'count' [-Wunused-variable]
int count = results->rawlen;
^
In file included from /Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/IRremote.h:24:0,
from /Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/IRremote.cpp:25:
/Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/IRremote.cpp: In function 'void cmt_isr()':
/Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/IRremoteInt.h:473:10: warning: unused variable 'tmp' [-Wunused-variable]
uint8_t tmp = CMT_MSC; \
^
/Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/IRremote.cpp:125:2: note: in expansion of macro 'TIMER_RESET'
TIMER_RESET;
^
In file included from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/core_pins.h:34:0,
from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/wiring.h:33,
from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/WProgram.h:15,
from /Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/Arduino.h:1,
from /Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/IRremoteInt.h:24,
from /Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/IRremote.h:24,
from /Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/irRecv.cpp:1:
/Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/irRecv.cpp: In member function 'void IRrecv::enableIRIn()':
/Applications/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/kinetis.h:3115:52: error: expression cannot be used as a function
#define CMT_CMD3  (*(volatile uint8_t  *)0x40062008) // CMT Modulator Data Register Space High
^
/Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/IRremoteInt.h:524:2: note: in expansion of macro 'CMT_CMD3'
CMT_CMD3   = 0;               \
^
/Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/irRecv.cpp:120:2: note: in expansion of macro 'TIMER_CONFIG_NORMAL'
TIMER_CONFIG_NORMAL();
^
In file included from /Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/IRremote.h:24:0,
from /Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/irRecv.cpp:1:
/Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/IRremoteInt.h:473:10: warning: unused variable 'tmp' [-Wunused-variable]
uint8_t tmp = CMT_MSC; \
^
/Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote-master/irRecv.cpp:125:2: note: in expansion of macro 'TIMER_RESET'
TIMER_RESET;
^
exit status 1
Error compiling.

```

I would appreciate any assistance in getting this updated library to compile.

AnalysIR, 2016-03-11:
IRrecord.ino compiles fine for me on Arduino IDE 1.6.5 or Teensy 3.1

..so maybe there is something wrong with your installation??

Do you have the library installed properly?

Do you have the teensy addons installed properly for the IDE?

Did you try to compile the blink sketch for the teensy 3.1?

Maybe rename the Irremote library folder from

`/Users/Gary/Dropbox/Arduino/libraries/Arduino-IRremote/`

to

`/Users/Gary/Dropbox/Arduino/libraries/IRremote/`

and restart the IDE

Garywoo, 2016-03-11:
I downgraded to Arduino IDE 1.6.5 to test as you have.

Installed library though Sketch > Include Library > Add .ZIP Library... then pointing to the downloaded folder from Github.

I had the teensy addons installed into the Arduino IDE, but I reinstalled for good measure after downgrading to 1.6.5.

The blink sketch for teensy 3.1 works fine, as does every other sketch I throw at it that doesn't include IRremote 2.0.1.

Folder renaming and restarting had no effect. :disappointed:

It's still producing the same error, on both my PC and Mac.

AnalysIR, 2016-03-18:
Unfortunately, I have no more suggestions...sorry

z3t0, 2016-03-23:
Hi @Garywoo somewhere along the way we seem to have broken teensy support #171 
Something I hope to fix once I get some time on my hands!

In the meantime the teensy provided library is still stable

bookerhsu, 2016-07-01:
I had same problem : I don't know what i can ? god !!!!!!!

\Program Files (x86)\Arduino/hardware/tools/arm/bin/arm-none-eabi-g++ -c -O -g -Wall -ffunction-sections -fdata-sections -MMD -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=128 -DARDUINO=10605 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3 -IC:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master r:\TEMP\build6393741459609701638.tmp\IRsendRawDemo.cpp -o r:\TEMP\build6393741459609701638.tmp\IRsendRawDemo.cpp.o 
C:\Program Files (x86)\Arduino/hardware/tools/arm/bin/arm-none-eabi-g++ -c -O -g -Wall -ffunction-sections -fdata-sections -MMD -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=128 -DARDUINO=10605 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3 -IC:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master -IC:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\utility C:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\irPronto.cpp -o r:\TEMP\build6393741459609701638.tmp\IRremote\irPronto.cpp.o 
C:\Program Files (x86)\Arduino/hardware/tools/arm/bin/arm-none-eabi-g++ -c -O -g -Wall -ffunction-sections -fdata-sections -MMD -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=128 -DARDUINO=10605 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3 -IC:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master -IC:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\utility C:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\irRecv.cpp -o r:\TEMP\build6393741459609701638.tmp\IRremote\irRecv.cpp.o 
In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/core_pins.h:34:0,
from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/wiring.h:33,
from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WProgram.h:15,
from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/Arduino.h:1,
from C:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\IRremoteInt.h:24,
from C:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\IRremote.h:24,
from C:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\irRecv.cpp:1:
C:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\irRecv.cpp: In member function 'void IRrecv::enableIRIn()':
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/kinetis.h:3315:52: error: expression cannot be used as a function
#define CMT_CMD3  (*(volatile uint8_t  *)0x40062008) // CMT Modulator Data Register Space High
^
C:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\IRremoteInt.h:524:2: note: in expansion of macro 'CMT_CMD3'
CMT_CMD3   = 0;               \
^
C:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\irRecv.cpp:120:2: note: in expansion of macro 'TIMER_CONFIG_NORMAL'
TIMER_CONFIG_NORMAL();
^
In file included from C:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\IRremote.h:24:0,
from C:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\irRecv.cpp:1:
C:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\IRremoteInt.h:473:10: warning: unused variable 'tmp' [-Wunused-variable]
uint8_t tmp = CMT_MSC; \
^
C:\Users\booker\Documents\Arduino\libraries\Arduino-IRremote-master\irRecv.cpp:125:2: note: in expansion of macro 'TIMER_RESET'
TIMER_RESET;
^
Error compiling.

AnalysIR, 2016-07-01:
It might be a good idea(quicker) to seek support on the Teensy forum & let us know the result.

Usually, @PaulStoffregen is proactive & a long time contributor to IRremote & lots more.....

..sorry I can't be of more help

TIP: provide more details about your system/setup/configration & what steps you have taken to debug yourself -  when seeking assistance

PaulStoffregen, 2016-08-01:
I've submitted a fix for this compile problem, and also a fix for the unsupported clock speeds.

https://github.com/z3t0/Arduino-IRremote/pull/343

PaulStoffregen, 2016-08-01:
Fixed

closed 2016-08-01 by PaulStoffregen