Arduino-IRremote/Arduino-IRremote #296: Implicit one-bit checksum by signal length

johnmayqwerty, 2016-03-17:
I'm using IRReceive for controlling some little robots - first, well done! it works very nicely.

My problem is data reliability - in order to solve it I'm using checksums, and of course that works fine, I'm trading reliability for bitrate.

BUT it occurred to me that there's something you could do, for many protocols, that could help to insure you've got good data.

Basically, for any protocol where bits are different lengths for ON or OFF... the total signal length in time ought to match the added up bit lengths.

If, when receiving the signal, you can measure the total signal length in time, either you in the library or I in the consuming code can compare that length to what it ought to be for the data received.

This would catch all one-bit errors, without changing the protocols or using any checksum. The "checksum" is just the total signal length itself.

z3t0, 2016-03-23:
Any chance you could propose a patch and share as a PR :wink: 

I'm slowly getting around to all the different issues/prs .... and PRs are mainly prioritized as implementations are provided and I simply have to check to make sure things work as expected, right now I simply do not have enough time to work on implementing features myself

ArminJo, 2020-10-12:
Use NEC protocol, it has a built in check for the command

closed 2020-10-12 by ArminJo