Arduino-IRremote/Arduino-IRremote #297: IR receiver giving different codes for the same button

darshkpatel, 2016-03-18:
****Board:** ARDUINO NANO
**Library Version:** 2.1.0
**Protocol:** Says UNKNOWN PROTOCOL**

I am Currently trying to capture 'Tatasky' (India) TV remote codes , but i get different codes every time i press the same button.

Receiving IR Codes for all the other remotes in my house work fine and the results are as expected , infact the remote of the same company DVR (ie. Tatasky), but a different Model also  works fine.

I have dumped the codes which are outputted when pressing the same button about 10 times

`Encoding  : UNKNOWN
Code      : 49EF3073 (32 bits)
Timing[35]: 
     + 450, - 250     + 150, - 300     + 200, - 550     + 150, - 600
     + 200, - 250     + 200, - 750     + 200, - 550     + 200, - 250
     + 200, - 250     + 150, - 600     + 200, - 550     + 200, - 450
     + 150, - 600     + 150, - 300     + 150, - 300     + 150, - 250
     + 200, - 250     + 150
unsigned int  rawData[35] = {450,250, 150,300, 200,550, 150,600, 200,250, 200,750, 200,550, 200,250, 200,250, 150,600, 200,550, 200,450, 150,600, 150,300, 150,300, 150,250, 200,250, 150};  // UNKNOWN 49EF3073

Encoding  : UNKNOWN
Code      : 6E277A3 (32 bits)
Timing[35]: 
     + 450, - 250     + 150, - 250     + 200, - 600     + 150, - 600
     + 200, - 250     + 150, - 800     + 150, - 600     + 150, - 300
     + 150, - 300     + 150, - 250     + 200, - 600     + 150, - 450
     + 150, - 600     + 200, - 250     + 150, - 300     + 150, - 250
     + 200, - 250     + 200
unsigned int  rawData[35] = {450,250, 150,250, 200,600, 150,600, 200,250, 150,800, 150,600, 150,300, 150,300, 150,250, 200,600, 150,450, 150,600, 200,250, 150,300, 150,250, 200,250, 200};  // UNKNOWN 6E277A3

Encoding  : UNKNOWN
Code      : B70D49E4 (32 bits)
Timing[35]: 
     + 450, - 250     + 150, - 300     + 150, - 600     + 150, - 600
     + 200, - 250     + 200, - 750     + 150, - 600     + 200, - 250
     + 200, - 250     + 150, - 600     + 200, - 600     + 150, - 450
     + 150, - 600     + 200, - 250     + 150, - 300     + 150, - 250
     + 200, - 250     + 200
unsigned int  rawData[35] = {450,250, 150,300, 150,600, 150,600, 200,250, 200,750, 150,600, 200,250, 200,250, 150,600, 200,600, 150,450, 150,600, 200,250, 150,300, 150,250, 200,250, 200};  // UNKNOWN B70D49E4

Encoding  : UNKNOWN
Code      : 25D6E01D (32 bits)
Timing[35]: 
     + 450, - 250     + 150, - 250     + 200, - 600     + 150, - 600
     + 200, - 250     + 150, - 800     + 150, - 600     + 200, - 250
     + 150, - 300     + 150, - 250     + 200, - 600     + 150, - 450
     + 150, - 600     + 200, - 250     + 150, - 300     + 150, - 300
     + 150, - 250     + 200
unsigned int  rawData[35] = {450,250, 150,250, 200,600, 150,600, 200,250, 150,800, 150,600, 200,250, 150,300, 150,250, 200,600, 150,450, 150,600, 200,250, 150,300, 150,300, 150,250, 200};  // UNKNOWN 25D6E01D

Encoding  : UNKNOWN
Code      : 40033D4D (32 bits)
Timing[35]: 
     + 450, - 250     + 200, - 250     + 150, - 600     + 200, - 600
     + 150, - 300     + 150, - 750     + 200, - 600     + 150, - 250
     + 200, - 250     + 200, - 600     + 150, - 600     + 150, - 450
     + 150, - 600     + 200, - 250     + 200, - 250     + 150, - 300
     + 150, - 250     + 200
unsigned int  rawData[35] = {450,250, 200,250, 150,600, 200,600, 150,300, 150,750, 200,600, 150,250, 200,250, 200,600, 150,600, 150,450, 150,600, 200,250, 200,250, 150,300, 150,250, 200};  // UNKNOWN 40033D4D

Encoding  : UNKNOWN
Code      : 6E277A3 (32 bits)
Timing[35]: 
     + 450, - 250     + 150, - 250     + 200, - 600     + 150, - 600
     + 200, - 250     + 150, - 800     + 150, - 600     + 150, - 300
     + 150, - 300     + 150, - 250     + 200, - 600     + 150, - 450
     + 150, - 600     + 200, - 250     + 150, - 300     + 150, - 250
     + 200, - 250     + 200
unsigned int  rawData[35] = {450,250, 150,250, 200,600, 150,600, 200,250, 150,800, 150,600, 150,300, 150,300, 150,250, 200,600, 150,450, 150,600, 200,250, 150,300, 150,250, 200,250, 200};  // UNKNOWN 6E277A3

Encoding  : UNKNOWN
Code      : 9A2A2562 (32 bits)
Timing[35]: 
     + 450, - 250     + 150, - 250     + 200, - 600     + 150, - 600
     + 200, - 250     + 150, - 800     + 150, - 600     + 150, - 300
     + 150, - 250     + 200, - 600     + 150, - 600     + 200, - 400
     + 200, - 600     + 150, - 300     + 150, - 250     + 200, - 250
     + 200, - 250     + 150
unsigned int  rawData[35] = {450,250, 150,250, 200,600, 150,600, 200,250, 150,800, 150,600, 150,300, 150,250, 200,600, 150,600, 200,400, 200,600, 150,300, 150,250, 200,250, 200,250, 150};  // UNKNOWN 9A2A2562

Encoding  : UNKNOWN
Code      : DF6C1994 (32 bits)
Timing[35]: 
     + 450, - 250     + 200, - 250     + 150, - 600     + 200, - 600
     + 150, - 300     + 150, - 750     + 200, - 600     + 150, - 250
     + 200, - 250     + 200, - 250     + 150, - 600     + 200, - 400
     + 200, - 600     + 150, - 300     + 150, - 250     + 200, - 250
     + 200, - 250     + 150
unsigned int  rawData[35] = {450,250, 200,250, 150,600, 200,600, 150,300, 150,750, 200,600, 150,250, 200,250, 200,250, 150,600, 200,400, 200,600, 150,300, 150,250, 200,250, 200,250, 150};  // UNKNOWN DF6C1994

Encoding  : UNKNOWN
Code      : 2EA1533E (32 bits)
Timing[35]: 
     + 450, - 200     + 200, - 250     + 200, - 600     + 150, - 600
     + 150, - 300     + 150, - 750     + 200, - 600     + 150, - 300
     + 150, - 250     + 200, - 600     + 150, - 600     + 200, - 400
     + 200, - 600     + 150, - 250     + 200, - 250     + 200, - 250
     + 150, - 300     + 150
unsigned int  rawData[35] = {450,200, 200,250, 200,600, 150,600, 150,300, 150,750, 200,600, 150,300, 150,250, 200,600, 150,600, 200,400, 200,600, 150,250, 200,250, 200,250, 150,300, 150};  // UNKNOWN 2EA1533E

Encoding  : UNKNOWN
Code      : 915E804D (32 bits)
Timing[35]: 
     + 450, - 250     + 150, - 300     + 150, - 600     + 200, - 550
     + 200, - 250     + 200, - 750     + 150, - 600     + 200, - 250
     + 200, - 250     + 150, - 300     + 150, - 600     + 200, - 400
     + 200, - 550     + 200, - 250     + 200, - 250     + 200, - 250
     + 150, - 300     + 150
unsigned int  rawData[35] = {450,250, 150,300, 150,600, 200,550, 200,250, 200,750, 150,600, 200,250, 200,250, 150,300, 150,600, 200,400, 200,550, 200,250, 200,250, 200,250, 150,300, 150};  // UNKNOWN 915E804D

Encoding  : UNKNOWN
Code      : 7B255A09 (32 bits)
Timing[35]: 
     + 400, - 250     + 200, - 250     + 200, - 550     + 200, - 600
     + 150, - 300     + 150, - 750     + 200, - 600     + 150, - 250
     + 200, - 250     + 200, - 600     + 150, - 600     + 150, - 450
     + 200, - 550     + 200, - 250     + 200, - 250     + 150, - 300
     + 150, - 300     + 150
unsigned int  rawData[35] = {400,250, 200,250, 200,550, 200,600, 150,300, 150,750, 200,600, 150,250, 200,250, 200,600, 150,600, 150,450, 200,550, 200,250, 200,250, 150,300, 150,300, 150};  // UNKNOWN 7B255A09

Encoding  : UNKNOWN
Code      : AD737632 (32 bits)
Timing[35]: 
     + 450, - 250     + 150, - 250     + 200, - 600     + 150, - 600
     + 200, - 250     + 200, - 750     + 150, - 600     + 200, - 250
     + 150, - 300     + 150, - 250     + 200, - 600     + 150, - 450
     + 150, - 600     + 200, - 250     + 200, - 250     + 150, - 300
     + 150, - 250     + 200
unsigned int  rawData[35] = {450,250, 150,250, 200,600, 150,600, 200,250, 200,750, 150,600, 200,250, 150,300, 150,250, 200,600, 150,450, 150,600, 200,250, 200,250, 150,300, 150,250, 200};  // UNKNOWN AD737632

`

AnalysIR, 2016-03-18:
Hi It looks like the RCMM protocol....but cant be 100% sure.

This is a very difficult protocol to capture, particularly with the 50uSecs granularity of IRremote.

I imported your 3rd signal into AnalysIR, cleaned it and got the following

```
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : RCMM :  : 28E0A600 : 32 : 0
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit http://www.AnalysIR.com or email info@....... for further details
int khz=36; //NB Change this default value as neccessary to the correct carrier frequency

unsigned int Signal_0_0[] = {417,278,167,278,167,611,167,611,167,278,167,778,167,611,167,278,167,278,167,611,167,611,167,444,167,611,167,278,167,278,167,278,167,278,167}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: RCMM, Key:  

```

However, bear in mind, that it may well be off because of the way IRremote captured the signal timings.
You may also need an RCMM specific IR receiver to get good captures.

Regading the varying codes output from IRremote...when it doesn't regognise the protocols it outputs a guess...so just ignore 'Code' when Encoding is 'Unknown'.

bengtmartensson, 2016-03-19:
IrScrutinizer identifies the captures as protocol = Nokia32, device = 40, subdevice = 224, obc = 0, misc = X=38, with different (alternating really) values of the toggle. I tried the first six of them, all decoded fine.

darshkpatel, 2016-03-19:
Can you help me implement it in the ir relay ,
Maybe a code sample ?

bengtmartensson, 2016-03-19:
I am not sure exactly what you want help with. In the case you are looking for an Arduino based Nokia-32 decoder, this [is not written yet](https://github.com/bengtmartensson/harctoolboxbundle/issues/97).

AnalysIR, 2016-03-19:
@darshkpatel 
You should be able to reuse the code I posted above...
...all you have to do is to replace the time values you get from IRremote to the nearest of
417 or 167 for +marks  (first mark is 417 and the rest are all 167)
and
278 or 444 or 611 or 778 for the -spaces

Using sendRAW is your only option at the moment with IRremote, but I suspect you will need ot correct the timings, as above.

However, if you use the IRLib library, RC-MM  is indeed supported.
[https://github.com/cyborg5/IRLib/blob/master/examples/rcmm/rcmm.ino](https://github.com/cyborg5/IRLib/blob/master/examples/rcmm/rcmm.ino)

darshkpatel, 2016-03-19:
Thank's @AnalysIR  @bengtmartensson  :smile:

darshkpatel, 2016-04-02:
I want to use IRremoteESP8266 With a remote which uses the RC-MM protocol  , I could not port the library here 
https://github.com/cyborg5/IRLib/blob/master/examples/rcmm/rcmm.ino
like @AnalysIR  mentioned 

to esp8266 as it uses some avr components too, 

can anyone help me add the rcmm protocol ?

AnalysIR, 2016-04-02:
I believe there is a version of IRremote ported to ESP8266(via google). However, you would have to use sendRAW for RCMM or alternatively write your oown add-on for RCMM.

Roseau, 2016-08-09:
Board : Arduino Uno
Library Version : 2.01
protocol : unknown with sanyo in between

Excuse me, im trying to capture the command of my remote(Chunghop universal Remote K-1028E) and get this code when pressing on/off button a couple of times :

49B0F626
Unknown Encoding : 49B0F626 (32 bits)
Raw (6): 1274 150 -5000 50 -4650 100 

FFFFFFFF
Decode SANYO : FFFFFFFF (0 bits)
Raw (101): -15500 3500 -1650 450 -400 450 -1250 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 400 -400 450 -400 450 -350 450 -1250 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -1250 400 -1300 450 -1250 450 -350 450 -400 450 -1250 450 -350 450 -400 450 -350 450 -400 400 -400 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 400 -400 500 -350 400 -400 450 -400 450 -350 450 -400 

687AD153
Unknown Encoding : 687AD153 (32 bits)
Raw (101): 5710 3500 -1700 450 -400 400 -1300 400 -400 400 -450 450 -350 400 -450 400 -400 450 -400 450 -350 400 -400 450 -400 450 -400 400 -400 400 -1300 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 450 -1200 450 -1300 450 -1250 400 -400 400 -450 400 -1300 400 -400 400 -450 400 -400 450 -350 450 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -350 450 -400 450 -400 400 -400 450 -400 

68D13F41
Unknown Encoding : 68D13F41 (32 bits)
Raw (101): 4386 3550 -1650 400 -400 450 -1250 450 -400 400 -400 450 -400 450 -350 450 -400 450 -350 450 -400 400 -400 450 -400 400 -400 450 -400 400 -1300 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -1250 450 -1250 450 -1250 450 -400 450 -350 450 -1250 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 400 -400 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 

68D13F41
Unknown Encoding : 68D13F41 (32 bits)
Raw (101): -11514 3500 -1700 450 -350 450 -1250 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -1250 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -1250 450 -1250 450 -1250 450 -400 450 -350 450 -1300 400 -400 450 -400 400 -400 450 -350 450 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -350 450 -400 450 -400 400 -400 450 -400 

68D13F41
Unknown Encoding : 68D13F41 (32 bits)
Raw (101): -19670 3500 -1700 450 -350 450 -1300 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -350 450 -400 450 -400 400 -400 450 -1250 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -1250 450 -1300 400 -1300 400 -400 450 -400 400 -1300 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 400 -400 450 -400 450 -350 450 -400 

FFFFFFFF
Decode SANYO : FFFFFFFF (0 bits)
Raw (101): -9900 3450 -1650 450 -400 450 -1250 450 -350 500 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -350 500 -1250 450 -350 450 -400 450 -350 450 -400 450 -350 450 -350 500 -350 450 -1250 450 -1300 400 -1300 400 -400 450 -350 450 -1300 400 -400 450 -400 400 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -400 400 -400 450 -400 400 -400 450 -1250 450 -350 500 -350 450 -400 450 -1200 500 -1250 450 -350 450 -350 500 -350 

FFFFFFFF
Decode SANYO : FFFFFFFF (0 bits)
Raw (101): -17600 3500 -1700 450 -350 450 -1250 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -1250 450 -350 450 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -1250 450 -1300 400 -1300 400 -400 450 -400 400 -1300 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 500 -350 450 -350 450 -400 450 -350 450 -400 400 -400 450 -400 450 -350 450 -400 

FFFFFFFF
Decode SANYO : FFFFFFFF (0 bits)
Raw (101): -6650 3550 -1650 450 -400 450 -1250 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 400 -400 450 -400 450 -350 450 -1250 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -1250 450 -1250 450 -1250 450 -350 450 -400 450 -1250 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -350 450 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 

FFFFFFFF
Decode SANYO : FFFFFFFF (0 bits)
Raw (101): -10400 3500 -1650 500 -350 450 -1250 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 500 -350 450 -400 400 -400 450 -400 400 -1300 400 -400 450 -400 400 -400 450 -400 400 -400 450 -400 400 -400 450 -1250 450 -1250 450 -1250 450 -400 450 -350 450 -1250 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 400 -400 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350 450 -400 450 -350

which raw code should i use? or there is something wrong with my receiver? thanks in advance!

AnalysIR, 2016-08-09:
try this sketch...to record long AC signals
[https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

you should then be able to use the output to resend using sendRaw of IRremote

Roseau, 2016-08-11:
Thank you for the help! and im terribly sorry for the late response. ive been trying the sketch that suggested and got this output from the serial
[http://imgur.com/a/qzrUr](url)
im using arduino mega 2560. And i put the output on pin 2. Am i doing something wrong?

AnalysIR, 2016-08-11:
make sure your selected baud rate matches what is defined in the sketch!

Roseau, 2016-08-11:
Alright! I got these following results, is this good enough? thanks in advance!

Press the button on the remote now - once only
Raw: (439) 3524, -1692, 448, -388, 444, -1276, 448, -388, 444, -392, 444, -388, 448, -388, 444, -388, 448, -388, 444, -392, 444, -388, 448, -388, 444, -392, 444, -388, 448, -1276, 444, -388, 448, -388, 444, -392, 444, -388, 448, -388, 448, -388, 444, -388, 448, -1276, 444, -1276, 444, -1280, 444, -388, 448, -388, 444, -1276, 444, -392, 444, -388, 448, -388, 444, -392, 444, -388, 448, -388, 448, -388, 444, -388, 448, -388, 448, -388, 444, -392, 444, -388, 448, -388, 448, -388, 444, -388, 448, -388, 448, -388, 444, -388, 448, -388, 448, -388, 448, -388, 444, -388, 448, -388, 448, -388, 444, -388, 448, -388, 448, -388, 444, -392, 444, -388, 448, -388, 448, -1276, 444, -1276, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -9964, 3496, -1696, 448, -388, 444, -1280, 444, -388, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -1276, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -388, 448, -388, 448, -1276, 444, -1280, 444, -1276, 444, -392, 444, -388, 448, -1276, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -388, 448, -388, 448, -388, 444, -392, 444, -388, 448, -388, 448, -388, 444, -392, 444, -1280, 444, -392, 440, -392, 444, -392, 444, -392, 444, -1276, 444, -392, 444, -392, 444, -392, 444, -1276, 444, -392, 444, -392, 444, -388, 448, -1276, 444, -392, 444, -388, 448, -388, 444, -392, 448, -388, 444, -392, 444, -388, 448, -388, 448, -388, 444, -1276, 448, -1276, 444, -1280, 444, -1280, 444, -1276, 444, -1280, 444, -1276, 444, -392, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -388, 448, -388, 448, -1276, 444, -1280, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -388, 448, -388, 448, -388, 448, -388, 444, -392, 444, -1276, 448, -1276, 444, -392, 444, -392, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -392, 444, -392, 444, -392, 444, -388, 444, -392, 444, -392, 444, -392, 444, -392, 444, -388, 448, -388, 448, -388, 444, -392, 444, -392, 444, -392, 444, -388, 448, -388, 444, -1280, 444, -392, 444, -392, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -392, 444, -1280, 444, -1280, 444, -392, 440, -392, 444, -392, 444, -392, 444, -392, 444, -392, 444, -392, 444, -1276, 448, -412, 424, -1300, 420, -1304, 420, -1304, 420, -1300, 424, 

Raw: (0) 
Press the button on the remote now - once only
Raw: (99) 3524, -1720, 444, -392, 444, -1276, 444, -388, 448, -388, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -388, 448, -1272, 448, -392, 444, -388, 448, -388, 448, -388, 444, -388, 448, -388, 448, -384, 448, -1276, 448, -1276, 444, -1300, 424, -408, 424, -388, 448, -1272, 448, -412, 424, -412, 424, -384, 452, -384, 448, -388, 448, -388, 444, -388, 448, -388, 448, -388, 448, -388, 444, -388, 448, -388, 448, -388, 444, -388, 452, -384, 448, -388, 448, -384, 448, -388, 448, -388, 448, -384, 448, -388, 448, 

Raw: (0) 
Press the button on the remote now - once only
Raw: (439) 3524, -1696, 444, -388, 448, -1276, 444, -388, 448, -388, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -388, 444, -392, 444, -392, 444, -388, 444, -1276, 448, -388, 444, -392, 444, -392, 444, -388, 448, -388, 448, -388, 444, -388, 448, -1276, 444, -1276, 444, -1280, 444, -388, 448, -388, 444, -1276, 448, -388, 448, -388, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -388, 444, -392, 444, -392, 444, -388, 448, -388, 448, -388, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -388, 448, -388, 448, -388, 444, -392, 444, -388, 448, -388, 448, -388, 444, -392, 444, -392, 444, -1276, 444, -1280, 444, -388, 448, -388, 448, -388, 444, -392, 444, -392, 444, -9940, 3496, -1724, 444, -388, 448, -1276, 444, -392, 444, -392, 444, -388, 448, -388, 448, -388, 444, -392, 444, -388, 448, -388, 448, -388, 444, -392, 444, -392, 444, -1276, 448, -388, 448, -388, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -1276, 448, -1276, 444, -1280, 444, -392, 444, -388, 448, -1276, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -388, 448, -1276, 444, -392, 444, -392, 444, -392, 444, -392, 444, -1276, 444, -392, 444, -392, 444, -388, 448, -1276, 444, -392, 444, -392, 444, -392, 444, -1276, 448, -388, 444, -392, 444, -392, 444, -392, 444, -388, 448, -388, 448, -388, 448, -388, 444, -392, 444, -1276, 448, -1276, 448, -1276, 444, -1280, 444, -1276, 444, -1280, 444, -1280, 444, -392, 444, -392, 444, -388, 448, -388, 444, -392, 444, -392, 444, -392, 444, -392, 444, -388, 448, -388, 448, -392, 440, -392, 444, -392, 444, -392, 444, -392, 444, -388, 448, -388, 448, -388, 444, -392, 444, -1280, 444, -1280, 444, -392, 444, -388, 448, -388, 448, -388, 444, -392, 444, -392, 444, -388, 448, -388, 448, -388, 448, -388, 448, -1276, 444, -1280, 444, -388, 448, -388, 448, -388, 444, -392, 444, -392, 444, -392, 444, -388, 448, -388, 448, -388, 448, -388, 448, -388, 448, -388, 444, -392, 444, -392, 444, -392, 444, -388, 448, -388, 448, -388, 448, -388, 448, -388, 444, -392, 444, -392, 444, -392, 444, -392, 444, -1276, 448, -388, 448, -388, 444, -392, 444, -392, 444, -392, 444, -392, 444, -392, 444, -392, 444, -392, 444, -1276, 448, -1276, 444, -392, 444, -392, 444, -392, 444, -392, 444, -392, 444, -392, 444, -392, 444, -1276, 448, -388, 448, -1276, 444, -1280, 444, -1280, 444, -1280, 444,

Roseau, 2016-08-11:
I found some raw codes have slightly different values. does those codes works? or there is a method to determine which code is correct?

AnalysIR, 2016-08-11:
You can easily verify by testing with your AC unit.

Roseau, 2016-08-11:
ill try that. Many thanks @AnalysIR ! :smile:

sid1202, 2017-01-02:
I was having similar issue trying to the remote for the Bell (Dish) 4100. Turns out the codes are 56Khz instead of 38Khz. Does this library support that frequency ? Is there a receiver that can read both 38 and 56 ?

Sid

travplan, 2017-01-26:
@sid1202 

I think I'm going through the same sort of thing as you. I have a Telus STB (Echostar De24) which as far as I can tell is the same as Bell and Dish. The IRrecvDumpV2 sketch i have been using returns all sorts   different results on the same repeated button press. After reading your comment, I have to wonder if its a frequency issue. I havent tried (or even looked at) the AnalysIR sketch yet, but I will give that some attention while waiting for your response.

Did you ever get to the bottom of this?

AnalysIR, 2017-01-26:
@sid1202 
> Turns out the codes are 56Khz instead of 38Khz. Does this library support that frequency ? 

Yes, IRremote supoorts all common carrier frequencies & should be fine anywhere from 30->60kHz when sending  (and likely beyond that range).

> Is there a receiver that can read both 38 and 56 ?

Standard IR receivers supported by the library are usually centered on one frequency. As the frequency moves away from this centre freq, the performance/ range degrades.

Quality IR receivers are also targeted at particular IR protocols, so you might have a 56kHz receiver that will not work well with some 56kHz protocols & similarly at other frequencies.

@travplan 
Sounds like your receiver may be an issue **OR** the Library cannot handle the protocol **OR** your wiring/interference could be an issue. Most likely the receiver.

sid1202, 2017-01-27:
@AnalysIR

Does this mean I would have a better chance of reading both frequencies with a 40 kHz sensor ? If so, could you recommend one to try ?

I'd like to use the 85 so I can only read 1 sensor.  Besides reading 2 sensor with 1 pin it not an easy task. I could just record the 38 kHz code in flash and send then when the are most often sent without actually reading them.

Also I was looking at the VISHAY line of sensors and I have not seen any that list Dish.

AnalysIR, 2017-01-31:
I don't have any specific suggestions at present. However, by increasing the IR power(current) it should perform better. Using a circuit like this:
[Constant current infrared LED emitter circuit](https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/)

you can control the IR power by changing the resistor value as explained in the post.

DISH is likely the brand name of the device/TV/STB. It does not mean that the IR protocol will have the same name (although often it does) so that can be a bit tricky.

In the absence of first-hand specifics, I would default to TSOP344XX, where XX is the carrier frequency.

As an aside, we will be launching a compact wideband IR receiver module soon (30-60kHz). You can make contact directly via [our website](https://www.analysir.com/blog/contact/) for more details.

anand021, 2017-07-17:
has anyone tried capturing Airtel DTH remote code ?? 
i am getting multiple different codes of unknown protocol on each button press..
Event the raw codes are different each time and tried with more than 12 Raw code for same button but the STB is not detecting the signal..
Does anybody has any idea what protocol they are using or any suggestions for capturing the codes??

bengtmartensson, 2017-07-17:
Please use a more precise description of the device you are dealing with. 

Having said that, [Global Cache's ControlTower](http://irdb.globalcache.com) has exactly one "Airtel" device, for which I generated [this IRremote code](https://gist.github.com/bengtmartensson/1ec8f5d31b6b88f8ea097fef72fe324a). Does this help?

anand021, 2017-07-17:
Thanks a lot @bengtmartensson , it worked for me too... 👍

bengtmartensson, 2017-07-18:
@Nice to hear.

This is what I did: Have ControlTower to email the codes. These consists of a CVS ("comma separated values") formatted email. This was copied to the clipboard in the mail reader. In IrScrutinizer, it was imported (Import -> Text -> Raw, name col. = 1, raw signal col. = 3 (or 2), field separator = , (comma)) form the clipboard. Subsequently it was exported as "Arduino Raw" to a file. For usage with the IRremote library, the pre-processor symbols were changed using a text editor (`#define IRREMOTE`).

abhijitdalavi, 2017-10-25:
hi sir ,
we are trying to decode the daikin  AC IR remote But we did not get any Library support from IRremote.h.
so which library should we use for Daikin ac ? and how can we decode the IR signals .
Thanks in advance.
Abhijit

AnalysIR, 2017-10-25:
AFAIK- IRremote does not generally support AC signals other than via RAW methods.

So record a raw signals and replay it using sendRAW, should work.

Otherwise, a search thru the open/closed issues & web search might help.

blackdevil8975, 2018-05-14:
Got the same problem :( 

Encoding  : UNKNOWN
Code      : 675AFC4B (32 bits)
Timing[11]: 
     +5250, -  50     +1700, - 100     +1300, - 400     + 800, - 500
     +  50, -2500     +  50
unsigned int  rawData[11] = {5250,50, 1700,100, 1300,400, 800,500, 50,2500, 50};  // UNKNOWN 675AFC4B

Encoding  : UNKNOWN
Code      : 60005DD7 (32 bits)
Timing[23]: 
     + 150, - 800     + 250, - 450     +1300, - 200     +1500, - 100
     +2450, -  50     +1600, -  50     +2550, -3200     +  50, -5050
     + 200, -1450     + 100, - 300     +1700, - 100     +7650
unsigned int  rawData[23] = {150,800, 250,450, 1300,200, 1500,100, 2450,50, 1600,50, 2550,3200, 50,5050, 200,1450, 100,300, 1700,100, 7650};  // UNKNOWN 60005DD7

Encoding  : UNKNOWN
Code      : 4CB0FADC (32 bits)
Timing[5]: 
     +9800, -1200     +  50, -1650     +  50
unsigned int  rawData[5] = {9800,1200, 50,1650, 50};  // UNKNOWN 4CB0FADC

Encoding  : UNKNOWN
Code      : 2F5CB35A (32 bits)
Timing[11]: 
     +2300, - 150     +1600, -  50     +1350, -  50     +3400, - 800
     + 100, -1600     + 100
unsigned int  rawData[11] = {2300,150, 1600,50, 1350,50, 3400,800, 100,1600, 100};  // UNKNOWN 2F5CB35A

Encoding  : UNKNOWN
Code      : D6C35DB (32 bits)
Timing[13]: 
     +7700, -  50     + 650, - 300     + 700, -1450     +  50, -1650
     +  50, - 350     +  50, - 400     +  50
unsigned int  rawData[13] = {7700,50, 650,300, 700,1450, 50,1650, 50,350, 50,400, 50};  // UNKNOWN D6C35DB

any suggestions?

AnalysIR, 2018-05-14:
> Got the same problem :(

Doubtful???....If it was the same problem then surely the same solution would apply!

Please open a new issue and follow the instructions for providing relevant information.
In your case, the make & model of the IR receiver/link. A photo of how you have wired it up. Plus **all** of the other stuff you will be prompted for when opening a new issue.

maxdd, 2018-08-03:
Could this be useful?
https://github.com/blafois/Daikin-IR-Reverse

shivansh9675, 2019-05-10:
Sir i hava a same problem(I am Currently trying to capture 'Tatasky' (India) TV remote codes , but i get different codes every time i press the same button.) 
but i don't what to do. please help and share  step by step guide

AnalysIR, 2019-05-10:
@shivansh9675 
I suggest you open a new issue and provide a lot more information.

shivansh9675, 2019-05-11:
### I am use this code to send raw data
#include <IRremote.h>
IRsend irsend;
void setup()
{

}

void loop() {
int khz = 38; // 38kHz carrier frequency for the NEC protocol
unsigned int irSignal[] = {9000, 4500, 560, 560, 560, 560, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 1690, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 39416, 9000, 2210, 560}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), khz); //Note the approach used to automatically calculate the size of the array.

delay(5000); //In this example, the signal will be repeated every 5 seconds, approximately.
}

### Use this code for receive that raw data
#include <IRremote.h>
int RECV_PIN = 11;
IRrecv irrecv(RECV_PIN);
IRsend irsend;
decode_results results;
void setup()
{
Serial.begin(9600);
irrecv.enableIRIn(); // Start the receiver

}

// Storage for the recorded code
int codeType = -1; // The type of code
unsigned long codeValue; // The code value if not raw
unsigned int rawCodes[RAWBUF]; // The durations if raw
int codeLen; // The length of the code
int toggle = 0; // The RC5/6 toggle state

// Stores the code for later playback
// Most of this code is just logging
void storeCode(decode_results *results) {
codeType = results->decode_type;
int count = results->rawlen;

if (codeType == UNKNOWN) {
Serial.println("Received unknown code, saving as raw");
codeLen = results->rawlen - 1;
// To store raw codes:
// Drop first value (gap)
// Convert from ticks to microseconds
// Tweak marks shorter, and spaces longer to cancel out IR receiver distortion
for (int i = 1; i <= codeLen; i++) {
if (i % 2) {
// Mark
rawCodes[i - 1] = results->rawbuf[i]*USECPERTICK - MARK_EXCESS;
Serial.print(" m");
}
else {
// Space
rawCodes[i - 1] = results->rawbuf[i]*USECPERTICK + MARK_EXCESS;
Serial.print(" s");
}
Serial.print(rawCodes[i - 1]);
}
Serial.println("");
}

}

void loop() {
// If button pressed, send the code.

if (irrecv.decode(&results)) {

storeCode(&results);
delay(300);
irrecv.resume(); // resume receiver
}

}

IN output it will not receive right raw code that is send by emitter.sometime it will receive NEC code some time diff. code
### **OUTPUT**
Received NEC: 20DF10EF
Received NEC: 20DF10EF
Received NEC: 20DF10EF
Received NEC: 20DF10EF
Received unknown code, saving as raw
 m850 s700 m700 s450 m1250
Received unknown code, saving as raw
 m1000 s500 m16500 s750 m6150 s3450 m58400 s1000 m4750 s550 m3250 s500 m13600 s450 m12050 s450 m21400 s850 m5300 s1100 m950 s400 m39550 s3300 m300 s750 m6150 s450 m100 s1350 m8000 s1500 m11700 s1250 m300 s1550 m6750 s300 m450 s800 m600 s400 m2800
Received unknown code, saving as raw
 m4000 s350 m18100 s700 m1050 s350 m850 s350 m2050 s450 m3150 s550 m550 s1550 m650
Received unknown code, saving as raw
 m11600 s350 m800 s300 m6900 s300 m10750 s700 m1700 s600 m450 s800 m1450 s550 m1000 s600 m2900 s900 m300
Received NEC: 20DF10EF
Received unknown code, saving as raw
 m4750 s800 m2000 s450 m750 s1650 m2100 s750 m3900 s400 m2400 s1050 m1250 s450 m2900 s300 m300 s300 m50 s1150 m3600 s800 m3000 s400 m400 s300 m50 s800 m50 s450 m1050 s300 m1000 s550 m200
Received unknown code, saving as raw
 m5250 s500 m500 s400 m5300 s400 m900 s500 m5450 s600 m250 s650 m5250 s400 m1750 s900 m3750 s450 m3400 s1000 m550 s450 m2350 s1250 m600
Received unknown code, saving as raw
 m6050 s600 m1500 s750 m950 s1050 m2400 s500 m2750 s550 m2200 s450 m2400 s1050 m350 s450 m2100 s400 m500 s300 m300 s350 m550 s350 m2150 s1000 m3600 s650 m900 s950 m50 s350 m800 s950 m2550
Received unknown code, saving as raw
 m9100 s300 m2750 s950 m50 s900 m200 s600 m5000 s650 m3500 s1100 m3900 s400 m50 s600 m2550 s350 m2050 s1250 m800 s500 m2450 s400 m2450
Received unknown code, saving as raw
 m1000 s1450 m4250 s700 m200 s300 m4450 s700 m900 s400 m800 s500 m400 s650 m0 s650 m500 s1350 m750 s2700 m950 s500 m5450 s600 m200 s1200 m1150 s500 m850 s1600 m1150 s700 m3000 s800 m750
Received unknown code, saving as raw
 m8800 s4600 m400 s700 m400 s700 m400 s1850 m400 s700 m400 s700 m400 s750 m450 s650 m400 s750 m350 s1850 m400 s1850 m400 s700 m400 s1850 m400 s1850 m350 s1850 m400 s1850 m350 s1850 m400 s750 m400 s700 m400 s700 m450 s1800 m400 s750 m350 s700 m450 s700 m400 s700 m400 s1850 m400 s1800 m400 s1900 m350 s700 m450 s1850 m350 s1850 m400 s1850 m400 s1800 m400
Received unknown code, saving as raw
 m500 s550 m200 s1300 m150 s1800 m200 s2950 m450 s2350 m400

yabinlove1, 2019-08-19:
> try this sketch...to record long AC signals
> https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/
> 
> you should then be able to use the output to resend using sendRaw of IRremote

I try use this code that you provide in this comment but i still read many different of raw 
and I try to choose one of them to test but it doesn't work 
So what should i do???
my remote is LG

AnalysIR, 2019-08-19:
1. Post the ouptut you got, with the last sketch.
2. Post the model # of your IR Receiver
3. Post the circuit diagramof your IR emitter (and/or a photo)
4. Post the platform you are using....eg Arduino UNO?

shivansh9675, 2019-08-19:
sir how to add new IR protocol in ESP8266 irremote.h  library

AnalysIR, 2019-08-19:
@shivansh9675 That library is a fork of the original IRremote. So best to ask your question on that repository, as it is not currently implemented here

PaulCmpbll, 2019-09-21:
Hi,

I'm trying to capture the IR codes for my electric wall fire remote.  The library says it's an unknown encoding.  Codes are not consistent, there are codes which appear more than others, but it depends on how long you press the button for.

I tried "IRScrutinizer" and it decodes the three buttons consistently as:
{38.0k,421,msb}<1,-3|3,-1>(A:12,-16)*{A=0xd88}
{38.0k,417,msb}<1,-3|3,-1>(A:12,-16)*{A=0xd84}
{38.0k,418,msb}<1,-3|3,-1>(A:12,-16)*{A=0xd82}

What I'm trying to work out is how do I get from the raw data to these codes ( I am assuming that 0xd82, 0xd84 and 0xd88 are the actual codes).   I realise that is probably a very open question, but maybe you could point me to some documentation on decoding these protocols.  I'm hoping this one should be fairly small/simple.

Raw code examples:
1232, 478,  1268, 398,  440, 1268,  1262, 424,  1262, 424,  418, 1270,  416, 1248,  412, 1296,  424, 1264,  418, 1268,  1266, 422,  390, 8028,  1282, 426,  1236, 430,  416, 1272,  1284, 424,  1264, 424,  420, 1246,  444, 1266,  418, 1268,  420, 1268,  422, 1266,  1238, 452,  420, 8020,  1266, 422,  1264, 422,  394, 1272,  1286, 402,  1284, 424,  418, 1268,  422, 1266,  422, 1264,  418, 1268,  418, 1268,  1262, 426,  418, 8020,  1262, 426,  1262, 424,  390, 1274,  1284, 404,  1286, 424,  418, 1268,  392, 1294,  390, 1296,  392, 1296,  392, 1296,  1264, 424,  392};  // UNKNOWN 18B1397

{1286, 424,  1290, 398,  420, 1268,  1286, 380,  1308, 400,  418, 1270,  418, 1268,  444, 1244,  444, 1242,  1290, 378,  466, 1220,  466, 7976,  1308, 400,  1288, 400,  418, 1270,  1288, 378,  1284, 426,  418, 1270,  444, 1244,  418, 1268,  420, 1248,  1308, 400,  442, 1244,  418, 8002,  1310, 376,  1310, 398,  444, 1244,  1288, 400,  1288, 398,  418, 1270,  418, 1270,  418, 1270,  418, 1270,  1290, 398,  418, 1268,  418, 8000,  1312, 398,  1288, 376,  440, 1268,  1286, 402,  1264, 424,  418, 1268,  418, 1248,  440, 1268,  418, 1270,  1262, 424,  442, 1222,  466};  // UNKNOWN EF1B843F

PaulCmpbll, 2019-09-21:
So I broke this down a little, having found the longer time around 8000uS.  So I have isolated the code down to:
+1265,-422,+1265,-422,+422,-1265,+1265,-422,+1265,-422,+422,-1265,+422,-1265,+422,-1265,+422,-1265,+422,-1265,+1265,-422,+422,-8023

Which http://irdb.tk gives me as:
Protocol F12, device 3, subdevice 1

So maybe I'll do some digging into that.  Obviously I have enough to send these codes, just tidying up the timing a little, but I also need to receive them in my project, so I'll look into how I can decode the raw myself.

Thanks.

AnalysIR, 2019-09-21:
Opening a new issue...would be a great help, intead of confusing this existing issue.
FYI: It decodes as Lasko on AnalysIR. The HEX of the last one is 0xD82, 12 bits. the other vsignal value is 0xD84.

[image]

saransh95, 2020-02-22:
> __**Board:_* ARDUINO NANO **Library Version:** 2.1.0 **Protocol:** Says UNKNOWN PROTOCOL_*
>
> I am Currently trying to capture 'Tatasky' (India) TV remote codes , but i get different codes every time [...]

were you able to find the ir codes? I do have a tata sky transfer box whose ir codes are completely diffrent from normal tata sky and I am not able to find ir codes for it.Let me know if you have any update for this.

shivansh9675, 2020-02-22:
use RCMM protocol to decode Airtel and Tata sky remote

closed 2020-07-04 by ArminJo

tech-freak, 2020-07-10:
> use RCMM protocol to decode Airtel and Tata sky remote
> […](#)

Can you share Airtel DTH remote codes?

AnalysIR, 2020-07-10:
@tech-freak  Did you notice that this issue is Closed?
:(