Arduino-IRremote/Arduino-IRremote #299: Add more details in library json

jefersonla, 2016-03-22:
Hello, can someone add "authors" to library.json file? I have a library that uses IRremote as a dependency, and is a little difficult add IRremote as a dependency on platformio. For example with this setup, platformio found 3 different libraries:
```json
{
      "name": "IRremote",
      "frameworks": "arduino",
      "platforms": "atmelavr"
}
```
**Platformio log**:
```
Installing dependencies:
Conflict: More than one dependent libraries have been found by request {"frameworks": "arduino", "name": "IRremote", "platforms": "atmelavr"}:
```
I think, if you add authors it will be more easier refine the search.

**P.S.**: According to [http://docs.platformio.org/en/latest/librarymanager/config.html](http://docs.platformio.org/en/latest/librarymanager/config.html#dependencies) there are only 4 requirements for dependency object, they are:
  * "name"
  * "authors"
  * "frameworks"
  * "platforms"

Best regards,
Jeferson

z3t0, 2016-03-23:
Hey @jefersonla Thanks for the suggestion I'll do that

jefersonla, 2016-03-23:
Thanks a lot @z3t0, I think this will remove the whole problem with the dependencies query, but your library.json seems to have problems:
```json
  [
      {
        name:"Rafi Khan",
        email:"zetoslab@gmail.com"
      }
      
      {
        name:"Ken Shirriff",
        email:"ken.shirriff@gmail.com"
      }
    ]
```
As you can see "name" and "email" need to be enclosed with **"**, and you forgot to put a **,** between the objects. The correct form is:
```json
[
     {
         "name":"Rafi Khan",
         "email":"zetoslab@gmail.com"
     },
     {
         "name":"Ken Shirriff",
         "email":"ken.shirriff@gmail.com"
     }
]
```
I don't know if this is a real problem, but is good avoid non default writting in json files.

z3t0, 2016-03-23:
oh haha my bad! I must have been half a sleep...

z3t0, 2016-03-23:
will merge #301 as soon as the pointless travis check passes :smile:

z3t0, 2016-03-23:
I've never actually looked into PlatformIO before and it seem really interesting and looks very similar to Github's atom. Looks like something interesting to look into

jefersonla, 2016-03-23:
PlatformIO is awesome, and the PlatformIO-IDE is based on the new versions of Github Atom, so like eclipse just open configurations and install the plugin you want, in this case the plugin name is PlatformIO IDE, after that, you'll have an IDE with code lint and dependency manager, much better than the Arduino IDE :)

closed 2016-07-11 by z3t0