Arduino-IRremote/Arduino-IRremote #307: Which pins should i use on microduino core+ (Avoiding any tx/rx pin)

dmsherazi, 2016-04-21:
**Board:** Microduino Core+
**Library Version:** 2.1.0

I am using [microduino core + ](https://wiki.microduino.cc/index.php?title=Microduino-Core
) and I am already using both the uarts. Which pins I can use for sending.

AnalysIR, 2016-04-21:
For the Atmega644P variant the library defaults to pin D14 for IR Tx. If you have the Atmega1284P variant, it is not defined in the library, at present,

If you are lucky, the defaults may output on D14 for both variants???

dmsherazi, 2016-04-21:
Thanks alot @AnalysIR ..

I would try this

dmsherazi, 2016-04-21:
I am having an issue I am trying the [LGACSendDemo.ino](https://github.com/z3t0/Arduino-IRremote/blob/3aebf42ca8daa865ce8ae572245328e7443e438e/examples/LGACSendDemo/LGACSendDemo.ino) .

When I observe the IR Led via a camera its ON all the time. I have connected it so that it sinks to the microcontroller ( **cathode **connected to processor while **anode **connected to **VCC**).

dmsherazi, 2016-05-17:
@AnalysIR  is this the correct connection?? https://camo.githubusercontent.com/cfcb72faa33d7f812e8496000bb62b09b6da9b76/687474703a2f2f617263666e2e636f6d2f696d616765732f69722d736368656d617469632e706e67

AnalysIR, 2016-05-17:
That looks fine for a standard Arduino. (eg UNO)

However, your Tx pin may be different as already discussed above.
Presumably you have defined pin 11 as the IR receive pin.

Regarding the IR receiver pinout...it looks fine, but as you have not stated the model number, it is not possible to confirm this. Different receivers will have different pinouts.

dmsherazi, 2016-05-17:
@AnalysIR  yes, basically I wanted to verify the direction of the sensor and tx led. For Tx Led is there any specifications? or just a standard IR led would be OK?

AnalysIR, 2016-05-17:
Yes a standard IR LED for remote control is fine. Make sure it is 944 nm (not 850nm). We use TSAL6100, TSAL6200 or TSAL6400.

The wiring above  for the LED is good. The longer lead is +ve and the shorter lead goes to GND.

dmsherazi, 2016-05-17:
@AnalysIR  great.. so I can take one from an old tv remote! :)

AnalysIR, 2016-05-17:
Yes - perfectly fine, but you will have to figure out which pin is -ve etc.

(Trace it back to the -ve terminal on the battery)

dmsherazi, 2016-05-17:
Yup.. Thanks alot @AnalysIR

closed 2016-05-17 by dmsherazi