Arduino-IRremote/Arduino-IRremote #310: Incorrect codes on Atmega328 with 8MHz clock

jBot-42, 2016-05-04:
**Board:** Atmega328 with internal 8MHz clock
**Library Version:** 2.1.0
**Protocol:** ??? 

I have tried the library on an Atmega328 with the internal 8MHz clock and a variety of remotes made between 1987 and 2010 and none seem to work correctly. The remotes are for a Philips DVD player,  a Sharp LCD TV, the 2nd generation Apple TV, and a Pioneer stereo. The results are unreliable. Pressing any button on any of the remotes returns a very long, constantly changing code. Based on the LIRC database, all of the codes I got are incorrect.

AnalysIR, 2016-05-04:
posting the 'unreliable results' may help?

jBot-42, 2016-05-04:
Here are some of the results:

Sharp GA535WJSA, pressing power, 1, 2, 3, 4, 5, 6, 7, 8, 9, 0 buttons in sequence:
`Encoding  : UNKNOWN
Code      : 6D5D1DB3 (32 bits)
Timing[31]: 
     + 200, - 850     + 200, - 300     + 200, - 300     + 200, - 300
     + 200, - 350     + 200, - 300     + 200, - 850     + 200, - 800
     + 200, - 350     + 200, - 800     + 200, - 350     + 200, - 300
     + 200, - 300     + 200, - 850     + 200, - 300     + 200
unsigned int  rawData[31] = {200,850, 200,300, 200,300, 200,300, 200,350, 200,300, 200,850, 200,800, 200,350, 200,800, 200,350, 200,300, 200,300, 200,850, 200,300, 200};  // UNKNOWN 6D5D1DB3

Encoding  : SANYO
Code      : FFFFFFFF (0 bits)
Timing[31]: 
     + 200, - 850     + 200, - 300     + 200, - 350     + 200, - 300
     + 200, - 300     + 200, - 350     + 200, - 800     + 200, - 850
     + 200, - 300     + 200, - 850     + 200, - 300     + 200, - 300
     + 200, - 350     + 200, - 800     + 200, - 350     + 200
unsigned int  rawData[31] = {200,850, 200,300, 200,350, 200,300, 200,300, 200,350, 200,800, 200,850, 200,300, 200,850, 200,300, 200,300, 200,350, 200,800, 200,350, 200};  // SANYO FFFFFFFF
unsigned int  data = 0xFFFFFFFF;

Encoding  : UNKNOWN
Code      : 6D5D1DB3 (32 bits)
Timing[31]: 
     + 200, - 850     + 200, - 300     + 200, - 350     + 200, - 300
     + 200, - 300     + 200, - 300     + 200, - 850     + 200, - 850
     + 200, - 300     + 200, - 850     + 200, - 300     + 200, - 300
     + 250, - 300     + 200, - 800     + 200, - 350     + 200
unsigned int  rawData[31] = {200,850, 200,300, 200,350, 200,300, 200,300, 200,300, 200,850, 200,850, 200,300, 200,850, 200,300, 200,300, 250,300, 200,800, 200,350, 200};  // UNKNOWN 6D5D1DB3

Encoding  : SANYO
Code      : FFFFFFFF (0 bits)
Timing[31]: 
     + 200, - 850     + 200, - 300     + 200, - 300     + 200, - 300
     + 250, - 300     + 200, - 300     + 200, - 850     + 200, - 800
     + 250, - 300     + 200, - 800     + 200, - 350     + 200, - 300
     + 200, - 300     + 200, - 850     + 200, - 300     + 200
unsigned int  rawData[31] = {200,850, 200,300, 200,300, 200,300, 250,300, 200,300, 200,850, 200,800, 250,300, 200,800, 200,350, 200,300, 200,300, 200,850, 200,300, 200};  // SANYO FFFFFFFF
unsigned int  data = 0xFFFFFFFF;

Encoding  : UNKNOWN
Code      : B341AB5D (32 bits)
Timing[31]: 
     + 200, - 800     + 200, - 350     + 150, - 350     + 200, - 300
     + 200, - 300     + 200, - 850     + 200, - 300     + 200, - 350
     + 150, - 350     + 200, - 300     + 200, - 300     + 200, - 350
     + 200, - 300     + 200, - 850     + 200, - 300     + 200
unsigned int  rawData[31] = {200,800, 200,350, 150,350, 200,300, 200,300, 200,850, 200,300, 200,350, 150,350, 200,300, 200,300, 200,350, 200,300, 200,850, 200,300, 200};  // UNKNOWN B341AB5D

Encoding  : SANYO
Code      : FFFFFFFF (0 bits)
Timing[31]: 
     + 200, - 800     + 250, - 300     + 200, - 300     + 200, - 300
     + 200, - 300     + 250, - 800     + 200, - 300     + 250, - 300
     + 200, - 300     + 200, - 300     + 200, - 300     + 250, - 300
     + 200, - 300     + 200, - 850     + 200, - 300     + 200
unsigned int  rawData[31] = {200,800, 250,300, 200,300, 200,300, 200,300, 250,800, 200,300, 250,300, 200,300, 200,300, 200,300, 250,300, 200,300, 200,850, 200,300, 200};  // SANYO FFFFFFFF
unsigned int  data = 0xFFFFFFFF;

Encoding  : UNKNOWN
Code      : BE2BA87F (32 bits)
Timing[31]: 
     + 200, - 850     + 200, - 300     + 200, - 300     + 200, - 350
     + 200, - 300     + 200, - 300     + 200, - 850     + 200, - 300
     + 200, - 300     + 200, - 350     + 150, - 350     + 200, - 300
     + 200, - 300     + 200, - 850     + 200, - 300     + 200
unsigned int  rawData[31] = {200,850, 200,300, 200,300, 200,350, 200,300, 200,300, 200,850, 200,300, 200,300, 200,350, 150,350, 200,300, 200,300, 200,850, 200,300, 200};  // UNKNOWN BE2BA87F

Encoding  : SANYO
[... 365 lines cut ...]
     + 300, - 750     + 350, - 200     + 300, - 750     + 300, - 200
     + 350, - 700     + 350, - 700     + 350, - 200     + 300, - 750
     + 350, - 150     + 350, - 200     + 350, - 150     + 350, - 750
     + 300, - 750     + 300, - 750     + 300, - 200     + 350, - 200
     + 300, - 200     + 350, - 700     + 350, - 700     + 350, - 200
     + 350, - 150     + 350, - 200     + 350, - 700     + 350, - 700
     + 350, - 700     + 350
unsigned int  rawData[67] = {4300,2050, 300,700, 350,200, 350,700, 350,200, 300,200, 350,700, 350,200, 300,750, 350,200, 300,750, 300,200, 350,700, 350,700, 350,200, 300,750, 350,150, 350,200, 350,150, 350,750, 300,750, 300,750, 300,200, 350,200, 300,200, 350,700, 350,700, 350,200, 350,150, 350,200, 350,700, 350,700, 350,700, 350};  // UNKNOWN 5FE20B96

Encoding  : UNKNOWN
Code      : FA322FB (32 bits)
Timing[67]: 
     +4300, -2000     + 350, - 750     + 300, - 200     + 350, - 700
     + 350, - 150     + 350, - 200     + 300, - 750     + 350, - 200
     + 300, - 750     + 300, - 200     + 350, - 700     + 350, - 200
     + 300, - 750     + 350, - 700     + 350, - 200     + 300, - 750
     + 300, - 200     + 350, - 200     + 300, - 200     + 350, - 700
     + 350, - 700     + 350, - 750     + 300, - 200     + 350, - 200
     + 300, - 200     + 350, - 700     + 350, - 700     + 350, - 200
     + 300, - 200     + 350, - 200     + 350, - 700     + 350, - 700
     + 300, - 750     + 350
unsigned int  rawData[67] = {4300,2000, 350,750, 300,200, 350,700, 350,150, 350,200, 300,750, 350,200, 300,750, 300,200, 350,700, 350,200, 300,750, 350,700, 350,200, 300,750, 300,200, 350,200, 300,200, 350,700, 350,700, 350,750, 300,200, 350,200, 300,200, 350,700, 350,700, 350,200, 300,200, 350,200, 350,700, 350,700, 300,750, 350};  // UNKNOWN FA322FB
`

I cannot figure out how to determine the model number of the IR receiver. There are very tiny numbers printed at the top that are too small to read. Based on appearance and pinout alone, it is a TSOP38238.

jrw42, 2016-06-11:
Today I just tried this again with the exact same configuration and a 16MHz Arduino. It did the same thing and after a few minutes the circuit started drawing 2 amps from the power supply and the sensor let out its magic smoke.

closed 2016-06-11 by jrw42