Arduino-IRremote/Arduino-IRremote #311: IR send problem

blackfoxtrot, 2016-05-04:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Sony 

I have a really weird problem. I want to control my sony tv with my IR led. so I use the basic code here :
#include <IRremoteInt.h>

#include <IRremote.h>
IRsend irsend;

void setup()
{
  Serial.begin(9600);
}

void loop() {
  
    for (int i = 0; i < 3; i++) {
      irsend.sendSony(0xa90, 12); // Sony TV power code
      delay(1000);
    }
  
}

And Normally my Tv have to switch off. BUT nothing change.

So for having more information, i took an other arduino with an IR receiver and I loaded the code for receiving the IR code.

So i had 2 arduino, one for sending and the other for receiving.

And my second arduino received well the sony code A90. So its weird my arduino understand the code but my tv dont.

Have anybody an idea for helping me ?

Thanks for your help

AnalysIR, 2016-05-04:
yes, with Sony you must send the burst at least 3 times, which your code does.

However, the gap is too long. Try 20 instead of 1000.

blackfoxtrot, 2016-05-04:
thanks really helpfull

blackfoxtrot, 2016-05-04:
still one issue : how can i improve the scope ? decreasing the gap ? or maybe increasing the number of time that i send the command ?

Do you have an idea ?

AnalysIR, 2016-05-04:
> how can i improve the scope  ?

Sorry I don't understand what you mean by this.

blackfoxtrot, 2016-05-04:
sorry i dont know how to say it in english. i want to say that i want to improve the distance needed for commanding my tv

AnalysIR, 2016-05-04:
Yes, I suspected that. I would say range or distance.
repeating the signal would not have any material impact.
the gap will have no impact on range, but altering duty cycle will.

If you post details about your IR emitter circuit, I may be able to make some suggestions.

see these blog posts:
https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/
https://www.analysir.com/blog/2014/10/03/driving-infrared-led-directly-arduino-pin/

or just google for IR LED circuit.

blackfoxtrot, 2016-05-07:
I use the exact same circuit as explain. I have an IR led and a resistor of 100ohm on the pin 3 of my uno.

Furthermore i still have an other question, when i use the irremote code for the detection of a code with one of my remote controleur, he say : 
Received unknown code, saving as raw
 m400 s350 m100 s300 m150 s600 m150 s350 m100 s600 m150 s800 m150 s350 m100 s350 m0 s350 m150 s650 m150 s600 m150 s450 m150 s650 m100 s400 m50 s400 m0 s850 m150 s350 m50

I understood that it doesnt know the kind of code that i m sending

but how can i include this code in my library or how can i send it with my led

Can you help me ?

AnalysIR, 2016-05-07:
**Range:** You will have to create a transitor based circuit to improve the range.

**Unknown Signal:**
Best to use the IRrecDumpV2 example sketch and then use the output with the SendRAW function.

However, It looks like your IR receiver is not handling the signal well. Also, I suspect you will not get a good decode with IRremote at all (It looks like a very difficult signal to record...might be XMP, RCMM or RECS80. Try this sketch instead (for improved timings)
[https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/](url)

You could also try the free IRLib library or [AnalysIR ($)](https://www.analysir.com/blog/get-analysir/)

closed 2016-07-11 by z3t0

Hengki-7idbox, 2017-11-27:
anybody want me to ask and ask for help?
I follow the tutorial from AnalysIR and get the Air conditioner code like this

(Button ON

Raw: (211) 3880, -1816, 540, -412, 544, -1308, 544, -408, 544, -1312, 544, -408, 544, -1308, 544, -408, 548, -1308, 544, -408, 544, -1336, 516, -412, 544, -1308, 544, -1308, 544, -412, 544, -1308, 544, -408, 544, -1308, 544, -1312, 544, -1308, 544, -1308, 548, -408, 544, -408, 544, -1308, 544, -1312, 544, -408, 544, -408, 544, -408, 544, -408, 548, 542, -408, 544, -408, 544, -408, 544, -1336, 520, -408, 544, -1308, 544, -408, 544, -408, 548, -436, 516, -412, 544, -1336, 516, -412, 544, -408, 544, -408, 544, -408, 548, -1308, 544, -408, 544, -436, 516, -1312, 544, -1308, 544, -408, 544, -412, 540, -436, 520, -408, 544, -408, 544, -436, 516, -436, 520, -408, 544, -408, 548, -408, 544, -408, 544, -412, 544, -408, 544, -1308, 544, -408, 548, -408, 544, -408, 544, -408, 548, -408, 544, -408, 544, -408, 548, -404, 548, -408, 544, -408, 544, -408, 548, -1308, 544, -408, 544, -408, 548, -404, 548, -408, 544, -408, 548, -404, 548 , -408, 544, -408, 548, -404, 548, -404, 548, -1308, 544, -408, 548, -1308, 544, -1308, 544, -1308, 548, -1308, 544 , -1308, 544, -408, 548, -408, 544, -408, 544, -1308, 548, -408, 544, -408, 544, -408, 520,

when I input into Sketch arduino, like this:

#include <IRremote.h>
 
IRsend irsend;

unsigned int  rawData[211] = {-1812, 544, -436, 516, -1336, 520, -408, 544, -1308, 544, -436, 520, -1308, 544, -436, 516, -1308, 544, -412, 544, -1308, 544, -408, 544, -1312, 544, -1308, 544, -408, 544, -1312, 544, -408, 544, -1308, 544, -1308, 548, -1308, 544, -1308, 544, -408, 548, -408, 544, -1308, 544, -1308, 544, -412, 544, -408, 544, -408, 544, -412, 544, -1308, 544, -436, 516, -408, 548, -408, 544, -1308, 544, -1312, 540, -412, 544, -1312, 540, -412, 540, -412, 544, -1312, 540, -1332, 520, -1312, 540, -412, 544, -408, 544, -412, 540, -412, 544, -1332, 520, -408, 544, -412, 540, -412, 544, -1312, 540, -436, 516, -412, 544, -1308, 544, -1308, 544, -412, 540, -412, 544, -408, 544, -412, 540, -408, 548, -408, 544, -412, 540, -412, 540, -412, 544, -412, 540, -436, 516, -412, 544, -408, 544, -1308, 544, -412, 540, -412, 544, -408, 544, -408, 544, -412, 544, -408, 544, -412, 540, -412, 540, -412, 544, -408, 544, -408, 544, -1312, 544, -412, 540, -408, 544, -408, 544, -412, 544, -408, 544, -408, 544, -408, 544, -412, 544, -408, 544, -408, 544, -1312, 544, -408, 544, -1308, 544, -1312, 540, -1312, 544, -1308, 544, -1312, 540, -412, 544, -408, 544, -408, 544, -412, 544, -1308, 544, -408, 544, -408, 492};  // ON

int khz = 38; // 38kHz carrier frequency for the NEC protocol

void setup()
{
  Serial.begin(9600);
}
 
void loop() {
  if (Serial.read() != -1) {
    for (int i = 0; i < 3; i++) {
      
 irsend.sendRaw(rawData,211, 38);
 Serial.println("Sending");
  delay(100); //In this example, the signal will be repeated every 5 seconds, approximately.

}
}
}

it seems everything is right, but why not work. Please help me

what is sketch there is something wrong
or writing wrong code ir

AnalysIR, 2017-11-27:
so far I see 3 issues:

- you have not copied the timing data correctly into the buffer
- the buffer definition is correctly defined as unsigned, but you have entered negative numbers...so remove the minus signs
- the delay is too short ...change it back to 5 seconds

you should end up with something like:
`unsigned int rawData[**nnn**] = {3880, 1816, 540, -412, ....`

AnalysIR, 2017-11-27:
Sorry....happy to point out potential issues with your own work/efforts......but very few, if any,  will do all of the work for you.

If you search thru the issues you will find examples that work. All you have to do is replace the timings for your signals.

Hengki-7idbox, 2017-11-28:
thanks for giving information .. i appreciate it ...

for the 3 points you mean there are some more that I want to clarify for easy I understand:

Point number 1: almost all the code is different, maybe I am wrong to enter the code, I will try now
Point number 2: what all minus signs in the code should be deleted.
Point number 3: why 5 seconds? can you explain in more detail why should 5 seconds

AnalysIR, 2017-11-28:
#2: the minus indicates that IR is OFF...otherwise it is ON. However the value is stored in the MCU as an unsigned number...so no need for the minus sign.

#3: If you have a short delay(100 mSecs), The target system may not see the short gap as a new signal and assume it is all only very long signal & ignore it or get confused. 
1 second is probably OK...but I suggested 5 seconds initially to allow time for the signals to be sent, the target device to respond and most important to give you time to be able to see what is happening.

Hengki-7idbox, 2017-11-30:
thanks for explanation and help from AnalysIR.

finally air conditioner can be controlled by suggestion and opinion from AnalysIR.

can i discuss about further infrared via email or whatsapp ...
because i still need help about infrared. to make receive and receive infrared from one sketch but I do not quite understand. because i am new in arduino

I want to create something like this. [https://www.youtube.com/watch?v=HYpWGzBteQg&t=13s]

AnalysIR, 2017-11-30:
> can i discuss about further infrared via email or whatsapp ...

Generally, private support is not provided here or for free. Also, if it is not directly related to IRremote you could try elsewhere. It is always better when everyone benefits from the answers/solutions etc.

e.g.
[The Arduino Forum](https://forum.arduino.cc/)
Or our own [IRForum](https://irforum.analysir.com/)
...or any similar location.

Regarding the video you linked...yes that is possible with an Arduino & IRremote. You just need to do some research using Google etc...as there are many examples online that you can follow. 

As you are new to Arduino, maybe take some time to start with simple projects, until you build up some more experience. 

Everyone starts at the beginning & the learning process goes on for years+++ :)

Hengki-7idbox, 2017-12-04:
if I ask your help to make a sketch like that video for arduino, what is the cost you want, to help me?

AnalysIR, 2017-12-04:
You may be able to get some help by posting [here](https://forum.arduino.cc/index.php?board=26.0) on the Arduino forum. Its a good place to post if you want other people to write sketches for you.

Hengki-7idbox, 2017-12-04:
thank you for the info.

Hengki-7idbox, 2017-12-04:
I want to ask again ...

please see my sketch
this can be decode or reicieve but when sending the ir signal does not work ... can you explain the location of this sketch error

#define IRpin_PIN PIND
#define IRpin 2
char val;

// The maximum pulse we'll listen for - 65 milliseconds is a long time
#define MAXPULSE 65000

// Timing resolution 
#define RESOLUTION 20

int IRledPin =  3;      // IR emitter LED connected to digital pin 3

int buttonState1 = 0;        
int buttonState2 = 0;

// Store up to 100 pulse pairs 
uint16_t pulses[300][2]; // Pair is high and low pulse
uint8_t currentpulse = 0; // Index for pulses we're storing
uint8_t sendpulse = 0; // Index for send pulses

void setup(void) {
    // Set uart baudrate
  Serial.begin(9600);
  
  
  
}

void loop(void) {

   if( Serial.available() >0 ) {
   val = Serial.read();
   Serial.println(val);}

/*============================================================================================================================================================================================================================*/
if (val == 'A') {   
    Serial.println("Sending IR signal");
    
    
    
    SendIRCode();
    //delay(15);  // wait 15 milliseconds before sending it again
    //SendIRCode();  // repeat IR code if it is neccesary
    Serial.println(sendpulse);
    
    
 
    delay(5000);  // wait 5 seconds to resend the code
    }
/*============================================================================================================================================================================================================================*/
 if ((val == 'a') || (currentpulse != 0)){
   
   if (val == 'a'){
    Serial.println("Ready to decode IR!");
    }

/*============================================================================================================================================================================================================================*/
  uint16_t highpulse, lowpulse; // Temporary storage timing
  highpulse = lowpulse = 0; // Start out with no pulse length
  
    while (IRpin_PIN & (1 << IRpin)) {
     // count off another few microseconds
     highpulse++;
     delayMicroseconds(RESOLUTION);

     // If the pulse is too long, we 'timed out' - either nothing
     // was received or the code is finished, so print what
     // we've grabbed so far, and then reset
     if ((highpulse >= MAXPULSE) && (currentpulse != 0)) {
       printpulses();
       sendpulse=currentpulse;
       currentpulse=0;
       return;
     }
  }
  // we didn't time out so lets stash the reading
  pulses[currentpulse][0] = highpulse;
  
  // same as above
  while (! (IRpin_PIN & _BV(IRpin))) {
     // pin is still LOW
     lowpulse++;
     delayMicroseconds(RESOLUTION);
     if ((lowpulse >= MAXPULSE) && (currentpulse != 0)) {
       printpulses();
       sendpulse=currentpulse;
       currentpulse=0;
       return;
     }
  }
  pulses[currentpulse][1] = lowpulse;

  // we read one high-low pulse successfully, continue!
  currentpulse++;
  
 }
}

void printpulses(void) {
  
  // Start declaration
  Serial.print("unsigned int  ");
  Serial.print("rawData[");
  Serial.print((currentpulse *2)-1, DEC);
  Serial.print("] = {");
  
  for (uint8_t i = 0; i < currentpulse; i++) {
  if (i != 0){
    
    Serial.print("");
    Serial.print(pulses
* RESOLUTION);

    Serial.print(", ");
    }
    Serial.print("");
    Serial.print(pulses[1] * RESOLUTION);
    Serial.print(", ");
  }
  // End declaration
  Serial.print("};");  //
  // Comment
 
  delay(1000);
  

}

// This procedure sends a 38KHz pulse to the IRledPin for a certain # of microseconds.
void pulseIR(long microsecs) {
 
  cli();  // Turn off any background interrupts
 
  while (microsecs > 0) {
   // 38 kHz is about 13 microseconds high and 13 microseconds low
   digitalWrite(IRledPin, HIGH);  // 3 microseconds
   delayMicroseconds(10);         /* hang out for 10 microseconds, 
                                     you can also change this to 9 if its not working */
   digitalWrite(IRledPin, LOW);   // 3 microseconds
   delayMicroseconds(10);         /* hang out for 10 microseconds, 
                                     you can also change this to 9 if its not working */
 
   // So 26 microseconds altogether
   microsecs -= 26;
  }
 
  sei();  // Turn them back on
}
 
void SendIRCode() {

  Serial.print(" irsend.sendRaw ({");
    
  

  for (uint8_t i = 0; i < sendpulse; i++) {
  
    if (i != 0){
    delayMicroseconds(pulses
* RESOLUTION);

    Serial.print(pulses
* RESOLUTION);

    Serial.print(", ");
    }
    
    pulseIR(pulses[1] * RESOLUTION);
    Serial.print(pulses[1] * RESOLUTION);
    Serial.print(", ");
  }
  // End declaration
  Serial.print("},"); 
  Serial.print((sendpulse *2)-1, DEC);
  Serial.print(",38);//"); 
  
  
}

AnalysIR, 2017-12-04:
Please re-edit your last post and place all of the code inside the code tags "<>" in the edit menu-bar.

AnalysIR, 2017-12-04:
Based on a very quick review of you most recent code:
- It seems you are no longer using the IRremote library to send IR
- and I could not spend the time to review it in more detail.

The earlier code you poster above was much better, using sendRAW (plus the things I suggested changing)

Hengki-7idbox, 2017-12-05:
is it true how i make a problem in this github. please come and check.[Here](https://github.com/Hengki-7idbox/arduino-receive-and-send-IR-Universal-Remote/issues/1)

I think the first code or sketch is better and works. but I want to create a flexible like code or a second sketch.

anurafika28, 2021-03-30:
My IR transmitter not sending, how to fixed it?

ArminJo, 2021-03-30:
Seems you forgot to press the send button.