Arduino-IRremote/Arduino-IRremote #312: libraries with atmel

nupuranmol, 2016-05-06:
How do i use the libraries given here with atmel studio. I am trying to transmit nec code via atmega 8 using NEC protocol.

AnalysIR, 2016-05-06:
Atmel Studio has a tool (visual micro) which I believe can accept arduino sketches & AFAIK is free.

Alternatively, you could also strip the code down by removing other protocols, which should make it easier to port.

Alternatively, check out the following blog post:
https://www.analysir.com/blog/2015/09/01/simple-infrared-pwm-on-arduino-part-3-hex-ir-signals/
This may be simpler if you are running with a 16MHz crystal - using the SendHexNEC or  SendRawBuf  functions

z3t0, 2016-05-31:
That is correct. 

http://www.visualmicro.com/ 

Visual Micro integrates with Atmel Studio to add Arduino functionality, it is very easy to use and free.

closed 2016-06-26 by z3t0