Arduino-IRremote/Arduino-IRremote #314: IR not received when running Adafruit Neopixel

sylvain-perez, 2016-05-07:
**Board:** ARDUINO MICRO
**Library Version:** 2.1.0
**Protocol:** https://www.sparkfun.com/products/11759

https://gist.github.com/sylvain-perez/225c8bfa980ed96afd070e4576585469

Hi, first of all, thank you for your time and effort building this great lib.
When using the library alone, it works great but when I use it with the Adafruit Neopixel, IR code are not recognized.

See code. 
When I comment the "strip" code, IR code are always recognized.
When I uncomment the "strip" code, IR are rarely recognized.
When I uncomment the "strip" code and change delay from 10 to 1000 it works like a charm but my strip moves too slowly.

There seems to be a weird delay issue or could it be that this is a limitation of the IR remote lib?
If this is the case, do you guys have any idea on how to have things going on while no IR signal is received?

Any help is welcome.

Thank you.

  Sylvain

AnalysIR, 2016-05-07:
> There seems to be a weird delay issue or could it be that this is a limitation of the IR remote lib?

...or a limitiation of the neopixel lib....:)

It could be a timer conflict.
Try using a different timer with the Neopixel lib.

If you can't do that, you can usuually change the timer used by IRremote. (Although I haven't checked the timers available for this platform)

In file IRremoteInt.h you will find defines for ATmega32U4/Teensy 2/Leonardo and you can select different timers to use. I am not sure which pin mappings are used on the micros (for sending).
You will need to reconfigure the timer used (rx and/or Tx) & the pin mapping(for tx)

Alternatively, contact the supplier of the Arduino micro or neopixel for assistance as they may have done this already.

sylvain-perez, 2016-05-09:
Hi AnalysIR,

many thanks for your prompt response.
I am completely new in the world of electronics and micro-controller
programming.
I have a small background in programming C/C++/Java and I am learning this
new world!
Having said that, I actually googled a bit and found that other people
encounter the same issue with Neopixel lib.
In forums they mention that Neopixel disable interrupts and that's what is
actually causing the issue.
How? I barely have no idea.
Nor I understand how timer/delay are related to interrupts.

From what I learned here and there is that when interrupts are disabled
delay function does not work anymore.
That is a bit strange then because if Neopixel lib disable interrupts then
putting a delay of 1000 in my code should not work.
Maybe Neopixel lib disable interrupts just before sending data to the led
strip and enable them right after.
In that case, how come that reading IR signals right after Neopixel lib
enabled back the interrupts does only work when putting a delay of 1000?
Is it because enabling back interrupts is not instant and take time?

Regarding your suggestion about changing the timer in case of a timer
conflict I don't know where to try this in IRremote.
I opened IRremoteInt.h and looked for the section ATmega32U4 but it's very
technical and I must admit I don't understand which thing to uncomment and
which to comment.
By the way I am only receiving IR. If you could guide me on this I'll
appreciate.

I also tried IRlib but I still have the issue.

The only workaround I found is to check if the received code is relevant.
If not then wait 1000 and read again.
By pressing two times rapidly on the button I get the correct received
code. I know it's a poor solution :(
I thought that leaving the button pressed after the delay of 1000 would do
the trick but it did not help neither most probably because there is a
"header" signal that IRremote wait and is resent.

Anyway, thank you very much for your help.
If you feel like helping me by trying things out on my board, please let me
know.
I am sure it will be helpful for others as I saw I am not the only one in
this case.

Thank you.

  Sylvain

AnalysIR, 2016-05-09:
@sylvain-perez 
It sounds like you should be contacting the support of the vendor(s) where you purchased the hardware.

Unfortunately, I am not in a position to trouble shoot vendors boards/libraries, particularly if I don't have one to test. (? maybe someone else monitoring this issue may be in a position to assist)

My only suggestion for you is to also try the IRLib library, which is similar to IRremote. It has a lot of documentation.

If what you say is correct and interrupts are disabled, then it is very unlikely you will be able to get IRremote to work for rx or tx.

sylvain-perez, 2016-05-11:
Hi, 
many thanks for your response.
After several readings here and there I think I understood where the problem comes from and there is no solution with the current hardware.
I am going to explain it here with my simple words and propose a solution.

So, IRremote is based on interrupts.
Interrupts are event that occurs at some point in time and which trigger some code registered with that particular event.
In the case of IRremote, there is an interrupt attached to the change of state of a certain pin (the pin receiving IR signal, usually pin 11). This means that whenever an IR signal is received by the sensor, in my case a TSOP38238, then an interrupt is triggered and some code is executed in IRremote library to process the IR signal bits and keep the resulting IR structured information for later user by user code. 
Later when user asks the lib if it has received a IR signal (using the decode function of IRremote) it receives true if a signal has been received and decode_results structure is filled with the IR signal info, including the holy IR value. That's very basically how IRremote works.

Now, it's the turn of Adafruit Neopixel library. It is said everywhere that this library rely on a very strict timing to send data to the led strip. Since interrupts results in execution of piece of code attached to it and delay the sending of data to the led, they don't have the choice and disable all the interrupts before sending data to the led strip and enable them back immediately after. To make it crystal clear, before strip.show is called, all interrupts are enabled, then somewhere inside strip.show interrupts are disabled and finally before strip.show returns all interrupts are enabled again.

OK, the problem comes when while interrupts are disabled to allow the sending of data to the led strip, if an IR signal is received then IRremote interrupt does not trigger the library code to start receiving IR signal data. Right after interrupts are enabled back then IRremote interrupt is triggered because IR signal is still being received but we loose part of the beginning of that signal resulting in invalid IR value!

Please Analysis, correct me if I am wrong with anything I wrote above.

There is in my opinion and as far as what I read is correct no solution without a new piece of hardware.
The solution is to offload the IR signal processing to another micro controller and have that micro controller communicate with the first one by any mean. The most easy being I2C bus.

I am going to use an ATTiny85 micro controller to process IR signal and have my Arduino micro request received IR value from ATTiny85 using I2C.

Question for you now Analysis is: Does IRremote works on ATTiny85?
I read may be in old posts that support is somewhere in a .h file but it gives compilation errors.
Please, keep in mind that I am only interested in the RX part.

I will act like this: if IRremote supports ATTiny85, then I will use IRremote otherwise I will use low level code to listen to IR signal.

In any case, if you don't mind Analysis, I will post here my working solution. This could help others who encounter the same problem as me.

Thank you.

  Sylvain

AnalysIR, 2016-05-11:
Some comments
1. Irremote uses a timer to poll the state of the IR signal every 50(default) uSecs.

2. Depending on how the neopixel works (?), it should be possible to decode IR signals using simple polling of the the input pin, particularly for a on-off solution.

3. I just checked and the Attiny85 is defined in IRremote - so it shouldn't be a problem.

4. If I was doing it your way, I would use a Nano or Pro Mini. You can get them from China incl postage for $1.70/$1.25 respectively and dont have to compromise on resources/crystal etc.

Look forward to hearing how you get on.

sylvain-perez, 2016-05-12:
Thank you for the reply and suggestions AnalysIR.

Could you please elaborate a bit more on the first point: *it should be
possible to decode IR signals using simple polling of the the input pin,
particularly for a on-off solution.*
Can IRremote do that? Another piece of code doing it?

Thank you,

  Sylvain

AnalysIR, 2016-05-12:
> Could you please elaborate a bit more on the first point: *it should be
possible to decode IR signals using simple polling of the the input pin,
particularly for a on-off solution.*
Can IRremote do that? Another piece of code doing it?

You would have to write this yourself. But if you are working with only one protocol, it shouldn't be too difficult. Just allow for a timing tolerance of up to +/-200uSecs on pulses.

> Oh and by the way are those Arduino clones found on China website ok in terms of quality?
I mean, will they behave exactly as original Arduino boards you can buy in Europe for example for 10 times the price?

Functionally equivalent. Quality is good. Can take 3-4 weeks for delivery from China. The main difference is that they usually use the CH340 serial/USB chipset to communicate with the PC. Some people prefer to purchase one or more 'official' boards to financially support the Arduino team indirectly others will only purchase official boards. However, a lot of the purchase price will go to the manufacturers/distributors & Arduino now have a donation option on their website. Either way seems like a cost effective replacement for an ATtiny85, with a lot more onboard resources.

sylvain-perez, 2016-05-17:
Hi there!

I tried your suggestion and wrote a sketch to demonstrate how IR polling and Adafruit Neopixel can "work". Here it is: https://gist.github.com/sylvain-perez/1602739f6183e2d32c92c5b21a12e132

One issue however, since in my case first pulse last 8700uSecs (almost 10ms) then animation is imposed such a delay and cannot be faster. I can live with that.
But because movePixel function takes between 1ms and 2ms it can happened that the first pulse is lost and IR array is not accurate.

If someone wanst the animation to be slower, then it's just a matter of increasing the FIRST_MAXPULSE constant. In my case it's 500 times 20uSecs that is to say 10.000 uSecs (10ms). If someone wants the readIR function to block for more when no IR signal is received then just increase FIRST_MAXPULSE!
 
As a personal conclusion I would say that having IR polling and Adafuit Neopixel working on the same board works with some limitation explained above.
Nevertheless because sometimes I can miss first IR pulses (and not find the right pressed button) and because I don't feel like comparing pulse arrays between them with an error margin, I decided to go for a second board to detect IR signals and the first board responsible for controlling the leds and checking second board using I2C if an IR button has been pressed.
I followed your advice and order a "chineese" arduino nano.
I'll post here my conclusion and code for others who want a solution that is more flexible.

Thank you.

 Sylvain

AnalysIR, 2016-05-17:
Thanls for the update....I just had a very qucik glance at the code.
You might find it better if you only call readIR() if the input from the IR reciver goes LOW, which means you have no delays when no signal is active.

I guess you should close this issue now!   :)

sylvain-perez, 2016-05-18:
Doing so makes my animation too fast so instead of putting a delay after my movePixel function and risk to loose IR signal transition to low, I prefer to have to wait in the readIR function instead.

Regarding the issue, I closed it. Thank you so much for your time and precious advice to a newbie like me. Like I said, I'll update the issue with my findings on the solution with the second board.

sylvain-perez, 2016-05-18:
This is not an issue of IRremote. It's just that IRremote relies on interrupts and that Adafruit Neopixel library disable all interrupts when the show function is called leading to times where interrupt is triggered after IR signal emission has started leading to incomplete IR pulses and then invalid IR code.

sylvain-perez, 2016-05-18:
This is not an issue of IRremote. It's just that IRremote relies on interrupts and that Adafruit Neopixel library disable all interrupts when the show function is called leading to times where interrupt is triggered after IR signal emission has started. Consequence is incomplete IR pulses and then invalid IR code.

closed 2016-05-18 by sylvain-perez

jassing, 2016-05-18:
You could always use a cheap pro mini to handle the IR stuff and query over
a soft-serial or i2c interface....

sylvain-perez, 2016-05-18:
Hi jassing. If you read the whole thread you will discover that it's
exactly what I will do ;-)

Thanks.

sylvain-perez, 2016-06-19:
Hi there,

just wanted to share with you that I finally managed to finish my project using an Arduino Micro to drive the led strip and listen to I2C messages coming from an Arduino Nano which listen to IR signal with IRRemote and then send IR code over I2C.
It's great and it worked.
Here is the code that I used:

IR Receiver : https://gist.github.com/sylvain-perez/48d7feecf2784ccff0d9d1f126611527
Main Program (I omitted extra code to keep the minimum for folks to use): https://gist.github.com/sylvain-perez/2b69b4e0828d9601938ebcb2794c050d

Enjoy ;-)

IAmOrion, 2017-03-15:
Hi sylvain-perez,

I realise this issue is "closed" and also almost a year old, but I very recently faced this exact situation and have looked at your code to see how you did it (I've never needed to use I2C Master/slave before).

The examples I find connect A4 - A4 and A5 - A5 (for the SDA & SDL connections between the 2 arduinos) however in your code you have: 

#define I2C_SLAVE_ADDRESS 8

Do I need to wire pin 8 to something?? I'm a little confused by this?  

Many thanks

AnalysIR, 2017-03-15:
@IAmOrion 

> I realise this issue is "closed" and also almost a year old

Issues are closed for a reason! You should really open a new issue, and link back to this one if relevant, by typing the following:

#314

Also, I didn't notice anything specific about the IRremote library in your post???

IAmOrion, 2017-03-15:
Hi AnalysIR,

The reason I didn't open new was because the issue isn't with the IR Remote library, It's due to the way the NeoPixels work.  I came across this 'post' during my research and the user was having the same problems as myself.  I have since resolved my issue but using the I2C method sylvain-perez suggests.  I was just trying to get him to clarify a few things but after an hour or so I managed to figure it out.

Again, the issue is created due the NeoPixel Library disabling interrupts - the IR library is perfectly fine and works great.

Apologies if my comment dragged up and old issue to the top of the list.

Kind regards

sylvain-perez, 2017-03-20:
Hi @IAmOrion,

Sorry for the delay in responding to your message. Yes, I2C_SLAVE_ADDRESS
is the address of the slave Arduino on the I2C bus.
Pick any value you want as long as on it is the same on master and slave
programs.
Happy to see that my solution worked for you.

  Sylvain

marcmerlin, 2017-04-03:
Ok, if that helps anyone else who stumbles on this issue and discussion. I had the same problem, but I came up with a different solution
1) don't use the adafruit neopixel lib
2) use FastLED instead, and that one has more platform specific code. Specifically for teensy 3.1, due, and other ARM (not AVR) 32bit CPUs, it re-enables interrupts in the middle of updating pixels. Those CPUs are fast enough that they are waiting before sending the next signal to the neopixel line, and during that time, there is just enough time for the infrared ISR to run 
I have tested with a teensy 3.1, and IR + neopixel, and it works fine.
My code: https://github.com/marcmerlin/Neopixel-IR

I believe (not had time to test yet) that it should also work with an ESP8266 for the same reason, but there is no ESP8266 IR support yet. If I get the time, I'll add code for that.
ESP32 will also work because it can talk to neopixels using the RMT hardware support which works independently from the CPU and therefore does not require interrupts to be ever stopped.
Hope this helps someone

z3t0, 2017-04-10:
@marcmerlin can you create a new issue with that info and I'll label it as wiki so I add it to the wiki once I get around to that.

marcmerlin, 2017-04-10:
You're welcome to take my text and paste it in a new issue :) but there you go, done:
https://github.com/z3t0/Arduino-IRremote/issues/435

DarkAbsent, 2018-07-05:
I had the same problem.
After examining the IRremote library, I found the isIdle method.
**if (irrecv.isIdle ()) strip.show ();**
In my simple project of a lamp this was enough.
Maybe someone will come in handy this decision.

marcmerlin, 2018-07-06:
@DarkAbsent  this is fixed by using at least an ESP8266, EP32, or teensy v3 and using FastLED instead of Adafruit::Neopixel
more details; http://marc.merlins.org/perso/arduino/post_2017-04-03_Arduino-328P-Uno-Teensy3_1-ESP8266-ESP32-IR-and-Neopixels.html

mgrybyk, 2021-10-05:
> I had the same problem. After examining the IRremote library, I found the isIdle method. **if (irrecv.isIdle ()) strip.show ();** In my simple project of a lamp this was enough. Maybe someone will come in handy this decision.

this one worked for me like a charm! Thanks a lot