Arduino-IRremote/Arduino-IRremote #317: Arduino IRremote error

MCButterMonster, 2016-06-10:
I was compiling some code for my Arduino "smart car robot" and this error showed up when uploading,

C:\Program Files (x86)\Arduino\libraries\RobotIRremote\src\IRremoteTools.cpp:5:16: error: 'TKD2' was not declared in this scope

 int RECV_PIN = TKD2; // the pin the IR receiver is connected to

                ^

exit status 1
Error compiling for board Arduino/Genuino Uno.

Here is my code,
[http://pastebin.com/Dvb7EmvF](url)

Thanks!

AnalysIR, 2016-06-10:
see point #5 in the readme for this library

[https://github.com/z3t0/Arduino-IRremote/blob/master/README.md](url)

MCButterMonster, 2016-06-11:
How do I delete that?

z3t0, 2016-06-14:
What operating system are you using

z3t0, 2016-06-14:
Never mind. Just find the folder in your original post and delete that.

closed 2016-06-26 by z3t0