Arduino-IRremote/Arduino-IRremote #318: Arduino Zero/MKR1000 (ARM Cortex-M0+ based) support

q2dg, 2016-06-16:
I suppose the answer will be the same than here (https://github.com/z3t0/Arduino-IRremote/issues/313) but maybe I'm lucky...
Thanks!!

z3t0, 2016-06-16:
Hah you're in luck. I am currently actually working on this in cooperation with another community member. 

Thanks, 
Rafi Khan

q2dg, 2016-06-19:
Yeah!! :-D
Thanks!!

digitalhen, 2016-07-20:
Hi guys,

Just wondering if there's any progress on this

z3t0, 2016-07-25:
Sorry, I have been quite busy for a while. But I will be getting to this at some point soon.

trodderus, 2016-08-08:
I am very keen for MKR1000 support also!

Cheers

z3t0, 2016-08-08:
Hi just wanted to let you know I am currently stuck with life and so wont be contributing anything to this project for a while, but if anyone does get around to this I am monitoring PRs all the time and will merge them

codehArt, 2016-08-30:
Hello @z3t0  ,
I want to contribute, if you can guide me, I might help myself.

bengtmartensson, 2016-08-30:
[This article](https://forum.arduino.cc/index.php?topic=344552.0) can probably serve as a starting point.

"It's left as an exercise for the reader to weave the ZERO functions in to the generic IRremote library"

codehArt, 2016-08-30:
Hello,
 @bengtmartensson  thanks for your guidance,
However I found all configurations of particular board in IRRemoteInt.h file ,
but unfortunately I don't know which kind of data should be added in  IRremoteInt.h file for arduino mkr 1000 or zero, so I can use this library .

 even I already found where to add mkr1000 board in boards configuration but I am not succeeded to found timer config for arduino mkr1000 32 bit timer
and there is another problem of mkr that it is 32 bit controller so as per my assumption we have to create another 32 bit timer.

agn0stico, 2016-10-02:
Hi guys.
I am also very interested in this platform support. Has there been any progress by any chance?
I tried to play around a bit with the code in the article that **@bengtmartensson** mentioned. I moved some code of the prototype sketch into _IRemoteInt.h_ to configure timer and PWM, but rather than getting it to compile for the MKR1000 board I didn't have luck getting it to work.

agn0stico, 2016-10-03:
**@codehArt** as far as I understood to enable support for this platform you need to define new timer macros (the various **TIMER_...** in _IRemoteInt.h_) accessing the specific Time Registries (TC/TCC) for the MKR1000 processor: this is necessary to control the PWM used to modulate/demodulate the IR signal.

I tried to give a look at the [ZERO IRremote demo](https://github.com/manitou48/ZERO/tree/master/IRtest) and the [SAM D21 datasheet](http://www.atmel.com/Images/Atmel-42181-SAM-D21_Datasheet.pdf) to see if I can understand something. Unfortunately, I don't have any experience in such low-level programming so even if I'd like to contribute I'm not much of a help here...

z3t0, 2016-10-04:
Okay so essentially the way you add support for a new board is to figure out the timer settings and add that information into IRremote.h

So the first step is to dig into the avr manual of the specific micro controller then from there you can find the registers which need to be changed to achieve the correct pwm settings, such as frequency and overflow etc.

Once you have both of those figured out you can share them here and I can guide you further.

It's really quite trivial to add support for other boards but does take a bit of digging through manuals and may seem a bit complex due to the nature of actually programming the microcontrolling by flipping bits and such instead of the easy overlay Arduino provides.

agn0stico, 2016-10-07:
Hi guys,

I somehow figured out how to get it working. I basically used the code of the **ZERO IRremote demo** plus some minor changes and integrated the code into _IRremoteInt.h_.

I forked the master and implemented the code there. You can find the repo with the changes at [https://github.com/agn0stico/Arduino-IRremote-MKR1000](https://github.com/agn0stico/Arduino-IRremote-MKR1000).

As far as I can see the library works for my project and it should be enough for me. I'm only using the receiver part though... Please feel free to try it out!

z3t0, 2016-10-08:
Thanks for sharing @agn0stico I took a quick glance and although it has laid the groundwork it is missing some of the finer details and instead implements some costly shortcuts. For example it uses digitalWrite to actually "blink" the led which is highly unrecommended as there is far too much overhead when calling such an operation for something this time sensitive.

However, since a lot of the ground work has been laid i will see if i can get a chance and build on that.

> I somehow figured out how to get it working. I basically used the code of the ZERO IRremote demo plus some minor changes and integrated the code into IRremoteInt.h.

Can you share the link for the source of the changes? Thanks

agn0stico, 2016-10-08:
Hi **@z3t0**, thanks for finding the time of looking into the changes.
To start implementing I basically gave a look at the demo code that was shared in previous comments. You can find the repo at [https://github.com/manitou48/ZERO/tree/master/IRtest](https://github.com/manitou48/ZERO/tree/master/IRtest).

z3t0, 2016-10-09:
> The Teensy-LC is also using an ARM Cortex M0+ processor which is exactly the same processor architecture of the MKR1000 (perhaps different manufacturers)

The last I can remember Teensy-LC support has been finicky as it had stopped working at some point along the development and no one has had a chance to look at the problem, although if I'm correct the manufacturer provides a version of this library which does support Teensy-LC so maybe take a look into that.

But I don't think these both use the same chip, I'm fairly certain they are different. Please share some information as to this as I am unclear.

> CMSIS? The Arduino IDE 1.6.10 seems to link by default to the CMSIS lib

I was not aware of this and will take a look. but also note that Teensy support has mostly been worked on by the team at @PaulStoffregen and so the credit goes there and I can't really say much as to the details. 

But if i were to take a hunch, I think Teensy does things differently from the Arduino IDE. Although, don't quote me on that.

z3t0, 2016-10-09:
Just thought I'd share the relevant [data sheet](http://www.atmel.com/Images/Atmel-42181-SAM-D21_Datasheet.pdf) and [pinout](http://www.giuseppecaccavale.it/wp-content/uploads/2016/04/MKR1000-Pinout-.png)

z3t0, 2016-10-09:
If we could find the arduino core definitions for the pins we could speed up this process greatly

agn0stico, 2016-10-09:
> But I don't think these both use the same chip, I'm fairly certain they are different. Please share some information as to this as I am unclear.

According to the [Teensy-LC specs](https://www.pjrc.com/teensy/teensyLC.html) the processor is indeed an ARM Cortex-M0+, produced by Freescale, vs the one produced by Atmel in the MKR1000. But as I said, I don't have any experience at this low-level so I might just be saying non-senses. I only noted that the processors architectures are the same and some Googling / talking to friends brought me to CMSIS. I have no clear idea then what the implications on programming are...

I know that @PaulStoffregen is doing a great effort to provide Teensy support for the most used libs so I was not proposing to take a look at that here. I was just throwing some ideas on the table :)

z3t0, 2016-10-09:
> I know that @PaulStoffregen is doing a great effort to provide Teensy support for the most used libs so I was not proposing to take a look at that here. I was just throwing some ideas on the table :)

I just meant if you're correct that they are the same processors then taking a look at their code and changing around the ifdefs might get you a working irremote library

PaulStoffregen, 2016-10-12:
Is Teensy LC support really not working?  So far, nobody has reported any issue.  Teensy 3.2 support was broken for some time.  I submitted a pull request a few months ago which fixed the issue, and expanded support to all clock speeds 8 MHz and higher (instead of only the original 3 speeds).

I must confess, it's been over a year since I've tested (and I no longer have the DVD player I was originally using for IRremote tests...)

Unfortunately, Atmel and Freescale (now NXP) designed their chips with very different timer hardware.  While the concept is similar, and pretty much the same as on AVR, the register names and finer details are completely different.  The code probably isn't very useful for porting to other chips, other than to show it can be done if you just dig into the timer hardware details and do careful testing.

agn0stico, 2016-10-12:
> Is Teensy LC support really not working? So far, nobody has reported any issue.

I think there was a misunderstanding between me and **@z3t0**. Before trying to implement specific changes for the Arduino MKR1000 I tried to use the same timer definitions that the Teensy LC was using, because I noticed that the two boards had the same processor architecture. I thought (hoped) that the timer implementation was done using some sort of common interface to the hardware (like CMSIS) but, as you said, the timer implementation uses specific registry names for that chip...

I'm straying from the point, but as far as I can tell we never really meant that there are currently issues with Teensy LC.

z3t0, 2016-11-06:
> Is Teensy LC support really not working? 

It's been a while so I can't recall whether the issue was fixed but as far as I can remember #172 records that only receiving works.

BriscoeTech, 2017-02-08:
Though I would message to see what the status of this project is since it has been afew months since the last post.

I have used the IRrecvDumpV2 and it appears to be receiving, I will be using my new IR analyzer when it arrives to confirm that the messages are correct.

z3t0, 2017-02-08:
I haven't had a chance to take a look in a while as I've been swamped with school... I do have a plan to make some major changes and add support for three boards lying around at home (including the Zero) but that won't be within the next few months...

Adminius, 2017-03-09:
Solution for faster blinking:
https://forum.arduino.cc/index.php?topic=334073.msg2308565#msg2308565

Adminius, 2017-04-11:
Adafruit has allready added support for M0:
https://github.com/adafruit/Arduino-IRremote

andrea689, 2019-12-05:
this library works with MKR1000:
https://github.com/cyborg5/IRLib2

ArminJo, 2020-10-17:
Samd (and Zero) is supported now.

closed 2020-10-17 by ArminJo