Arduino-IRremote/Arduino-IRremote #320: How to receive at multiple pins ?

Pikesh, 2016-06-19:
If I want to receive at both pin11 & pin10, what shall I do ?

Pikesh, 2016-06-19:
#include<IRremote.h>

IRrecv irrecv2(10);
IRrecv irrecv(11);

decode_results results;
decode_results results2;

void setup()
{
  Serial.begin(9600);
  // Start the receiver
  irrecv2.enableIRIn();
  irrecv.enableIRIn(); 
  
}

void loop()
{
  
  if (irrecv2.decode(&results))
    {
     Serial.println(results.value, HEX);
     irrecv2.resume(); // Receive the next value
    }
  
  if (irrecv.decode(&results2))
    {
     Serial.println(results2.value, HEX);
     irrecv.resume(); // Receive the next value
    }
    
    
}

For this code only pin11 is receiving, When I interchange line 3 & 4, pin10 receives only

z3t0, 2016-06-26:
I don't believe multireceiver functionality has yet been implemented #20 should point you in the right way however

closed 2016-06-26 by z3t0

Pikesh, 2016-06-26:
https://github.com/etix/Arduino-IRremote/commit/23caf5a832d05cfaf87fe8b792c5caf83b51afe8

this is Etix repository and its working fine for multiple receivers