Arduino-IRremote/Arduino-IRremote #322: IRremote with ATtiny2313 using arduino uno

harishshankar, 2016-06-22:
**Board:** ARDUINO UNO

i got this error when i upload IRremote program to ATtiny2313 .
can anyone fix this error 
thank you.

error:

Arduino: 1.6.9 (Windows 10), Board: "ATtiny x313 series, ATtiny2313, 1 MHz (internal), B.O.D. Disabled, no"

Build options changed, rebuilding all
In file included from C:\arduino-1.6.9\libraries\IRremote\IRremote.h:24:0,

                 from C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:1:

C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp: In member function 'void IRrecv::enableIRIn()':

C:\arduino-1.6.9\libraries\IRremote\IRremoteInt.h:239:3: error: 'TCCR2A' was not declared in this scope

   TCCR2A = _BV(WGM21); \

   ^

C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:125:2: note: in expansion of macro 'TIMER_CONFIG_NORMAL'

  TIMER_CONFIG_NORMAL();

  ^

In file included from c:\arduino-1.6.9\hardware\tools\avr\avr\include\avr\io.h:99:0,

                 from c:\arduino-1.6.9\hardware\tools\avr\avr\include\avr\pgmspace.h:88,

                 from C:\Users\haris\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.1\cores\tiny/Arduino.h:8,

                 from C:\arduino-1.6.9\libraries\IRremote\IRremoteInt.h:24,

                 from C:\arduino-1.6.9\libraries\IRremote\IRremote.h:24,

                 from C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:1:

C:\arduino-1.6.9\libraries\IRremote\IRremoteInt.h:239:16: error: 'WGM21' was not declared in this scope

   TCCR2A = _BV(WGM21); \

                ^

C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:125:2: note: in expansion of macro 'TIMER_CONFIG_NORMAL'

  TIMER_CONFIG_NORMAL();

  ^

In file included from C:\arduino-1.6.9\libraries\IRremote\IRremote.h:24:0,

                 from C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:1:

C:\arduino-1.6.9\libraries\IRremote\IRremoteInt.h:240:3: error: 'TCCR2B' was not declared in this scope

   TCCR2B = _BV(CS20); \

   ^

C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:125:2: note: in expansion of macro 'TIMER_CONFIG_NORMAL'

  TIMER_CONFIG_NORMAL();

  ^

In file included from c:\arduino-1.6.9\hardware\tools\avr\avr\include\avr\io.h:99:0,

Multiple libraries were found for "IRremote.h"
                 from c:\arduino-1.6.9\hardware\tools\avr\avr\include\avr\pgmspace.h:88,

 Used: C:\arduino-1.6.9\libraries\IRremote
 Not used: C:\arduino-1.6.9\libraries\RobotIRremote
                 from C:\Users\haris\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.1\cores\tiny/Arduino.h:8,

                 from C:\arduino-1.6.9\libraries\IRremote\IRremoteInt.h:24,

                 from C:\arduino-1.6.9\libraries\IRremote\IRremote.h:24,

                 from C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:1:

C:\arduino-1.6.9\libraries\IRremote\IRremoteInt.h:240:16: error: 'CS20' was not declared in this scope

   TCCR2B = _BV(CS20); \

                ^

C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:125:2: note: in expansion of macro 'TIMER_CONFIG_NORMAL'

  TIMER_CONFIG_NORMAL();

  ^

In file included from C:\arduino-1.6.9\libraries\IRremote\IRremote.h:24:0,

                 from C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:1:

C:\arduino-1.6.9\libraries\IRremote\IRremoteInt.h:241:3: error: 'OCR2A' was not declared in this scope

   OCR2A  = TIMER_COUNT_TOP; \

   ^

C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:125:2: note: in expansion of macro 'TIMER_CONFIG_NORMAL'

  TIMER_CONFIG_NORMAL();

  ^

C:\arduino-1.6.9\libraries\IRremote\IRremoteInt.h:242:3: error: 'TCNT2' was not declared in this scope

   TCNT2  = 0; \

   ^

C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:125:2: note: in expansion of macro 'TIMER_CONFIG_NORMAL'

  TIMER_CONFIG_NORMAL();

  ^

C:\arduino-1.6.9\libraries\IRremote\IRremoteInt.h:222:30: error: 'TIMSK2' was not declared in this scope

 #define TIMER_ENABLE_INTR   (TIMSK2 = _BV(OCIE2A))

                              ^

C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:128:2: note: in expansion of macro 'TIMER_ENABLE_INTR'

  TIMER_ENABLE_INTR;

  ^

In file included from c:\arduino-1.6.9\hardware\tools\avr\avr\include\avr\io.h:99:0,

                 from c:\arduino-1.6.9\hardware\tools\avr\avr\include\avr\pgmspace.h:88,

                 from C:\Users\haris\AppData\Local\Arduino15\packages\ATTinyCore\hardware\avr\1.1.1\cores\tiny/Arduino.h:8,

                 from C:\arduino-1.6.9\libraries\IRremote\IRremoteInt.h:24,

                 from C:\arduino-1.6.9\libraries\IRremote\IRremote.h:24,

                 from C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:1:

C:\arduino-1.6.9\libraries\IRremote\IRremoteInt.h:222:43: error: 'OCIE2A' was not declared in this scope

 #define TIMER_ENABLE_INTR   (TIMSK2 = _BV(OCIE2A))

                                           ^

C:\arduino-1.6.9\libraries\IRremote\irRecv.cpp:128:2: note: in expansion of macro 'TIMER_ENABLE_INTR'

  TIMER_ENABLE_INTR;

  ^

exit status 1
Error compiling for board ATtiny x313 series.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

AnalysIR, 2016-06-22:
Hi 
It appears the ATtiny2313 is not a supported platform for this library.

However, your error looks like it is related to Timer2 and I suspect the ATtiny2313 does not have a Timer 2. (?)

One possible approach would be to change the default timer to Timer1 from Timer2. You will just need to change the deafult setting around lines 206 & 207 of IRremoteInt.h.

So comment the line for Timer2 and uncomment the line for Timer1 after the #else 

There is no guarantee this will work, but it is the best approach. If not, search for any fork of IRremote that may support this MCU.

z3t0, 2016-07-11:
@harishshankar This board is currently unsupported. We are currently in the process of adding support for a number of boards so we are unavailable to work on adding support for this particular board, however you are free to work on it and let us know via a PR if you get anywhere.

If you do decide to try and add support yourself, feel free to create an issue to keep track of your progress while also being able to ask the community questions if you hit any walls.

closed 2016-07-11 by z3t0