Arduino-IRremote/Arduino-IRremote #327: Compile Error Buffer Overflow

Felarof23, 2016-06-30:
**Board:** ATTINY85
**Library Version:** 2.1.0
**Protocol:** 

**Code Block:**
```c

#include <IRremote.h> 

.....

```
I get this error when I compile for Attiny85 through Arduino UNO as ISP.

my code: http://pastebin.com/uB8GLBLc
the error output: http://pastebin.com/Zu3cCxBx

I use Arduino IDE 1.6.9 on Windows 10 whit their Arduino ISP sketch.

The attiny support taken from : https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json

Thank you very much

P.D. Sorry for my Google english xD

AnalysIR, 2016-07-02:
look like a memory issue.

try to compile in only the protocols you need by using the #defines provided

e.g. in IRremote.h

```
...
#define DECODE_RC5           1  //
#define SEND_RC5 1
...
```

by setting the 1 to 0 for any protocol on the send or decode #define, you can reduce the Flash required.

I suggest you set them all to 0 and set the option you require to 1

Felarof23, 2016-07-05:
Thank you, it's works fine, now i'm fightinhg with irsend because it does not send the code correctly

Grettings!

closed 2016-07-05 by Felarof23