Arduino-IRremote/Arduino-IRremote #329: LG decode - something incorrect

Gustavomurta, 2016-07-04:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** LG

**Code Block:**
```c

#include <IRremote.h> 

.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided
- [] The title of the issue is helpful and relevant 

The above is a short template allowing you to make detailed issues!

Testing your library, I found some inconsistent results. 

See my tutorial about decoding IR codes (only in portuguese) 
If press channel UP I found code **04 FB 00 FF**. ( correct code - LG Manual) 
With your code , I found **20 FD 00 FF.** 
The correct decode uses least significant bits first. 

http://labdegaragem.com/profiles/blogs/6223006:BlogPost:315534?id=6223006%3ABlogPost%3A315534&page=4#comments

Regards , 

Gustavo Murta ( Brazil) 
jgustavoam@gmail.com

bengtmartensson, 2016-07-04:
Just a comment:

When analyzing IR protocols, there is a large amount of arbitrary decisions involved. This amounts both to the naming of the protocols and their parametrization. So for example, the protocol here referred as "LG" is known in other communities (like jp1) as "NEC2". You are right, "normally" the parameters are interpreted in this protocol with the LSB first.

Having said that, with a non-standard parametrization, you will be consistent. The expressions "2 in LSB" or "64 in MSB"  are mathematically the same. The problem occurs when communicating with others.

This is not the only place where IRremote does it differently from "the rest of the world". Also the NEC signals (possibly others) suffer from the same problem.

Gustavomurta, 2016-07-04:
Bengt, 
I suggested change, because the official protocol of LG is least significant bits first. 
I think is not difficult to change Library to do correct thing. 

[http://www.proaudioinc.com/Dealer_Area/RS232.pdf](url)

Regards. 

Gustavo Murta

z3t0, 2016-07-11:
I think maybe we should look into finding a source of protocol definitions and go from there, as @bengtmartensson has mentioned the naming conventions used are rather arbitrary, so if we used fixed definitions then we could cite them in our implementations to clear things up.

z3t0, 2016-07-11:
Added a note in #331

closed 2016-07-11 by z3t0