Arduino-IRremote/Arduino-IRremote #330: Arduino Nano V3 has not decoding properly

antonyenergy, 2016-07-07:
MARK_EXCESS    100
in IRremoteInt.h should be set to 0 to make it wokrk.
As for comment in code
// Due to sensor lag, when received, Marks  tend to be 100us too long and
//                                   Spaces tend to be 100us too short
Probably arduino Nano has no this lag.

z3t0, 2016-07-11:
I have no way to test this so I cannot confirm it, but if you want feel free to create a pull request with the change, and using a define to check for nano v3, and I will put it on a nano branch in case anyone wants to test

bengtmartensson, 2016-07-11:
Probably a property of the sensor (which the OP does not specify), not the processor board.

z3t0, 2016-07-11:
@bengtmartensson Agreed.

@antonyenergy can you share with us the details of your sensor?

antonyenergy, 2016-07-28:
Sensor? You mean HX1838.[http://uamper.com/products/datasheet/HX1838.pdf](url)

closed 2020-08-17 by ArminJo