Arduino-IRremote/Arduino-IRremote #333: Protocols Implementation

z3t0, 2016-07-11:
@adamlhumphreys @Gustavomurta 

It has come to the attention of many users that many of the protocol implementations are questionable and possibly incorrect. This issue will be used to document the effort to find "reputable" sources for the specifications of the different IR protocols so that they can then be referenced and so the library has a way to explain what arbitrary names such as "LG" really mean.

#331

bengtmartensson, 2016-07-11:
I am happy that this issue has been brought up. Actually, it is an recurrent issue at remotecentral.com and hifi-remote.com that someone reports e.g. an NEC signal captured by IRremote, but the parametrization is different from the rest of the world, creating a fair amount of misunderstanding. 

I have been working on maintaining a list of IR protocols, [this one](https://raw.githubusercontent.com/bengtmartensson/harctoolboxbundle/master/IrpMaster/src/main/config/IrpProtocols.ini). It is in machine readable form, in the so-called [IRP notation](http://www.hifi-remote.com/wiki/index.php?title=IRP_Notation) a slightly cryptical, man- and machien readable notation for IR protocols. It is used in IrScrutinizer, which can generate the raw IR signals from the IRP form in the data base. It is based on the library DecodeIR by John Fine, with subsequent improvements by the [JP1 community](http://www.hifi-remote.com/forums). So I personally do not take credit for it, nor claim copyright. It is therefore compatible with DecodeIR (which is a library for IR signal decoding, used in IrScrutinizer, RemoteMaster, IrScope), and the "general consensus" on IR protocols and their interpretations in the JP1 community. It does not represent the "truth", for example in the sense of a standard by a standardization organization like ISO. IR protocols are not registered by a central registration, so there is a fair amount of arbitrariness involved. (For example, the protocol therein referred to as "denon" is by the company Denon called "Sharp" (!)). 

This is not a finalized list; improvements are welcome (possibly in particular to the documentation), but it should be kept consistent with the JP1 community and DecodeIR.

So if you are considering "canonicalizing", please consider being compatible with this list. Work is in progress for generating decoder/rendering code directly from this list... 

"LG" corresponds to "Nec2" in "my" list.

z3t0, 2016-07-11:
@bengtmartensson 

Thanks for the detailed comment, I will look into IRP Notation and see if I can find a way to make it work with this library. Also I think it would be a great idea to base it off some sort of community standard specification (such as "your"  list) for the community is the main target in either case.

The main concern is as you said to "canonicalize" so that a standard, albeit arbitrary?, is documented.

bengtmartensson, 2016-07-12:
I was "playing" with code generation today, and it struck me that there is so much more to "canonicalization"  than just renaming and changing bit order and such. For example, the Nec protocol: By "most", it is considered to have three 8-bit parameters, called D (device), S (subdevice) and F (function). In IRremote, all these are lumped together in a 32-bit integer; the forth byte being the one-complement of F. Then there is the different types of repeats, which is not really contained in the IRremote model. And the ending silence is essentially just ignored from IRremote.  Comments?

You may like to look at Infrared4Arduino as a comparison.

z3t0, 2016-07-12:
@bengtmartensson That's very interesting while also adding quite a fair bit of complexity. One idea I had a while back but never really worked on was that the library itself could provide a very basic implementation for sending arbitrary signals, basically only what is required to interact with the hardware. While the actual "protocol" specification can be abstracted away into "implementations" which will be within example sketch files. 

This kind of ignores the whole issue with irregularities between protocols with similar names and etc. because that way the basic protocol is given by the examples, with documentation, and then anyone wanting to add more specific functionality for the protocol can do so, e.g add different repeat patterns.

The main focus for this would be that we could make the footprint of the library smaller, and allow for explicitly "choosing" protocols without the whole library define mess we currently use :smile:

Thoughts?

bengtmartensson, 2016-07-13:
For the discussion, I think you should have a look at Infrared4Arduino, which I think should at least in principle satisfy your requirements. There are classes IrSequence and IrSignal; there are sending classes consuming IrSequences and IrSignals, and receiving classes producing IrSequences. There are rendering classes producing IrSignals from parameters, and decoding classes producing parameters from IrSequences. To use, just use the classes you want by selecting the right #include's in the sketch.

z3t0, 2016-07-13:
@bengtmartensson Thanks, that definitely seems like what I am looking for.

Just for discussion, what would you think about transitioning the library to the model I suggested above? Any pros or cons that come to mind?

adamlhumphreys, 2016-07-23:
Sorry I'm a bit late. I'd be for the idea of a basic sending implementation. Or course I may be biased in that I wrote a "protocol" for receiving most basic implementations that returned values consistent with the existing "protocols" in IRremote rather than a more arbitrary hash. (Of course I realize I've yet to add/update that to my pull request. Been working on a pinball/general purpose driver PCB.)

z3t0, 2016-07-24:
@adamlhumphreys okay then when i get time I will start working on something that can work as a basic sending implementation.

How's your side project going? :smile:

adamlhumphreys, 2016-07-26:
That'd be awesome. :D

Well, almost ready to have some barebones boards produced. Already sourced the general SMT parts I'd need to make the prototypes, (possibly the finals as well if I can get have a successful Kickstarter), but I'm still working out the crystal pad landing. I may just opt to support both the smaller and normal sized HC-49US SMT packages _and_ the classic through hole so I don't have to worry about paying crazy amounts for the smaller SMT just for the prototype. :simple_smile:

closed 2020-10-19 by ArminJo