Arduino-IRremote/Arduino-IRremote #335: Mitsubishi AC Protocol Creation

janvikalra, 2016-07-11:
**Board:** Arduino Uno
**Library Version:** 2.1.0
**Protocol:** Creation of Mitsubishi AC Protocol

If anyone is up to it, would you be so kind as to create a Mitsubishi **Electric** AC protocol? The current Mitsubishi protocol is for a Mitsubishi TV (16 bits) and, thus, doesn't work for an AC (32 bits)

Or, could you kindly guide me on what I would have to do create my own protocol?

z3t0, 2016-07-11:
Sure, could you find some documentation regarding the protocol. As per #333 we are trying to use names like "Mitsubishi" to make it easier to communicate ideas but are shifting towards documenting the actual specs of the protocol to clear up confusion

janvikalra, 2016-07-12:
Yes, I can provide as much documentation as needed. I am currently using the MSY-GE 18 VA Mitsubishi HVAC that is also compatible with MSY-GE 24 VA and MSY-GE 10 VA models. I have my Arduino set up with an IR Receiver and IR Transmitter, so I was able to print a raw dump through Ken Sheiff's IRrecv sketch in his IRLibrary. This is what prints in the serial monitor for the on/off button: 

[image]

Unfortunately, however, the hex value changes each time I press the on/off button. 
I checked out the Mitsubishi HVAC Protocol that I found on [github](https://github.com/r45635/HVAC-IR-Control) but that doesn't work because it's for the MSZ-GF60VE model :(

z3t0, 2016-07-12:
Would it be possible for you to try and record the signal being sent using some sort of oscilloscope?

I remember using a sketch when I was testing different remotes which allowed me to get much more detailed information by recording a signal on an arduino using something that I believe @bengtmartensson worked on. I'll see if I can find it

z3t0, 2016-07-12:
I think I was using this: https://github.com/bengtmartensson/harctoolboxbundle/tree/master/IrScrutinizer
Its very useful for analyzing unknown signals, once you are able to get the information for the signal we can move forward

janvikalra, 2016-07-13:
I am using a Mac an have downloaded IrScrutinizer multiple times (included different versions) but it crashes every time right after it launches. Would information from AnalysisIR about my remote work instead?

z3t0, 2016-07-13:
I believe so

bengtmartensson, 2016-07-13:
@janvikalra : create an issue at https://github.com/bengtmartensson/harctoolboxbundle/issues. There is no way I can fix problems that I am not aware of.

AnalysisIR is Windows only.

z3t0, 2016-07-13:
Just thought I'd mention that I have used irscrutinizer previously on OS X
and it has worked great.

janvikalra, 2016-07-13:
@bengtmartensson will do :) 
@zt30 hm, alright I'll try downloading it again on a another computer then.

z3t0, 2016-07-13:
@janvikalra  I could be wrong on this, but last I remember you need to install java so maybe that was missing

llemtt, 2016-07-13:
Hi, someone already tried this https://github.com/r45635/HVAC-IR-Control ?

(I compiled and uploaded but not tested yet)

z3t0, 2016-07-18:
@llemtt 
>Hi, someone already tried this https://github.com/r45635/HVAC-IR-Control ?
(I compiled and uploaded but not tested yet)

From OP:
>I checked out the Mitsubishi HVAC Protocol that I found on github but that doesn't work because it's for the MSZ-GF60VE model :(

llemtt, 2016-07-20:
I'm not sure my IR led is working properly (check out later today) but code in https://github.com/r45635/HVAC-IR-Control doesn't work on my MSZ-GA35 although my remote is identical to MSZ-GF60VE.

@z3t0 @janvikalra 

for MSY you should try also

https://github.com/ToniA/arduino-heatpumpir

janvikalra, 2016-07-21:
@llemtt you can check if your IR led is working by going to a dark room, setting the sketch so that a signal is being sent to it every 1 millisec and looking at the LED through your camera. Make sure to look at your LED from a bird's eye view. If it works you'll see a cherry-red light flickering. 
Also, thanks for the link! 

@z3t0 Thanks for agreeing to write a Mitsubishi protocol. While I was trying to get the specific data for this post, using AnalysIR, I actually solved my own problem. I found that sending the raw data that AnalyIR gave me and sending it via IrRemote's SendRaw worked like a charm. 

@llemtt you might want to try to the above if you can't manage to control your AC using any of the existing github codes.

llemtt, 2016-07-21:
Ok my IR led actually works! although low powered so I had to move closer...

https://github.com/r45635/HVAC-IR-Control  works on my MSZ-GA35 but it's not completely right, this suggests that the base "18 bytes" protocol is valid for more models but requires tweaking for each one.

Now I'm going to reverse mine's protocol (remote model KM05B).

I suggest to integrate the base protocol into this library while managing to design a way to distinguish different remote models. (maybe I'll do that once my project is finished)

llemtt, 2016-11-28:
Consider also some Mitsubishi have a serial port!

neskweek, 2017-11-11:
Hello there
@llemtt : I wanted to know if you had any success on reversing KM05B protocol and if so if you could explain this protocol.
I have those remote controls too for my AC and want to integrate those in Jeedom.

llemtt, 2017-11-13:
Hi neskweek, I'm sorry but since I've discovered the serial port I dropped my IR projects...

llemtt, 2018-11-05:
cn105

page 23 bottom middle just under jp34

neskweek, 2018-11-06:
Marvelous ! Thank you very much !

Suprabat, 2019-02-16:
> @llemtt you can check if your IR led is working by going to a dark room, setting the sketch so that a signal is being sent to it every 1 millisec and looking at the LED through your camera. Make [...]

Hi @janvikalra , did the link 
https://github.com/ToniA/arduino-heatpumpir provided by @llemtt actually work for your MSY? Thanks. I'm thinking of doing this project on my MSY too, so am just doing some initial research.

darkfader, 2019-06-03:
> Hi @janvikalra , did the link
> https://github.com/ToniA/arduino-heatpumpir provided by @llemtt actually work for your MSY? Thanks. I'm thinking of doing this project on my MSY too, so am just doing some initial research.

I have SRK25ZS-W and it seems to accept those codes. I was trying the codes with the checksum byte but didn't work. My Xiaomi IR Remote wanted compressed data too which was the other reason I didnt get it working. I better had started with the serial port in the first place..

lumanga, 2019-06-16:
Hi All, 
I have **Mitsubishi MSZ-FD25VA** and I'm having the same problem, to get always different IR code for the same button pressed and same settings.
To capture IR code sent from original remote (code KM09D 0314438) I'm using this windows software with BROADLINK MINI 3:
https://sourceforge.net/projects/broadlink-manager/

Could someone give me the file .json with the principal command for this model?
(I'm afraid to don't send all the code needed... including PLASMA,CLEAN,AREA,WideVane..) 

What happen if I don't send all the command??  not sending these special feature above that not all the Mitsubishi models have???

AnalysIR, 2019-06-16:
Please try to relate your query to the IRremote library if possible or even better open a new issue and do that.

You do not generally get signals unrelated to IRremote here, so it may be more productive to try a site like remotecentral.

Alternatively, start by using the examples provided with IRremote and post any isuues arising.

> What happen if I don't send all the command??

Most of the time, it is neccessary to send the full/complete IR signal. _(assuming that is what you were referring to)_

closed 2020-10-17 by ArminJo

FIROGR, 2021-02-13:
> Hi neskweek, I'm sorry but since I've discovered the serial port I dropped my IR projects...

Do you have some example how this serial port works? I would like to control my Mitsubishi air conditioner with arduino GMS shield

llemtt, 2021-02-13:
@FIROGR https://github.com/SwiCago/HeatPump