Arduino-IRremote/Arduino-IRremote #341: ATtiny 85 - Receiving (and possibly sending) does not work

r00li, 2016-07-25:
**Board:** Custom - Based on ATtiny 85
**Library Version:** 2.1.0

I am having some issues getting this library to work on one of my custom boards using the ATtiny85. Below is the code that I am running - based on the example provided:

```c
#include <IRremote.h>
#include <SoftwareSerial.h>

int RECV_PIN = 0;
SoftwareSerial TinySerial(3, 4); // RX, TX

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  TinySerial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    TinySerial.println("RECEIVED");
    TinySerial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
  TinySerial.println("TEST");
  delay(1000);
}
```

When I plug in a serial monitor and run the code I get this output:
> TEST

And that is it. No additional lines, nothing happens if I press a key on a remote. I attached a logic analyzer to the attiny receive pin and I did actually get a proper signal. So the IR receiver is working properly. Of course I tried this without that TEST print as well.

I am running this with the ATtiny core linked in the readme of this library (using 8MHz internal oscillator). 

I've also noticed weird things when sending IR codes as well. Using unmodified IRSendDemo the timing is off. I measured the send pin using a logic analyzer. Instead of 5000ms delay, the delay was only 90ms. And instead of 40ms delay (in the for loop) the delay was 1ms. I've also tried adding a `TinySerial.println("TEST");` at the end of the loop. The output was much faster than expected. Removing the IRremote library and just using a blank loop with `delay(5000); TinySerial.println("TEST");` in it works properly. As soon as I add the IRremote library the timing gets messed up.

middelink, 2016-08-04:
Ok, fly-by commenting here. Did you make sure F_CPU matches your actual cpu speed? It determines the delays made in the _delay function. Guess so, else it would not be fine without the IR lib.

My guess is that both IRremote (timer TINY0 -> timer 0) and the arduino delay function (depends on arduino source code) use the same timer and step on each others foot. Try using _delay_ms(xxx)? it works without timer, but i don't think you can do 5000ms in a single call. Look at its documentation.

r00li, 2016-08-07:
I am currently on vacation for the next 10 days so I can't really test any of this at the moment. But I will get to it when I get home.

If I remember correctly I was already inspecting the F_CPU and it seemed fine. I definitely was playing around it.

The second one could be the problem. I don't think I've ever tried it without delay. I just hope that SoftwareSerial doesn't use it or timer 0.

cklim01, 2016-09-01:
Hi @r00li , did you manage to solve this problem? i experience the same thing when using IRremote and Softwareserial on Attiny85.

r00li, 2016-09-01:
@cklim01 : No, I did not. I've been a bit busy with other stuff, but hopefully I'll have some time during the weekend.

cklim01, 2016-09-02:
@r00li Thanks for update. Hope you can solve this problem.

janbobis, 2016-09-07:
I have made separate tests on the latest 2.2.1 and 2.1.0 release using ATtiny85 using the internal 8Mhz oscillator and i confirm that it is not working. Even if i stripped the unneeded protocols and just use Sony for sending and literally define the SYSCLOCK to 8000000 it is still not working. The only positive side here is that i do get a consistent burst of IR data which is 2Axxxx00.

Doing some tests later and revert back with the results once done.

AnalysIR, 2016-09-07:
Are you confident the AVR fuses of the ATtine85 are configured for Internal Osc @ 8MHz.
Try a 1 second Led blink sketch as a basic test.

If that works it might be worthwhile trying to calibrate the Internal Osc using OSCCAL(?)

_(Of course using an Oscilloscope would be a better approach.)_

janbobis, 2016-09-07:
already did.
fused to Internal OSC @8Mhz
calibrated OSCCAL, tried plenty, from 0x00 to 0x7F.
Would love to test using oscilloscope which i don't have access to currently

ccoenen, 2016-09-12:
I tried the sketch above on a attiny85 / 8Mhz (internal) with a slight OSCCAL calibration added.
Currently, I have IRemote 2.0.1 on my system, and the sketch runs for a little while (and it also seems to receive stuff), but it stops after a short while.

kutysam, 2017-02-11:
Hi, is anyone even able to run IRRAWSENDDEMO?

I can't seem to run it. and it could be because my clock speed etc. may be wrong.
Could anyone who has managed to run it, provide their source code.

My Tiny is running at 8mhz

AnalysIR, 2017-02-14:
@kutysam 

FYI: there is enough information already posted above in this thread for you eliminate at least one of the issues you mentioned. 

Please read the '**Issue Template**' on the main page of this library & also consider opening a new topic with a title that more suits your own issue.

kutysam, 2017-02-14:
I believe the issue isn't solved and I believe my issue is similar to the others above.

I somehow feel, arduino delay is already using timer0 and if i'm not wrong, IRremote is also using TINY0 timer. Thus, a conflict in the 'delay'.

I believe that having a demo that doesn't work properly is an issue.
My delay is not 5seconds as listed in the demo. Its like there is almost no delay.
ATTINY85 fuses are tagged to 8mhz and program is at 8mhz too.

Moving this to timer1 would be better i think, cause it is 'not used by arduino by default.'

AnalysIR, 2017-02-14:
OK

z3t0, 2017-02-15:
> I somehow feel, arduino delay is already using timer0 and if i'm not wrong, IRremote is also using TINY0 timer. Thus, a conflict in the 'delay'.

It's hard to simultaneously support many different boards :)

If you can maybe point us to some reading material about whether delay uses timer0 that would be nice. But in general delay is a bad programming approach for microcontrollers. 

Also, since this issue is related I guess you could continue in this thread but it would be a good idea to create a new issue, explain your info inside the auto generated template and post it as a comment here.

kutysam, 2017-02-17:
Hi z3t0 :)
Unfortunately I can't find any official documentation.
I hope these will assist in showing that timer0 is being used by delay

http://playground.arduino.cc/Main/TimerPWMCheatsheet
"Changing the prescale factor on Timer0 will affect functions
millis(), micros(), delay(),..."

https://electronics.stackexchange.com/questions/60480/any-reason-not-to-use-timer0-on-avr
"For the Arduino, Timer0 is best avoided, as the millis(), delay() and all internal timekeeping in the Arduino libraries use this timer. Changing it's time constant will impact these operations."

https://arduino-info.wikispaces.com/Timers-Arduino
"Timer0: 8bit timer.
In the Arduino world timer0 is been used for the software Sketch timer functions, like __delay()__, __millis()__ and __micros()__. If you change timer0 registers, this may influence the Arduino timer function. So you should know what you are doing."

z3t0, 2017-02-17:
That seems sufficient, I guess any interested party could work on a PR to use another timer

kutysam, 2017-02-18:
I don't know if this will help.
https://gist.github.com/SeeJayDee/caa9b5cc29246df44e45b8e7d1b1cdc5

He seems to be using timer1 for ATTINY85.
I tried it and delay seems to work.

ArminJo, 2020-10-12:
IRremote and [IRMP ](https://github.com/ukw100/IRMP) now supports ATtinies and Digispark.

closed 2020-10-12 by ArminJo