Arduino-IRremote/Arduino-IRremote #342: irrecv.decode works fine, then starts returning random values after using FastLED functions

CarrotShapedTurtle, 2016-07-30:
Hi all, 
I'm new to the world of Arduino, so I apologize if the solution should be obvious.

I'm trying to use the IRRemote and FastLED libraries together so I can control LED animation with unused buttons on my TV remote.
An adapted version of the irrecvdemo code sample works fine all the way until I actually want to change FastLED parameters within loop() -- then the decoder only returns seemingly random HEX values regardless of what button I press on the remote (i.e. 3 different HEX values when pressing the same button 3x).
I've narrowed it down to this reduced [code sample](https://gist.github.com/CarrotShapedTurtle/ec697d04b0d8f1db11238fa60d727961), which does the following: 

> SETUP: 
> initialises the IR receiver
> sets FastLED parameters (to show that the problem doesn't just come from using both libraries in the same sketch)
> 
> LOOP:
> decodes IR signals via irrecv.decode
> prints received HEX values to serial for debugging
> sets the mode variable when one of four known button codes is received + prints the button colour to serial
> switches between different modes, all of which are deactivated for debugging, except the blue button which simply clears LED output (the switch here is necessary since modes 1 and 2 can switch back and forth from within the used functions)

Here's some example output from Serial Monitor: 
> FEAA55
> red
> FE6A95
> green
> FEEA15
> yellow
> FEEA15
> yellow
> FE28D7
> blue
> 5104C00E
> FAFAC709
> FC9C4D61
> 7ADF5C26
> A447D8D3
> 5A074F9F
> AC6A85DE

As you can see, the decoder works fine until the blue button is pressed, which runs FastLED.clear(true); and causes subsequent random output (i pressed the yellow button repeatedly).

How can I avoid this problem and keep reading actual IR codes?

AnalysIR, 2016-07-30:
FastLED is probably using the same timer which results in a conflict. (guess)

Possible options
1. reconfigure IRremote to use a different timer - in file IRremoteInt.h
2. re-initialise the IRremote timer by calling `irrecv.enableIRIn(); // Start the receiver` after you set the colour.

I think option  would be the cleanest method.

jpoppe, 2016-07-30:
@CarrotShapedTurtle: https://github.com/FastLED/FastLED/issues/198

Seems to work pretty OK for me ;)

CarrotShapedTurtle, 2016-07-31:
Thanks to both of you for your help! 
@AnalysIR, I had tried switching IRremoteInt to Timer 1, no effect.
@jpoppe's link did the trick -- but rather than halting the entire loop, which would freeze the Bambilight output I am processing in one of the modes, I'll just skip IR decoding whenever its in use, using 
`if(irrecv.isIdle()) { ... }`

Given that the loop is processed at high frequency, this still catches almost all remote button signals correctly.

closed 2020-10-17 by ArminJo