Arduino-IRremote/Arduino-IRremote #344: IR LED weak signal

ArGaspar, 2016-08-02:
**Board:** ARDUINO UNO / MEGA

Hello, 
I don´t know why but the IR LED is transmitting a really weak signal, i can check it with the phone camera because if i just put any pin high the LED gets really shiny but when im running the library and sending signals it dosnt get even a half as it would do with just a pin high.
Sorry for my english.
Many Thanks.
Gaspar

AnalysIR, 2016-08-02:
..because an IR signal uses a carrier and is pulsed & on average is only on for 20% or less of the time.

So it all makes sense that it is dimmer.

However, you should not connect the IR LED direct to the GPIO pin without a current limiting resistor.

see:
[https://www.analysir.com/blog/2014/10/03/driving-infrared-led-directly-arduino-pin/](https://www.analysir.com/blog/2014/10/03/driving-infrared-led-directly-arduino-pin/)

and 
[https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/](https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/)

or search google for IR led driver circuit.

bobby1sc, 2016-08-07:
i think you should try IR LED from Vishay InfraRed Emitter TSAL6400...
i already tried it.. if you use normal IR LED you need about 2 IR LED.. but with IR LED TSAL6400 only need 1 IR LED for long range .. ^_^

ArGaspar, 2016-08-15:
Many thanks guys and sorry for the late answer.
@AnalysIR , I read your article and its awesome, I just have one doubt with constant current circuit, how much range did you reach?, Im trying to make a Remote control so I need to get same range as a "normal remote".

AnalysIR, 2016-08-15:
If the answer is not addressed in the article......you should carry out some tests and measure the results - to eliminate your doubts :)

> **_Quote_**: _you can adjust the resistors to match the current you want._

(resistors = R1 **or** R4)

ArGaspar, 2016-08-23:
Hello @AnalysIR , 
Finaly I have done some test, and definitely im doing doing something wrong but dont know what, I have 100mA Emmiter, so following your intructions, 0,5v/0,1A = 5 ohm, so I put 5ohm resistor but im getting  50 ma of current dont know why because I have test V and its 0.5v, if i just connect directly  i get 100mA the weird thing is that if I change R2/3 for 3.3ohms i get 300ma directly and 100 ma with the resistor so then its working BUT im getting around 58ma of the arduino pin, all this results are with blink example if I try to send an IR signal i cant get more than 5ma with the resistor,
I can only imagine that the transistor is not suitable for the job because in your article you dont talk about changing R2/3 to get more mA,! and as far as i know changing the base mA  it should not change emmisor ma right? im using this transistor 2n3904.
https://www.sparkfun.com/datasheets/Components/2N3904.pdf

Many thanks for your help!

AnalysIR, 2016-08-23:
your transistor has a max rated current of 200mA, which may be a limiting actor?.

You havent mentioned how you are measuring current...so its not clear to me. IS your Vcc @ 5V or 3V3?
Also note that when sending IR the 'average' current will be less because of the carrier and the mark/space ratios.

What is the 100mA emitter...the IR Led?

You said you test V @ 0.5V, but didn't reference where the measurement was taken and other important info.

Maybe update your post and make use of the measuremnt points/labels in the original circuit diagram.

ArGaspar, 2016-08-23:
Hi @AnalysIR thanks again,
I have updated the post the only thing that I have forget to mention is that instead of the IR signal im just turning high pin 13 on arduino. and 0.5V is always measured on pins of R1.

AnalysIR, 2016-08-23:
>  and 0.5V is always measured on pins of R1. 

Yes that is the idea, you should be getting around that voltage on the emitter of the transistor. We exploit this to drive a constant current based on the value of R1.

This voltage will be more or less contant (when active) & you then get to control the current by setting the value of the resistor R1.

So if Ve (emitter) is 0.5V and R1 is 5r =>Ir1 = 0.5/5 = 0.1 = 100mA
to get 200mA set R1=2R5.

NB: Remember the absolute max current limit for your transistor is 200mA(?)...so maybe another Transistor would be better if you need higher currents. (>100mA)

Are you measuring current with a DMM or a simulator? 

You should not have a short for R1.

ArGaspar, 2016-08-23:
Im measuring with a DMM, I just need 100mA so it should be ok.
I understand that changing R1  can calibrate current of emitter but Its supposed to change current on the emitter of the transistor if I change R2? because that is happening as you can see on diagrams.

AnalysIR, 2016-08-23:
>  Its supposed to change current on the emitter of the transistor if I change R2?

My guess is that by reducing R3 significantly to 3R3, you are likely increasing Vb, which in turn increase Ve which results in more current thru R1.  (Ve is the voltage at the emiiter/top of R1)
_BTW: This (3R3) could have caused damage to your Diodes, transistor and/or Arduino!_

However, that is way outside the scope of the circuit & if you want to understand more just play around with the options then test and measure until you understand what is happening.

> PD: if i connect the DMM directly to emitter and 0V im getting 0.5V when the arduino is low but when arduino is high i get more that 1V. wtf?

Consider posting on the Arduino or similar forum. When the Arduino is outputting LOW, Ve should be 0V. If not check your ground wiring or perhaps you may have damaged one or more of the components or MCU.

> Im measuring with a DMM, I just need 100mA so it should be ok.

A good time to close this issue as you have a working solution.

ArGaspar, 2016-08-23:
Right, thanks again.

closed 2016-08-23 by ArGaspar