Arduino-IRremote/Arduino-IRremote #345: Recieve and send i same sketch

mjelund, 2016-08-08:
Hi, please advise me.
I'm trying to do a device-in-the-middle to replace a lost tv remote, converting a IR input from my small Apple remote, to send a specific IR code to my LG television. Decoding these signals was easy. I can use the Apple input and it is working. I can send LG commands and it is also working. BUT; when adding these into same sketch, it halts! Even if i remove the sending IR LED (to not interfere with the reciever and cause a endless loop! 

**Board:** ARDUINO NANO , IDE 1.6.10
**Library Version:** 2.0.1
**Protocol:** NEC (if any)

**Code Block:**
long unsigned int input;
#include <IRremote.h> 
....
input = results->value, HEX;
    if (input == 0x77E1D014){  //Apple small remote VOL+
      b = b+10;  //Control brighness of a small LED to verify it works
      irsend.sendNEC(0x20DF40BF, 32); //LG TV VOL+
    }
.....

yesyesuk, 2016-08-08:
I seem to remember that after sending you need to enable receiving again with irrecv.enableIRIn(); But I may remember this wrong...

mjelund, 2016-08-08:
Hello yesyesuk
You have a brilliant memory, and I thank you a lot! That did the trick. I thought the irrecv.resume(); in the loop (taken from one of the examples) was sufficient. 
Now I have the nicest small IR Remote from Apple, controlling my LG TV. As i only use Chromecast, the volume+/-, Power and input (for the Playstation) is more than enough for me!
Thanks alot!
Br
M

yesyesuk, 2016-08-08:
I'm glad I could help for once.. :D

Adminius, 2017-02-10:
@mjelund , can you provide you sketch? thx!

AnalysIR, 2017-02-10:
@Adminius 

You can easily check the examples provided with the library.

@z3t0  I suspect this issue should have been closed before??

closed 2017-02-10 by z3t0

farziney, 2018-01-10:
I want to receive remote TV signal by Arduino and send it back to Ardino to tv the program code show me?

AnalysIR, 2018-01-10:
@farziney I suggest you open a new issue, as this one is closed.

It would also be a good idea to provide details of the efforts you have made up until now. 
The example sketches provided with the library is always a good place to start.