Arduino-IRremote/Arduino-IRremote #346: RedBearLab BLE Nano

llemtt, 2016-08-10:
Has anyone tried to support the BLE Nano?

It's based on Nordic nRF51822 (ARM Cortex-M0).

z3t0, 2016-08-23:
Not that i know of, you may want to look into the code for teensy as I believe one of them (LC?) uses a Cortex M0. And then with a little bit of expertise it should be fairly trivial to add support for the requested micro controller.

BandonRandon, 2016-11-28:
seems #318 is a duplicate (with more comments) regarding M0 support.

llemtt, 2016-11-28:
As written in #318 M0 is the name of the "core" (think CPU) of the MCU/SOC, all the peripherals (TIMERS..) are different from family to family. AVRs, STM32s, nRF51s,... each one requires a different implementation.

Anyway I managed to have some working code on the BLE Nano but since then I discovered my HVAC has also a serial port..

ArminJo, 2020-10-12:
[IRMP ](https://github.com/ukw100/IRMP)has mbed (Nano 33 BLE) support.

closed 2020-10-12 by ArminJo