Arduino-IRremote/Arduino-IRremote #348: Interrupt support

rkoptev, 2016-08-11:
Hi!

rkoptev, 2016-08-23:
@z3t0 , yes

z3t0, 2016-08-23:
OK, that should be fairly simple, except it would still wait and decode. So basically the best you can achieve is:
- signal received
- callback called: decodes signal and does something

But you will still have the same issue as decoding/ receiving takes time and the arduino can't multitask.

Thoughts?

rkoptev, 2016-08-23:
Yes, approximately. I say that the time of execution of code in the loop can be more than 100ms, which certainly will not give a good responsiveness to the remote. I would like to be able to executable code and perform a certain task when buttons on the remote pushed. This logic is very similar to a standard interrupt for Arduino that we can attach to some pins with *attachInterrupt()* function.

z3t0, 2016-08-24:
Could you please share your code in A gist and link it here just for clarification. Thanks.

mikrob24RUS, 2017-04-12:
Hi! I have the same problem. I can't solve it for a long time...
Sometimes it turns out that the IR signal is either lost, or contains errors.
`#include <IRremote.h>

int IRPin = 12; // порт инфракрасного приёмника
IRrecv irrecv(IRPin);
decode_results results;

unsigned long micros_sp = 0;
volatile float sp; // мгновенная скорость
volatile float spAVG = 0; // средняя скорость

float CCSpeedDelta = 0.25; // отклонение от скорости КК для реагирования
unsigned long serviceTime; // временная метка нажатия сервисной кнопки (power off)
int serviceMode = 0; // вариант сервисного режима

int sensorBrPin = 4; // сигнал с педали тормоза
int hallMinPin = 5; // Датчик Холла минимум
int hallMaxPin = 6; // Датчик Холла максимум
int motorPinF = 7; // управление двигателем 1 канал
int motorPinR = 8; // управление двигателем 2 канал
int enabledMotor = 9; // разрешающий сигнал драйверу мотора
int ledHallMaxPin = 10; // светодиод CC_FRW
int buzzerPin = 11; // динамик
int ledHallMinPin = 13; // светодиод CC_REV
int ledCCOnPin = 14; // светодиод CC_On

int acceleratorPin = 5; // сигнал с датчика БДЗ (аналоговый)

boolean ccActive = 0; // активность КК
int timePeriod = 1000;
unsigned long timeMotorStop = 0; // время остановки мотора
int SpeedFail = 0; // счётчик нарушений скорости для безопасности
int SpeedReduction = 0; // счётчик кол-ва последовательного снижения скорости
//int SpeedReduction2 = 0; // счётчик кол-ва последовательного снижения скорости
int SpeedIncrease = 0; // счётчик кол-ва последовательного увеличения скорости
float CCSpeed = 0; // скорость КК
const int CCSpeedMin = 5; // 
const int CCSpeedMax = 5; //
int CurrDZ;
int DZPosMin; // минимальное положение ДЗ ~95
int DZPosMax; // максимальное положение ДЗ ~750
float OldSpeed = 0; // скорость прошлого замера
float SpeedUp = 0; // ускорение
//float SpeedOld = 0; // старая скорость в расчёте ускорения
float CalcSpeedUp = 0; // расчётное ускорение
int CSDK = 15; // коэффициент резкости расчётного ускорения (рекомендуется от 15 до 30, чем больше тем резче разгон)

unsigned long timeOldSpeed = 0; // время замера прошлой скорости

void myTone(int hertz,int longs) {
  unsigned long startTime = millis() + longs;
  float period = 1000000 / hertz;
  while (millis() <= startTime)
  {
    digitalWrite(buzzerPin, HIGH);
    delayMicroseconds(period / 2);
    digitalWrite(buzzerPin, LOW);
    delayMicroseconds(period / 2);
  }
}

void indicator(float val) {
  myTone(1500, 100);
  digitalWrite(ledHallMinPin, LOW);
  digitalWrite(ledCCOnPin, LOW);
  digitalWrite(ledHallMaxPin, LOW);
  delay(2000);
  int cel;

[... 519 lines cut ...]
      }// *снизилась
      if (spAVG > OldSpeed) {// увеличилась
        if (spAVG < CCSpeed) {
          SpeedIncrease = 1;
        }
        else
        {
          SpeedIncrease = int(spAVG - CCSpeed) + 1;
        }
        int rev;
        rev = 6;
//          rev = int(spAVG * abs(SpeedUp) / 4);
        if (rev < 6 ) { rev = 6; }
        rev = rev  * SpeedIncrease;
        setDZ(CurrDZ - rev);
      }// *увеличилась     
  
//        delay(500);
      // запоминаем скорость как старую
      OldSpeed = ReadSP_AVG();
      timeOldSpeed = millis();
      }// *скорость отклонилась на 1 км/ч
    // *скорость в пределах нормы              
  }// *проверяем включен ли КК и настало ли время для след.прохода
}`

z3t0, 2017-04-12:
Please share the relevant piece of code and create a new issue, follow the template on the new issue.

mikrob24RUS, 2017-04-12:
I do not quite understand what You mean. Sorry I English is not know use a translator. I wrote my comments on the link - https://gist.github.com/z3t0/e3a110138b072467b9a454293035c8a9/5d968e65d53bd5109d997a8968186085817fd99b#file-a-ino-L336.

z3t0, 2017-04-12:
https://github.com/z3t0/Arduino-IRremote/issues/new

mikrob24RUS, 2017-05-15:
Thank you very much for participating in my issue! It turned out that the recognition signal prevent electrical interference affecting the cable going to the IR receiver. Replaced the cable with a shielded and all became good to recognize!
Thanks again!

closed 2020-10-17 by ArminJo