Arduino-IRremote/Arduino-IRremote #354: Chuwi ILIFE V7 Protocol Creation

Oitzu, 2016-08-27:
**Board:** ARDUINO NANO
**Library Version:** 2.1.0
**Protocol:** Creation of Chuwi ILIFE V7

I'm a little bit lost in protocol creation for die Chuwi ILIFE V7 (robot vacuum cleaner).

I do not have a IR-Receiver at hand but a IR-Photosensitive diode and an oscilloscope.
Connecting the oscilloscope to the circuit and pressing the up button on the remote gives me following picture:
[image]

I tried the IRanalyze code on it but got weird results. Do i understand it right that a IR-Receiver normally would invert the signal?

bengtmartensson, 2016-08-27:
Do you have the numbers too?

> a IR-Receiver normally would invert the signal?

"normally" in the sense of "people are normally right-handed"

Oitzu, 2016-08-27:
Well i have seen that most protocols are defining: HDR_MARK, HDR_SPACE, BIT_MARK, ONE_SPACE and ZERO_SPACE but have not idea which timig between which 2 points i should exactly take for those. 

Well, if a receiver would invert the signal: 
HDR_SPACE: The HIGH-Signal timing at the beginning
HDR_MARK: The LOW-Signal timing after the HIGH at the beginning
BIT_MARK: The LOW timing between of 2 of the HIGH signals
ONE_SPACE: The duration of one high
ZERO_SPACE: ???
(LOW and HIGH referenced to my osci picture)

IRanalyze gave me following:
```
Samples=10
Decoded Unknown: Value:0 (0 bits)
Raw samples(68): Gap:2621
  Head: m4250  s750
0:m350 s740	1:m365 s740		 2:m365 s740	3:m365 s730		 
4:m370 s740	5:m365 s735		 6:m1480 s725	7:m400 s735		 
8:m1480 s735	9:m365 s735		 10:m1500 s735	11:m365 s730		 
12:m1485 s735	13:m370 s735		 14:m1495 s735	15:m380 s730		 

16:m370 s735	17:m1495 s725		 18:m375 s730	19:m1490 s740		 
20:m360 s735	21:m1495 s740		 22:m360 s740	23:m1500 s720		 
24:m1505 s740	25:m360 s730		 26:m1495 s730	27:m370 s730		 
28:m1495 s735	29:m370 s730		 30:m1490 s730	31:m395 s725		 

32:m2966
Extent=57861
Mark  min:350	 max:2966
Space min:720	 max:740

	 #	Low	High	Avg.
Mark	33	350	2966	890
Space	32	720	740	733

Marks		 #	Low	High	Avg.
Above Avg	14	1480	2966	1597
Belows Avg.	19	350	400	369

Spaces		 #	Low	High	Avg.
Above Avg.	19	735	740	737
Below Avg.	13	720	730	728

Ratios		Below	Above
Head Mark	11.52	2.66
Head Space	1.03	1.02

Mark Above/Below=	4.33
Space Above/Below=	1.01
```
But i think its wrong because it don't get the inverted signal it would normally expect?

Oitzu, 2016-08-28:
Looking again at the code this would be the right things to measure?:
[image]

Oitzu, 2016-08-28:
If i measure that on the osci i get:
HDR_MARK: 9000
HDR_SPACE: 4500
MARK: 500
ZERO_SPACE: 600
ONE_SPACE: 1700

Carrier Frequency 38kHz

Does this sound plausible?

AnalysIR, 2016-08-28:
FYI: looks like standard NEC 32 bits

HDR_MARK: 9000
HDR_SPACE: 4500
MARK: 560
ZERO_SPACE: 560
ONE_SPACE: 1690

You should be able to use the sendNEC function - but first you need to 'manually' record the Hex value.

If you have an IR LED you could try this approach with an Arduino.
[https://www.analysir.com/blog/2014/05/28/poor-makers-infrared-receiver-2/](https://www.analysir.com/blog/2014/05/28/poor-makers-infrared-receiver-2/)
...as a bonus the sketch output also provides carrier frequency

Oitzu, 2016-08-28:
@AnalysIR in the meantime i created a own protocol called "ILIFE" with the measured timing and was successful in sending the the robot a "CLEAN" (0x02aa22dd) Signal. But you are right, it looks near enough (in sense of timing) to the NEC 32 bit protocol, to also work with that. 

I will also try the NEC protocol, to not create any redundant protocols.

Is there any place to document the compatibility and the found IR-codes, for other users?

AnalysIR, 2016-08-28:
> Is there any place to document the compatibility and the found IR-codes, for other users?

You can make changes to the code & docs (or add anew file) using GitHub and submit a pull request for inclusion into the library.

The NEC IR protocol is used by many manufacturers...not just by NEC. I think its a safe bet that it is NEC (for the purposes of IRremote).

Oitzu, 2016-08-28:
Well.. yeah, i see where i would place new protocols but not where to place codes or the compatibility between hardware and protocol. 
I leave the codes, i gathered so far with the osci, here for the moment, till i found a better place:
```
UP 0x02aa55aa
CLEAN 0x02aa22dd
DOWN/MAX 0x02aa6699
LEFT 0x02aa33cc
RIGHT 0x02aa44bb
SPOT 0x02aa7788
HOME 0x02aa2277
EDGE 0x02aa9966
```
These are not complete, the timer settings are missing. I need a better working receiver to acquire these.

frankja2, 2017-04-08:
@Oitzu how to implement those codes to lirc?

Oitzu, 2017-04-08:
@frankja2 i would take a existing config of a NEC device (by example this: http://lirc.sourceforge.net/remotes/nec/RC-1077E) as base and just change the name and codes.

frankja2, 2017-04-08:
@Oitzu wow :+1: it's working! I've spent few hours of useless clicking remote buttons from Ilife A4 and lirc couldn't read it. I don't know why.

Oitzu, 2017-04-09:
@frankja2 great that it works.
Have a look at the picture of comment https://github.com/z3t0/Arduino-IRremote/issues/354#issuecomment-242971847 .
You see the osci output of the "up" signal.
Spikes in the signal represent pulses. A short space between pulses represents a logical 0 and a long space between to pulses represents a logical 1.
The rest is just reading 4 bits at a time and convert them to hex.
As example: Binary: 1010 Decimal: 10 Hexadecimal: A

If this is still unclear have a look here https://en.wikipedia.org/wiki/Hexadecimal#Written_representation
on the right side.

closed 2020-10-17 by ArminJo

ArminJo, 2022-03-17:
Just guess the code 😀

Oitzu, 2022-03-18:
Well i never actually used the codes. But i tested them back then, and they worked.
Maybe the "Home Button" has a slightly different code for the A4?
Maybe guessing the code is a good idea, there should be only 256 valid combinations, that could be quickly tested with a small program.
Im now using a tuya compatible ir blaster where i only learned the clean button to.

About the other button codes:
I assume with other buttons you talking about the clock and timer configuration?
They dont work on single button presses but the remote sends out the whole configuration on each press.
I never bothered to work out how that works.