Arduino-IRremote/Arduino-IRremote #355: Intel based Arduino support

MeNicolas, 2016-08-30:
**Board:** ARDUINO 101
**Library Version:** 2.1.0
**Protocol:** any

Will Arduino 101/Edison/etc be supported by this library since they haven't AVR support ? 

Thanks a lot

AnalysIR, 2016-09-01:
IRremote can support more than AVR platforms eg Teensy.

However, adding  new platforms like the ones you mentioned, depends on someone with the Board developing, testing & contributing the code required. Usually it is not that difficult for an experienced coder to do.

Another approach would be to contact the manufacturer or supplier of the board directly to request that this be done. For example, the Teensy guys are very proactive in this regard.

AnalysIR, 2016-09-01:
I forgot to mention that there could be forks of this IRremote library which could support these platforms. However, it would still require someone to contribute those forks back to the main library.

So maybe a search might pop-up existing support in a forked library.

z3t0, 2017-08-22:
Closing due to inactivity.

closed 2017-08-22 by z3t0