Arduino-IRremote/Arduino-IRremote #356: Issue sending NEC codes, but raw works fine

robbiet480, 2016-09-01:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** NEC (maybe?)

Hey,

So I know my setup is working since sending raw data controls my Seville Classics fan. The IRRecvDemo says that it's getting NEC codes, and some existing code on Github makes me think it's NEC, but whenever I send NEC out the fan won't respond. If I send the raw data it works fine. I posted about this issue in the [IRemoteESP8266 repo](https://github.com/markszabo/IRremoteESP8266/issues/42) but I am also posting here because I have the same issue when I try this library (which the ESP8266 one is based on) on my Uno.

Wondering if it's a library issue or a protocol issue. I'm thinking it might be library because it's such a long code to send (according to the timing?). Please note: IR & NEC newbie speaking here.

Here's example output from the receiver:

```
Encoding  : NEC
Code      : 10203040 (32 bits)
Timing[131]: 
     +9400, -4800     + 600, - 500     + 600, - 500     + 600, - 550
     + 600, -1600     + 600, - 500     + 600, - 550     + 600, - 500
     + 550, - 550     + 600, - 550     + 550, - 550     + 600, -1600
     + 600, - 500     + 600, - 550     + 550, - 550     + 600, - 500
     + 600, - 550     + 600, - 500     + 600, - 550     + 550, -1650
     + 600, -1600     + 600, - 500     + 550, - 550     + 600, - 550
     + 600, - 500     + 600, - 500     + 600, -1650     + 550, - 550
     + 600, - 500     + 600, - 550     + 600, - 500     + 600, - 550
     + 550, - 550     + 600, - 500     + 600, -1600     + 600, - 550
     + 550, -1650     + 600, - 500     + 600, - 550     + 550, - 550
     + 600, - 500     + 600, - 550     + 600, - 500     + 550, - 550
     + 600, - 550     + 600, - 500     + 550, - 550     + 600, - 550
     + 600, - 500     + 600, - 500     + 600, -1600     + 600, -1650
     + 550, - 550     + 600, - 500     + 600, - 550     + 600, - 500
     + 600, - 500     + 600, - 550     + 550, -1650     + 550, -1650
     + 550, -1650     + 550, - 550     + 600, - 550     + 600, - 500
     + 600, - 500     + 600
unsigned int  rawData[131] = {9400,4800, 600,500, 600,500, 600,550, 600,1600, 600,500, 600,550, 600,500, 550,550, 600,550, 550,550, 600,1600, 600,500, 600,550, 550,550, 600,500, 600,550, 600,500, 600,550, 550,1650, 600,1600, 600,500, 550,550, 600,550, 600,500, 600,500, 600,1650, 550,550, 600,500, 600,550, 600,500, 600,550, 550,550, 600,500, 600,1600, 600,550, 550,1650, 600,500, 600,550, 550,550, 600,500, 600,550, 600,500, 550,550, 600,550, 600,500, 550,550, 600,550, 600,500, 600,500, 600,1600, 600,1650, 550,550, 600,500, 600,550, 600,500, 600,500, 600,550, 550,1650, 550,1650, 550,1650, 550,550, 600,550, 600,500, 600,500, 600};  // NEC 10203040
unsigned int  data = 0x10203040;
```

**Code Block:**
```c++
irsend.sendNEC(0x10203040, 32);
```

Can anyone help?

**checklist:**
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided
- [] The title of the issue is helpful and relevant 

The above is a short template allowing you to make detailed issues!

AnalysIR, 2016-09-01:
Standard NEC is 32 bits...your signal is much longer than 32 bits. (64 bits?)

you should also change the timings to (9000 / 4500 / 560 / 1690) for better performance

Irremote only sends NEC (sendNEC) for up to 32 bits.

If you store each hex byte in an array, you could hack the existing sendNEC code to work with this signal...otherwise sendRAW is the way to go.

AnalysIR, 2016-09-01:
PS: you will find a similar approach here...
[https://www.analysir.com/blog/2015/09/01/simple-infrared-pwm-on-arduino-part-3-hex-ir-signals/](https://www.analysir.com/blog/2015/09/01/simple-infrared-pwm-on-arduino-part-3-hex-ir-signals/)

robbiet480, 2016-09-01:
@AnalysIR Thanks for that! I was looking at your blog posts yesterday but was unsure if I was on the right path or not.

So it looks like AnalysIR will do everything I need to get this done and be able to send hex. Just gotta find a Windows computer first :(

AnalysIR, 2016-09-02:
or a VM :).....but yes Windows only!

..runs fine on VirtualBox

closed 2020-10-17 by ArminJo