Arduino-IRremote/Arduino-IRremote #358: Consult to how to decode sharp TV IRremote signals and send

liang-jia, 2016-09-06:
- 

 ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:**Sharp (if any)

HI,
  I met a problem when I make use of a IR learning device to decode Sharp TV IRremote by our own software  and then send signal to control the Sharp TV. Because the software is available  when it decode and send other chinese  electric appliance IRremote ,so I don't know whether the sharp TV IRremote 'signal has some special fomation  ,so that I must use some special method for the decoding and sending software .
    On your web ,I have seen your  irRecv.cpp   and irSend.cpp , can you tell me thar the two cpp are availble for all IRremote signal?
    And,i have seen the ir_Sharp.cpp, Is it the special method for sharp?
     But in ‘ir_sharp.cpp’,I have only seen the SEND_SHARP,not seen the decodeSharp,can you tell me whether the cpp is enough for the Sharp Irremote ?
 

Thanks

z3t0, 2016-09-24:
If you look inside the mentioned files you will see that we have provided protocol definitions for certain popular remotes such as Sharp, Sony and NEC. 

But yes each vendor uses their own encoding for the signal.

As of now, we have not included a routine for receiving SHARP signals, the define is there, but the method is not implemented. You should still be able to achieve similar results by trying to understand how your remote sends signals, and then replicate it using sendRaw or adding your own protocol definition to the library.

buestad, 2016-11-12:
Hi. I am working on the Sharp protocol as well. I have written the decodeSharp function but it is not 100% finished yet but works OK for my project. I can send you my ir_sharp.cpp if I can figure out how

Adam-Morris, 2017-01-01:
Hi @buestad, are you able to paste your ir_sharp.cpp please?

buestad, 2017-01-01:
It's new years eve man! Coding can wait :) 

But if you can't, go to pull requests. There is a link to my fork of the project. There is also an other user who made an other version.

z3t0, 2017-08-22:
Closing due to inactivity.

closed 2017-08-22 by z3t0