Arduino-IRremote/Arduino-IRremote #359: Difference of pressing IRremote button for a long time or a shor time

liang-jia, 2016-09-06:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Sony (if any)

HI,
 I have a problem  that the learning device sometimes can't succede learning when I press the IRremote button for a long time or a shor time ,can you tell me that it's diffrent for pressing long time or short ? Has the software some method to enhance the learning succeding rate for IRremote?

Thanks

z3t0, 2016-09-24:
I am not certain about all remotes but I know for my generic remote which is recognized as NEC, sends FFFF for repeat keys. 

IRremote doesn't necessarily handle this as different protocols may use different methods to signal long press (I'm not too sure), but you could easily program this as a part of your sketch.

I would recommend loading up IRRecvDumpV2 and take a look at what your remote does when holding down buttons

z3t0, 2016-09-24:
For example, the following is how the NEC works.
https://i.imgur.com/TcyNC24.png

The only way to determine how long it was pressed is to count up how many repeat signals are sent by the remote.

bengtmartensson, 2017-08-22:
IrScrutinizer can be used; it identifies a repeating portion of an ir sequence, and counts the number of occurances (on the plot: intro: red, repeat: blue, ending (when present): green).

closed 2017-08-22 by z3t0