Arduino-IRremote/Arduino-IRremote #360: IR Remote & RC Switch libraries incompatibility

jlgoasis, 2016-09-06:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** any

I'm working in my new project were I'll like to use my LG TV IR remote control as controller to remote some RF433MHz 220V light switches like that (http://es.aliexpress.com/store/product/High-Quality-EU-3-Pack-Wireless-Remote-Control-Power-Outlet-Light-Switch-Plug-Socket-OD-S/1333800_32566929234.html?spm=2114.04020208.3.2.tSkYTi&ws_ab_test=searchweb201556_0,searchweb201602_1_10057_10065_10056_10055_10037_10054_10044_10043_10033_10059_10058_10032_10017_107_10041_10042_10060_10061_10052_414_10062_10053_10050_10051,searchweb201603_2&btsid=a03c2c6b-7626-4ada-8ab8-b4bc004c36bb).
 
For this project I have an ATMega328P (or Arduino Uno R3), one IR Sensor VS1838B and one 433MHz transmitter module (http://www.aliexpress.com/item/1set-RF-module-433-Mhz-superheterodyne-receiver-and-transmitter-Support-ASK-OOK-small-size-low-power/32571703475.html?spm=2114.13010608.0.91.R3ntQ6)
 
I’m using the IR remote library plus the standard and well know RC switch library (https://github.com/sui77/rc-switch). 

Both library are working fine separately, IR remote as receiver and RC switch as transmitter, but when use those together doesn't work and the RC switch library doesn't send anything to my light switches devices, probably by some timer issue than I don't understand.
 
Find below a simple example. I think the problem is when is invoked **irrecv.enableIRIn()**; (in void setup())

Thanks

**********************************
#include <IRremote.h>
#include <RCSwitch.h>

#define RECV_PIN  11     //for IR Remote
#define TRAN_PIN  10   //for RC Switch

RCSwitch mySwitch = RCSwitch();
IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(9600);

  mySwitch.enableTransmit(TRAN_PIN);  // Transmitter is connected to Arduino Pin #10

  // Just to test RF344MHz is working fine
  mySwitch.send(16762196, 24);    // CH A ON
  delay(2000);
  mySwitch.send(16762193, 24);    // CH A OFF

  irrecv.enableIRIn(); // Start the receiver
}

void loop() {
  if (irrecv.decode(&results))  {
    switch (results.value) {
      case 0x20DF4EB1:              // CH A ON
        Serial.println("CH A ON");
        mySwitch.send(16762196, 24);
        break;

      case 0x20DF8E71:              // CH A OFF
        Serial.println("CH B ON");
        mySwitch.send(16762193, 24);
        break;

      case 0x20DFC639:              // CH B ON
        Serial.println("CH B ON");
        mySwitch.send(16765268, 24);
        break;

      case 0x20DF8679:              // CH B ON
        Serial.println("CH B OFF");
        mySwitch.send(16765265, 24);
        break;

      default:
        Serial.println("NG.");
    }
    delay(50);
    irrecv.resume(); // recive siguiente valor
  }
}

z3t0, 2016-09-24:
I would assume both libraries are trying to use the same timers and that is causing a conflict

jlgoasis, 2016-09-24:
Thanks. I Will check it with timer2
BR
Jacinto

z3t0, 2016-09-24:
Please refer to this line of code to change the timer and see if that helps :smile"
[Timer](https://github.com/z3t0/Arduino-IRremote/blob/master/IRremoteInt.h#L223)

tvixen, 2017-01-01:
So did it help ?

z3t0, 2017-01-14:
Just a note that the link will no longer work due to code changes

smartboy82, 2017-01-26:
I have the same problem. If someone resolve this? How can I control RF switch with IR remote.

AnalysIR, 2017-01-26:
> I have the same problem. If someone resolve this?

If you have the same problem, why not try the suggestions above and post your own results.

> How can I control RF switch with IR remote.

Many different ways, depending on the equipment & resorces you are using.

smartboy82, 2017-01-26:
I don't understand what must do. It is my first project.
IR receiver work alone perfect and RF transmitter too but together doesn't work

`#include <IRremote.h>
#include <RCSwitch.h>

int RECV_PIN = 11;
IRrecv irrecv(RECV_PIN);
decode_results results;
RCSwitch mySwitch = RCSwitch();

void setup() {

  Serial.begin(9600);
 irrecv.enableIRIn(); // Start the IR receiver
  
  // Transmitter is connected to Arduino Pin #10  
 mySwitch.enableTransmit(10);

  // Optional set pulse length.
 mySwitch.setPulseLength(498);
  
  // Optional set protocol (default is 1, will work for most outlets)
 mySwitch.setProtocol(5);
  
  // Optional set number of transmission repetitions.
  // mySwitch.setRepeatTransmit(15);
 
 
}

void loop() {
 if (irrecv.decode(&results)) {
   switch(results.value){
     case 3672802284:
       mySwitch.send (16302791,24);
       break;
     case 732942060:
       mySwitch.send (16660487,24);
       break;
   
   }
 Serial.println(results.value, DEC);
 irrecv.resume(); // Receive the next value
 }
  
  
}`

tvixen, 2017-01-27:
Try look at my code https://github.com/tvixen/Wemo-for-Alexa
I'm using the IR library and ESP8266 ... you should try it, it works like a charm.

smartboy82, 2017-01-27:
I can't see that you use 433MHz transmitter on your project...anyway thank you for reply it is good idea for upgrade. At this moment I just need IR to RF control

AnalysIR, 2017-01-31:
It is possible that both libraries are using the same timer.

Try changing the timer used with IRremote....in the file boarddefs.h or IRremoteboarddefs.h
you can comment/uncomment as appropriate.

AnalysIR, 2017-01-31:
It may be that both libraries are usign the same timer.

Try changing the timer used by IRremote in boaddefs.h (or IRremoteboarddefs.h) by commenting/uncommenting the timer used as appropriate.

smartboy82, 2017-01-31:
I don't have that file...

AnalysIR, 2017-01-31:
use the latest version of the library

[https://github.com/z3t0/Arduino-IRremote/blob/master/boarddefs.h](https://github.com/z3t0/Arduino-IRremote/blob/master/boarddefs.h)

smartboy82, 2017-01-31:
which timer do I must comment?

AnalysIR, 2017-02-01:
Sorry, I can't say as it is different for different platforms.

Please read the README, where you will see the answer to your question & the pin options & timers for various platforms.

BlackyS, 2017-02-02:
I (unfortunately) have the same issue with this setup... :(

Each individual libraries works perfect (stand alone), but as soon as I combine both libs and call the IR function "enableIRIn()" I'm still able to receive IR but I'm not able to send anything via RC anymore.

As I have an Arduino Nano clone I also tried to changed the timer in IRremoteInt.h file to the following:
(in my understanding this would switch the IR Code to use Timer1 instead of Timer1)

```
// Arduino Duemilanove, Diecimila, LilyPad, Mini, Fio, Nano, etc
#else
      #define IR_USE_TIMER1   // tx = pin 9
      //#define IR_USE_TIMER2     // tx = pin 3
#endif

```

=> the result: it still does NOT work.

I have the impression that this problem has nothing to do with timer problems (as suggested above ... maybe an Interrupt  issue?

AnalysIR, 2017-02-02:
@BlackyS 
As enablIRin essentially sets up the timer & sets the IR rx pin to input....it does appear to be a timer related conflict.

Perhaps contacting the author of the RC library my shed some light (e.g does the RC library require both Timer1 & Timer2???)

>  I also tried to changed the timer in IRremoteInt.h file

It looks like you are not using the latest version of the library....please update

> I have the impression that this problem has nothing to do with timer problems (as suggested above ... maybe an Interrupt issue?

FYI: In IRremote & usually Timer use involves the use of interrupts.

BlackyS, 2017-02-02:
THX for your fast response!

I updated my library (which indeed was not the latest - the update manager of the Arduino IDE obviously was not aware of the latest version ;) )

Hence I tried both Timer Settings via the newer boarddefs.h file:
```
// Arduino Duemilanove, Diecimila, LilyPad, Mini, Fio, Nano, etc
// ATmega48, ATmega88, ATmega168, ATmega328
#else
	//#define IR_USE_TIMER1   // tx = pin 9
	#define IR_USE_TIMER2     // tx = pin 3

#endif
```

=> ... still the same ... does NOT work for me either ... same behavior 

I also checked the RCSwitch Code a bit in more in details (as far as I'm capable of) and I only noticed that the "receiving()" function of the lib is using an interrupt (but I could not see any reference to a timer)
But as I'm ONLY just using the "sending" function of the lib, which neither seams to use interrupts nor timers :( ... it seams very confusing to me ...

On the other hand the enablIRin() Function of the IR lib obviously is preparing the usage of timer + interrupts ...

AnalysIR, 2017-02-02:
I believe enableIR**in** is for IR rx only (not 100% certain without more checking)

So if you are just sending....leave it out and try again.

AnalysIR, 2017-02-02:
One more thought ....

enablIRin does disable all interrupts and then enables all interrupts....so maybe RC does not want interrupts enabled at that time...worth trying to comment out the sei() in enablIRin  to see if it helps?????

....long shot!

BlackyS, 2017-02-02:
Oh ... sorry, I maybe was not clear on that ...

I'm **receiving an IR Signal** from a TV Remote (=> enableIRin is the one I'm looking for)
But I'm only **sending out an 433MHZ RC code** in order to switch on/off an 433MHz Plug (=> just sending)

THX for the hint ... I will give it a try and comment out that sei() ... defiantly worth a try ;) 

The strange thing is: my piece of code was in a working environment for more than a 1/2 year now, and I just wanted to replace my broken RC-433MHz-Plug with a new one ... so I just wanted to changed the 433MHz Datagram (to match the new Plug) and after the new upload it stopped working.... 
... very confusing ...

BlackyS, 2017-02-02:
Nope ... disabling sei() in that function did not solve it either :(

BlackyS, 2017-02-03:
I found a - maybe not very clean - solution ... but better than nothing.

I now simply disabled all interrupt before calling the RC-funtion send() -in order to send a 433MHZ datagram, and enable all interrrupts right after it.

```
  noInterrupts();
        myRCSwitch.send(17492,24);
  interrupts();
```
... as I said: better than nothing :)

z3t0, 2017-02-04:
This issue is being closed as there is no problem within the library and instead the issue is with conflicting libraries which try to use the same hardware. We could certainly tell you how to solve your issue but it would be specific to your particular project and hardware. Instead I recommend taking a look at the different hardware components used by this library and the conflicting one to determine what needs to be changed.

If you end up figuring that out and then have specific questions about setting the timer to timerX then feel free to create a new issue, and please follow the guidelines of issues when you create a new one.

closed 2017-02-04 by z3t0

qqq2001, 2018-04-18:
Actually is is not only a problem of the RC Switch Library. It is caused by the delayMicrosecond function (which is used in the RCSwitch Library).

So RCSwitch does send data but with wrong timing (checked with oscilloscope).

If you use a simple example like:
`digitalWrite(10, HIGH);`
`delayMicroseconds(1400);`
`digitalWrite(10,LOW);`

you get different results depending on the existence of "irrecv.enableIRIn();" in your code. Since delayMicroseconds doesnt use a timer but CPU instructions to cause a delay I guess that each Timer2 interrupt of IRRemote (each 50µs) somehow messes up the timing.  So disabling interrupts (like BlackyS said) actually works. But i dont know if IRRemote is working properly if you do that.