Arduino-IRremote/Arduino-IRremote #361: Attiny85 (SpenceKonde TinnyCore) - code hangs on irrecv.enableIRIn()

gosewski, 2016-09-10:
**Board:** Attiny85@8MHz (SpenceKonde TinnyCore)
**Library Version:** 2.1.0
**Protocol:** RAW 32 bit

Problem is as described in the title.

HINT: The tiny_IRremote library from https://gist.github.com/SeeJayDee solves the code hanging problem but it spoils receiving raw 32 bit data so it is not a good solution.

ccoenen, 2016-09-12:
I currently have 2.0.1 installed and running (more or less, but that's not the point). In that version, `enableIRIn` did work on a Attiny85.

Aztectrev, 2016-09-30:
I am also having this issue but with at Attiny84. What is the cause?

ccoenen, 2016-09-30:
@Aztectrev that is extremely hard to say with the current amount of information in this ticket.

@Aztectrev / @gosewski Could you also double-check with 2.0.1? Personally, I was able to run 2.0.1 on an ATtiny85. So knowing if it works for you, too will have two advantages:
a) you might have a working version in the meantime, and
b) if 2.0.1 works and 2.1 does not, the amount of code to be checked is much smaller than a wild goose chase in _all_ of 2.1's code.

Aztectrev, 2016-10-01:
For the 84 I've tried all release versions that compiled and none seemed to work. I've ordered another 84 and a few 85. I've not tested all releases on the 85, but latest didn't work and found code, from a different post on here (at least I think it was). Will find the link. Was a cut down version of the library just for the 85, which when o tried to use for the 84, was missing a lot of definitions.

I don't have access to the chips at the moment and need to wait for my new order to arrive as I got a few projects I want to do with this

z3t0, 2016-10-06:
Please try the following as it is the release where support for both 84 and 85 were added.

https://github.com/z3t0/Arduino-IRremote/blob/05cc3e683c87569eef19c4c7ac6bd2c752c686ac/changelog.md#201---20150726---release

ArminJo, 2020-10-12:
IRremote and [IRMP ](https://github.com/ukw100/IRMP) now supports ATtinies and Digispark.

closed 2020-10-12 by ArminJo