Arduino-IRremote/Arduino-IRremote #362: Generated code to big for ATTINY85

snoopy32, 2016-09-13:
**Board:** ATTINY85
**Library Version:** 2.1.0
**Protocol:** 

I tested compiling the example "IRrecvDemo" in the Arduino IDE v. 1.6.10 and the generated code is to big to fit into the ATTINY85.
Is there any optimisation that I can do or is there a problem?

z3t0, 2016-09-23:
I would recommend removing definitions for protocols you are not using.

kyet, 2017-04-08:
Also try to use LTO(Link Time Optimization) feature.

My experiment result (IRrecvDemo with SoftwareSerial)
disable: sketch 1898 bytes (build fail)
enable: sketch 7984 bytes (97%), global vars 371 bytes (72%)

refer -> https://github.com/SpenceKonde/ATTinyCore 
> In version 1.1.2 and later, this core supports Link Time Optimization (lto). This can substantially reduce the compiled size of your sketch. Version 1.6.13 or later of the official AVR boards package (included with 1.6.11 and later of the IDE) is required for this functionality. Link time optimization can be enabled from the option in the tools menu.

closed 2020-10-12 by ArminJo