Arduino-IRremote/Arduino-IRremote #364: Add architectures to library.properties

z3t0, 2016-09-24:
Once the Library Manager is released check for the supported architectures and add them to the library.properties file.

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

z3t0, 2016-10-06:
Edit: The documentation is outdated and I just realized I have been using Library manager all along as it is a part of the IDE... Will look into the architectures if i can find some updated documentation

ArminJo, 2020-10-17:
@z3t0 Hi Rafi, is this still an issue?
@bengtmartensson What is your opinion?
I can do it, but I do not want to decide it alone.

bengtmartensson, 2020-10-17:
Currently, it is 

    architectures=*

which is simply not true: the library does not run on all architectures. So, it is still an issue.

Gleaning on Infrared4Arduino, it should probably start with 

    architectures=avr,megaavr,samd,sam,esp32

possibly with some additions (attiny?)

ArminJo, 2020-10-17:
I changed it to avr,megaavr,samd,esp32,mbed, since sam (Arduino DUE is AFIK not supported).

closed 2020-10-17 by ArminJo