Arduino-IRremote/Arduino-IRremote #365: Support for RC6 36 bits

cklim01, 2016-09-29:
Hi, does latest IRremote library support to send RC6 36 bits code? If yes how to send? If no, how to modify it? Hope expert here can help. Thanks

z3t0, 2016-11-06:
It seems to not matter how many bits are used, though the implementation may differ in other ways such as the header?

AnalysIR, 2016-11-06:
Not 100% sure but the main difference could be the toggle bit position & no of bits (TBC). If the toggle bit is in a different location, the RC6 code would need to be adjusted to handle that.

You can always use sendRAW to send any IR signal (with a carrier).

z3t0, 2016-11-06:
> Not 100% sure but the main difference could be the toggle bit position & no of bits

Good point

AnalysIR, 2016-11-06:
....just checked the code and the max number of bits supported for RC6 in the library is 32 (an unsigned long value)...

...so for 36 bits, you need to use sendRAW for sending and some custom code for decoding...

...alternatively change the relevant variables to `long long` instead of `long `within the library. (for your use case)

_

> ....a hack that might work is to send the first 32 bits intially, immediately followed by the next 2 bits and immediately followed by the last 2 bits.
> So 3 RC6 signals sent one after another (32 bits->2 bits->2 bits). The last two avoid the toggle insertion if less than 4 bits long. However, sendRAW may be a whole lot simpler

_

z3t0, 2016-11-06:
>just checked the code and the max number of bits supported for RC6 in the library is 32

@AnalysIR but the bit of code below would suggest that the number of bits is given and not hardcoded? Please correct me if I am wrong as I haven't looked at the code for a while so I may just be going crazy haha.

https://github.com/z3t0/Arduino-IRremote/blob/master/ir_RC5_RC6.cpp#L58

AnalysIR, 2016-11-06:
That is correct...but the max is 32, which is of course less than the required 36 :)

BTW: The 32 bit limit is enforced by the unsigned long variable only.

cklim01, 2016-11-07:
i used the older version of library (i think is 2.0.1) and follow instruction in http://www.righto.com/2010/12/64-bit-rc6-codes-arduino-and-xbox.html then able to send 36 bits.

AnalysIR, 2016-11-07:
Great...it seems to use the method I suggested with long long variables.

...please close this issue.

z3t0, 2016-11-07:
> BTW: The 32 bit limit is enforced by the unsigned long variable only.

Ohh my bad, I missed that

z3t0, 2016-11-09:
@AnalysIR 
> BTW: The 32 bit limit is enforced by the unsigned long variable only.

Would you like me to update the current code to use `unsigned long long` ? as you are suggesting as that would solve the issue?

AnalysIR, 2016-11-09:
No.

I think changing this type could impact many of the existing protocols(?) & user projects for both sending and receiving.

My suggestion was merely a workaround for this particular user's needs. In the end, the OP found an existing implementation elsewhere which works.

Of course if someone was prepared to refactor the whole code base and test all the existing protocols for send & receive then the question would become "Is it desireable?"

In my opinion it is not a pressing requirement and can be parked.

z3t0, 2016-11-10:
I understand, thank you for the explanation

bengtmartensson, 2016-11-10:
The general issues has been discussed in #333.

ArminJo, 2020-10-19:
Is implemented now 😀

closed 2020-10-19 by ArminJo