Arduino-IRremote/Arduino-IRremote #366: Atmega328P on internal 8MHz oscillator

fire0shadow, 2016-09-30:
**Board:** Custom board with Atmega328P running on internal 8MHz oscillator
**Library Version:** 2.1.0
**Protocol:** any

**Code Block:**
```
#define F_CPU 8000000
#include "IRremote.h"

IRsend irsend;

void setup() {}

void loop() {
    irsend.sendSony(0x04);
    delay(1000);
}
```

Hello. When i run the library on Atmega328P with internal 8MHz oscillator, i receiving a double-timed signal:

What i expect:
2400 (header) - 1200 - 600 - 600 - 1200
What i got:
4800 - 2400 - 1200 - 1200 - 2400

On Atmega328P with external 16MHz oscillator, everything works perfectly. Is the library does not support the internal 8MHz oscillator of Atmega328P?

AnalysIR, 2016-09-30:
looks like it is actaully @ 16MHz?

Have oyu configured the fuses correctly? (CLKDIV)

try #define F_CPU 16000000
or leave it out

fire0shadow, 2016-10-01:
Fuses are set as follows:
LFUSE: 0xE2
HFUSE: 0xDA
EFUSE: 0x05
It's fuses for internal oscillator, as it says arduino.cc site or [AVR Fuse calculator](http://www.engbedded.com/fusecalc/).
I tried to set F_CPU to 16000000 or just delete it, but nothing happened.

fire0shadow, 2016-10-01:
Temporary hack: i just divided by 2 header, space and mark variables, but it's kinda... stupid.

AnalysIR, 2016-10-01:
Make sure to set the board up as a Pro Min @ 8MHz
(If you select a UNO or Nano then it assumes the clock is 16MHz)

Make sure you have the correct bootloader for the Board type selected.

Alternatively create your own "boards" file for the Arduino IDE.

Finally, post the output from the IRremoteInfo example sketch.

fire0shadow, 2016-10-01:
Sorry, i forgot to say: i'm not using bootloader, firmware uploads directly to Atmega via USBasp. Fuses set up similar as Pro Mini on 8MHz. Due to PCB design i don't have access to serial, so i can't post output of IRremoteInfo. All configuration is default.

AnalysIR, 2016-10-01:
The settings on the Arduino IDE are also important. Have you selected Pro Mini 328P @ 8MHz under the tools menu?

Another thing to try is to blink a LED 5 secs on 5 secs off to see if the default Timer is configured correctly.

===============
Maybe the best way to resolve this is to force the value for F_CPU within the library by adjusting the following lines of IRremoteInt.h 

> //------------------------------------------------------------------------------
> // CPU Frequency
> //
>  #ifdef F_CPU
>  #define SYSCLOCK  F_CPU     // main Arduino clock
>  #else
>  #define SYSCLOCK  16000000  // main Arduino clock
>  #endif

Maybe the value for F_CPU is not getting thru to the library & default is 16MHz

closed 2016-11-22 by fire0shadow