Arduino-IRremote/Arduino-IRremote #369: ATTiny84 IR Output Pin Options?

Wesstun, 2016-10-11:
Hey guys, I'm working on a project trying to send IR codes with an ATTiny84.  It works great if I use pin 6 (physical pin 7 on the DIP/SOIC-14 package) as intended by the library.

Unfortunately, pin 6 is a critical pin, because it's used for MOSI and SDA with the USI, so assigning it to my IR LED knocks out both hardware SPI and I2C, which is painful.

Is there any practical way that pin 7 (physical pin 6 on DIP/SOIC) could be used as an output pin for the IR send instead?  I realize that not just any pin can be used, and that's it's very timer/architecture dependent.  I just don't have enough expertise to know what's possible.  I see that on the ATTiny84, pin 6 is associated with the OC1A timer, and pin 7 is associated with the OC0B timer, but does that even mean anything for the IR Library?  It seems to me like the common denominator for the default pin in all of these different chips is that it's a PWM-capable pin, but that not all PWM pins work for timer-related reasons that are beyond me.

I also really need to keep pin 8 available, because I need the interrupt.  So even though that pin is also associated with OC1A (like pin 6), it wouldn't be a great alternative, at least for me.

Even if I do put the IR LED on pin 6, that MOSI pin still needs to be connected to the ICSP header for programming, and the IR LED goes nuts when I'm programming a board from the MOSI pulses.  Pin 7 would be so much more convenient if there's any way.

If there's no way to use a different pin for the IR send, and I really do need SPI functionality, what's my best option?  Probably just bit-banging some software SPI over different pins?  I'm ok with that, I just wanted to check to make sure that there are no other options.  I can't possibly be the only person who wants to use the IR library and serial functions at the same time with the ATTiny family of chips.

AnalysIR, 2016-10-11:
I guess your 2 best options are to:
- bit-bang the SPI, or
- use a softPWM(essentially bit-banging the carrier) approach for generating the IR carrier (ie replace the mark funtion).

We have 3 part series on our blog with examples of softPWM for IR. 
[Simple Infrared PWM on Arduino](https://www.analysir.com/blog/2015/05/12/simple-infrared-pwm-on-arduino/)

Of course a better approach would be to use a more suitable model of AVR MCU for the task.

Wesstun, 2016-10-12:
Thanks for the response!

I agree that my life would be a whole lot easier if I was just using a different micro.

I read through your tutorial on 'bit banging' the carrier for the IR signal.  I've never seen it before but you guys did an excellent job putting that together.  Very clear and helpful.

If I have to bit bang something though, I'm more comfortable doing it with SPI.  My particular project has lighter-duty serial needs, and I guess I can live with the IR LED blasting away while I program my micro.  Compromises.

Thanks again.

Wesstun, 2016-10-12:
Just out of curiosity, is the reason that it wouldn't work due to timer conflicts or what?  Is the USI on the ATTiny family pretty much out of commission when using the IRremote library?

I ask because the ATTiny85 has the same problem.  Its IR transmit pin (1) is also its MOSI pin.  So hardware SPI with the USI is out.  HOWEVER, unlike the ATTiny84, its I2C pins are in the clear.  I'm wondering if would be possible to use I2C and the IRremote library at the same time on an ATTiny85.

Not that it would particularly matter for me . . . the SCL pin is also the interrupt!  Back to the bit-banging board.

yesyesuk, 2016-10-12:
This is probably quite off-topic, and I apologise...
But this discussion (and many others here) make me wonder...
I'm very happy that this library exists for Arduino (and others) but it seems that the ATmel chips are not very well suited for this kind of thing.
If one had to design a completely new device with IR send and receive and it should still be able to do other things and you had free choice of which micro controller you could use, which one would you choose? What do commercial devices use for IR?

AnalysIR, 2016-10-12:
> but it seems that the ATmel chips are not very well suited for this kind of thing

I disagree with the above as a general statement. Every supplier has a range of MCUs designed to fit particular tasks & budgets.

Any MCU with 512+ bits of RAM, 8K Flash & PWM running at 16MIPS(possibly 8MIPS) can handle 99%+ of IR remote control projects. The problems usually arise when trying to do too much with an MCU that is designed for less. You always need to match the specs of the MCU to the requirements.

Usually custom ICs (COBs) are used for consumer IR remote controls.

AnalysIR, 2016-10-12:
@Wesstun 
> Just out of curiosity, is the reason that it wouldn't work due to timer conflicts or what? 

Lack of resources/pins on this MCU for your particular requirements. As above its a matter of matching the resources of the MCU to the requirements of the project. Just because you have a particular MCU available, doesn't mean its the right/best one for the task.

FYI: you can get an Arduino Nano clone delivered from Aliexpress for $2.

Wesstun, 2016-10-12:
Hey guys,

If price were no object, I'd definitely use a full ATMega328 and be done with it.  I'd have full hardware SPI, an interrupt, a dedicated pin for IR send, UART for debugging, I2C if I need it, a dozen other functions and way more RAM and program space than I would ever need for this project.  And for my proof of concept on this project that's exactly what I used.  It worked great.

I do want to to turn this into a product though, and I'm targeting a complete BOM cost of less than $5 for the final product.  Going from an ATTiny84 which costs $0.70 in volume to an ATMega328 which costs about $2 in volume, just to make things a tiny bit easier on myself, just isn't worth it for me.  I finished bit banging my SPI last night with direct port writes and it's fast enough that it will in no way affect the customer experience.  It was just a little more painful for me getting it going.  Compromises were made. =)

And yesyesuk I'm kinda with you . . . it kinda makes me crazy that on, for example, the ATTiny85, its external interrupt pin is on one of the same pins that's required for SPI communication, which means that this micro is an immediate non-starter for any project that requires both, but I can't really fault Atmel for not giving me all the goodness that's on its other chips which cost 3X more.

yesyesuk, 2016-10-12:
I understand that these controllers were designed for something completely different and then us enthusiasts come along and re-purpose them. ;-)
I agree with both of you. I guess I was just asking if there is any micro controller that is better suited for things like IR send and receive than the ATmel series, specifically the ones used in Arduinos.
I'm just used to these chips because I started off with Arduinos. But maybe there is an alternative that can send/receive IR and control LEDs and send/receive serial data and so on without any conflict of timers and the like because their architecture is better suited for these kinds of things...

closed 2020-10-12 by ArminJo