Arduino-IRremote/Arduino-IRremote #371: change requirement for using pin #3 for output

movingelectrons, 2016-10-17:
**Board:** ARDUINO UNO r3
**Library Version:** 2.1.0

I am wondering why it is (if it is) necessary to use pin 3 for the IR led output.  I am using a 2.4 inch lcd/touchscreen shield and unfortunately pin #3 in is use.. however pins 10 and 11 are free, among a few others.  I poked around in the source code for a little while and did not find out where the output pin # is set (yet)

AnalysIR, 2016-10-17:
you should be able to use pin 9 by changing this code in IRremoteint.h

```
// Arduino Duemilanove, Diecimila, LilyPad, Mini, Fio, Nano, etc
// ATmega48, ATmega88, ATmega168, ATmega328
#else
	//#define IR_USE_TIMER1   // tx = pin 9
	#define IR_USE_TIMER2     // tx = pin 3

#endif
```
......simply comment/uncomment the relevant line and restart the IDE to make sure it is recompliled.

Pins 10 & 11 may also be possible, but you would have to figure out the register settings and make the appropriate changes etc....

Of course the other alternative is to change the lcd library :)

AnalysIR, 2016-10-17:
....forgot to mention that you would be changing the timer used which could cause a conflict with other libraries.

movingelectrons, 2016-10-18:
what about if I used a MEGA instead of a UNO r3?

thanks

AnalysIR, 2016-10-18:
yes -you get more options

```
// Arduino Mega
#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
	//#define IR_USE_TIMER1   // tx = pin 11
	#define IR_USE_TIMER2     // tx = pin 9
	//#define IR_USE_TIMER3   // tx = pin 5
	//#define IR_USE_TIMER4   // tx = pin 6
//#define IR_USE_TIMER5   // tx = pin 46
```
Note your LCD may use different pins on the Mega!

pin 46/Timer5 looks a safe bet - default is set to pin 9

**(!!! all untested !!!)**

AnalysIR, 2016-10-18:
FYI - this is all explained in the [IRremote README](https://github.com/z3t0/Arduino-IRremote)

z3t0, 2016-10-27:
@movingelectrons  It might be a good idea to try and figure out first which pins are being used by your other libraries so then we can figure out which pin / timer is free for the ir remote library

z3t0, 2016-11-20:
Closing as the question seems to have been answered

closed 2016-11-20 by z3t0