Arduino-IRremote/Arduino-IRremote #372: Try to create a B&O remote but get to view data

wolfgangschneider, 2016-10-18:
**Board:** ARDUINO Nano
**Library Version:** 2.1.0
**Protocol:** Bang & Olusen

**Code Block:**
I can send Data with the irsend.sendRaw function and everything works fine. If I try to recive data using the IRrecvDump example , the app switch to UNKNOWN protokol, and use the decodeHash function the output looks like:

25AE7EE0
Unknown encoding: 25AE7EE0 (32 bits)
Raw (6): 200 -2900 200 -2900 200

the data looks good but there are to view , i schuld get:
{200, 3125, 200, 3125, 200, 15625, 200, 3125, 200, 6250, 200, 6250, 200, 6250, 200, 6250, 200, 6250, 200, 6250, 200, 6250, 200, 6250, 200, 9375, 200, 3125, 200, 6250, 200, 6250, 200, 6250, 200, 6250, 200, 6250, 200, 6250, 200, 12500, 200}; 

What can I do to recive more all data? As a test/hack I try several thinks e.g set results->rawlen to 20 in decodedHash but than i get 
Raw (20): 200 -2850 250 -2850 200 -0 0 -0 0 -0 0 -0 0 -0 0 -0 0 -0 0 

It looks like that the irparams.rawbuf has no more Data. BTW I didn't find out where the irparams.rawbuf will be filled.

could you please so kind, and bring me to the right direction

Wolfgang

#include <IRremote.h> 

.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided
- [] The title of the issue is helpful and relevant 

The above is a short template allowing you to make detailed issues!

AnalysIR, 2016-10-18:
As you may know B&O uses a carrier of 455kHz.

So it would not be surprising is a typical IR receiver would not receive the signal. (i.e. @30-56kHz)

What model IR receiver are you using?

Also, B&O is not supported by IRremote so you will not get a meaningful HEX value.

wolfgangschneider, 2016-10-18:
Found it :)
In case that B&O sends with 455 kHz i set in IRRemoteInt.h
// Minimum gap between IR transmissions
#define _GAP            20000//  org 5000

closed 2016-10-18 by wolfgangschneider