Arduino-IRremote/Arduino-IRremote #373: Wrong HDR sent

lettierimarco, 2016-10-21:
**Board:** ARDUINO Mega 2560
**Library Version:** 2.1.0
**Protocol:** NEC
Hex: 20DF40BF

[image]

Hi, via @AnalysIR  software i'm debugging a sent signal from my arduino mega.
The first in the image attached is the command from the remote controller and second is from my arduino with Arduino-IRremote, i2C (wire library) and Freqmeasure library (that i disabled for testing thinking it's a ISR problem).

Very often (up to 60%) of commands sent from the arduino have a wrong first HDR mark duration like in the image.
This doesn't happens only with nec but with every brand or raw.
Have you ever had the same issue?

lettierimarco, 2016-10-21:
He's a RAW command
[image]

AnalysIR, 2016-10-21:
I see a shorter header in the first image and a missing header in the second image.

everything else seems to be OK.

I suggest you try the following sketch to test your IRremote install & Arduino config.

```
void loop() {
	irsend.sendNEC(0x5A5A5A5A, 32);
	delay(5000); //5 second delay between each signal burst
}
```
This should isolate any issues to IRremote, your IR Receiver or your ARDuino/IDE etc.

If that does not work, post the results of the following sketch:
[IRremoteInfo](https://github.com/z3t0/Arduino-IRremote/tree/master/examples/IRremoteInfo)

PS: one final thought....
....post the model#/link of your IR receiver - as it could also be that (if it truncates the first part of any signal), which could also explain your images

lettierimarco, 2016-10-21:
Waiting your reply i've added a delay 20 in IRSend.cpp, enableIROut after TIMER_CONFIG_KHZ(khz); 
Now seems work... probably it must have a more time to config frequency or disable timer

AnalysIR, 2016-10-21:
...strange...thanks for the update

z3t0, 2016-10-27:
This seems to have been solved, feel free to reopen if that is not the case.

closed 2016-10-27 by z3t0