Arduino-IRremote/Arduino-IRremote #374: Wrong Decoded IR Code

asyncer, 2016-10-30:
Hello,

I've used the sample below to decode my SAMSUNG TV IR codes for test, But I can see that the result is not OK and it differs from what is in https://irdb.globalcache.com/Home/Database,
And as IRDB codes are working fine with my IR sender, I guess the IR codes that below program is dumping are wrong.
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino

Anyone has experience on how to adjust the codes IRrecvDumpv2 ?

One example of wrong code:
The decoded IR code for SAMSUNG power is:
unsigned int rawData[67] = {4550,4450, 650,1650, 650,1650, 650,1650, 650,500, 650,500, 650,550, 650,550, 650,550, 650,1650, 650,1650, 650,1650, 650,550, 650,550, 650,550, 650,550, 650,500, 650,550, 650,1650, 650,500, 650,550, 650,500, 650,500, 650,550, 650,500, 650,1650, 650,550, 650,1600, 650,1650, 650,1650, 650,1600, 650,1600, 650,1650, 650}; // SAMSUNG E0E040BF

And the one from IRDB is:
unsigned int Samsung_power_toggle[71] = {38000,1,1,170,170,20,63,20,63,20,63,20,20,20,20,20,20,20,20,20,20,20,63,20,63,20,63,20,20,20,20,20,20,20,20,20,20,20,20,20,63,20,20,20,20,20,20,20,20,20,20,20,20,20,63,20,20,20,63,20,63,20,63,20,63,20,63,20,63,20,1798};

Both data and length are different and IRDB code is working fine, but I need to be able decode the signals myself.

bengtmartensson, 2016-10-30:
The problem is that the last code is in a completely different format. It is GlobalCache's irsend format (used by hardware manufactured by said firm), with "sendir" removed, and a ":" changed to a ",". I severely doubt the statement that it is working for you. (What hardware/software are you using?) For example, most numbers denote a duration, not in microsecods, but in multiples of the modulation frequency period.

You can use current IrScrutinizer (version 1.3)  to convert the mail with the codes to Arduino code in with a few clicks: First copy it to the clipboard. In IrScrutinizer, select the Import -> Text -> Raw pane, Select "Name col! to 1, "Raw signal col" to 3 (or 2), and "Field separator" to the comma ",". Press "load from clipboard", then "import all" (or "import all/raw"). Go to the export pane and select "Arduino Raw" as export format. Press ""export param. remote" (or "export raw remote", if you imported with "import all raw"). This generates an *.ino file for you.

asyncer, 2016-10-30:
I'm using a simple IR Receiver module connected to ESP8266 and using the code mentioned in below link:
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino

My issue is not with simple data formatting, As you can see the decoded IR codes are totally different from IRDB, I need to capture correct codes so I can use this system to decode IR codes of infamous devices.

AnalysIR, 2016-10-30:
IRremote records the times in chunks of 50uSecs by default. I believe this parameter can be adjusted.

Your IR receiver will also 'skew' the timings of mark/spaces up to +/-200uSecs (more often <100uSecs).

Our sketch will get you better timings than IRremote, via:
[Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

FYI: The timings for both your signals were out a bit when I checked them. As it happens the timings for the RAW signals recorded by IRremote was more accurate, than the one from the DB. !!

closed 2020-10-17 by ArminJo