Arduino-IRremote/Arduino-IRremote #375: Recommend more specific header file name

PaulStoffregen, 2016-11-10:
Please consider renaming "boarddefs.h" to something more specific, like "IRremoteBoarddefs.h".

Arduino indexes all .h files.  If another library also uses this generic name, a library matching conflict can occur, where Arduino may infer the use of the wrong library.  These issues are not common, but when they do strike the result is terribly confusing.

Changing the name to be IRremote specific is a simple and easy thing.

bengtmartensson, 2016-11-10:
If this is an issue (I am not saying or implying it isn't), this should go into some general library guide lines. (Possibly there is already one?)  Where is the indexing described?

But them, are the existing file names specific enough?

z3t0, 2016-11-12:
Are there any other files that should be renamed?

>  Where is the indexing documented or described?

Perhaps as @bengtmartensson has suggested there is somewhere we can check?

z3t0, 2016-12-04:
One way to move forward with this?

> Finally, I noticed the recent update with a boarddefs.h file. You might consider putting this into a sub directory with one file per 'board/MCU' supported, along with conditional compiles to include the correct board file. I believe this will make it easier for newcomers to grasp the details vs one increasingly large file.

Quoted from: @AnalysIR

bengtmartensson, 2016-12-06:
I opened [this thread](http://forum.arduino.cc/index.php?topic=439926.0) in the Arduino forum.

    One way to move forward with this?...

It would rather amplify the problem,  with files like, say, boarddefs_uno.h

bengtmartensson, 2016-12-08:
I tried to understand what really happens.

When calling the compiler, it is given a list of include directories to search for include files (-I...). This can potentially cause problem if a file is found in several of these. However, boarddefs.h is included from IRremoteInt.h by
     
    #include "boarddefs.h"
not by

     #include <boarddefs.h>

This means that it is searched in the same directory as IRremoteInt.h, so that if the correct version of IRremoteInt.h is found, the correct version of boarddefs.h will be used, also if there are other versions in other include directories. I have even tried it. boarddefs.h is a private, internal file that should not be directly included by any file not belonging to the library. So, IRremote is immune to other libraries using boarddefs.h.

The only thing that can bite is if another library decides to introduce a file called boarddefs.h, and to make it public in the sense of recommending users to `#include <...>` it (as opposed to `#include "..."`). Under some circumstances, it may find the  IRremote file. 

It is my opinion that, since "our file" does not break any published guidelines, it is not a "bug", and no change is necessary. 

I should say that if Rafi decides to rename it "just to be on the safe side", I have no problem with that either. But I would prefer if it is settled soon.

bengtmartensson, 2017-01-20:
I would politely request that this issue is resolved soon.

As I wrote above, IMHO a fix is not necessary. I disagree with the "Bug" classification/label. On the other hand, (again written above) I have no problem with a renaming either. But what consider bad is the uncertainty.

It should not take more than minutes...

z3t0, 2017-03-26:
Will continue in #405

closed 2017-03-26 by z3t0