Arduino-IRremote/Arduino-IRremote #376: IRrecieve and IRsend in same sketch hangs

buestad, 2016-11-11:
I am working on a project which has both a receiver and sender. My problem is that the reicever dies after the sender has sent something. I made a small sketch as example. I use the Sharp protocol which I have modified, but the error occurs when sending using Sony as well.

I use a Arduino Uno with a W5100 ethernet shield

```
#include <IRremote.h>
IRrecv irrecv(2);
IRsend irsend;
decode_results results;
unsigned long lastIR = millis();
void setup() 
{
  Serial.begin(9600);
  delay(100);
  irrecv.enableIRIn(); // Start the receiver
}
void loop() 
{
  if (irrecv.decode(&results)) 
  {
    if (millis() - lastIR > 250)
    {
      Serial.println(results.value, HEX);
      if(results.decode_type == SHARP && results.address == 0x02 && results.value == 0xE1) 
      {
        delay(1000);
        //irsend.sendSharp(0x02,0xE2);
        irsend.sendSony(0xa90, 12);
        delay(1000);
      }
    } 
    lastIR = millis();
    irrecv.resume(); // Receive the next value
  }
}
```

AnalysIR, 2016-11-11:
You must enable the receiver after every send!  

`irrecv.enableIRIn(); // Start the receiver`

buestad, 2016-11-12:
It works!

closed 2016-11-12 by buestad