Arduino-IRremote/Arduino-IRremote #378: ghost "RobotIRRemote" file being accessed

beau-dids, 2016-11-13:
I am trying to run the example file IRrecvDemo to get the hex code on my remote control. When I compile the code I get an error that "IRremote.h" is being accessed through RobotIRRemote library and not this one. I never downloaded the RobotITRemote files so I don't understand where it is accessing these files. I have tried adjusting the name of the header file but that causes further problems.

yesyesuk, 2016-11-13:
The RobotIRRemote is part of the Arduino IDE installation. I have deleted it from my Libraries folder to avoid this confusion.

z3t0, 2016-11-13:
@beau-dids https://github.com/z3t0/Arduino-IRremote#installation

as @yesyesuk said the RobotIRRemote Library is one provided by the IDE, ironically I have never seen anyone actually use it? Maybe we should open an issue on the arduino repository as it definitely is quite confusing for people just picking up this library?

Thoughts?

z3t0, 2016-11-13:
I'll just close this as the README answers the issue

closed 2016-11-13 by z3t0