Arduino-IRremote/Arduino-IRremote #380: Sharp message is split into 3 messages

buestad, 2016-11-14:
I am working on a project that uses the Sharp protocol. The decodeSharp function is missing in the ir_sharp.cpp as of today, but I have made it in my fork. 
https://github.com/buestad/Arduino-IRremote

My problem is that the message from one command from the remote is split into three messages before I read it in the decodeSharp function. One Sharp message contains the data three times. First all bits are normal, then all are inverted and lastly normal again. In between these is a pause of roughly 40ms. 

Is there a way to merge theses three messages together before the decode functions are called?

AnalysIR, 2016-11-14:
Probably a good idea to post the model of device & remote (also as a comment in the code), as there will be more than one Sharp IR protocol.

It would be useful to post a sample set of RAW signals here for reference.

buestad, 2016-11-14:
I use a Denon RC-940 remote control. This has a bunch of different protocols I believe but according to the IR code document from Denon (assets.denon.com/documentmaster/us/denon%20master%20ir%20hex.xls) it uses the Sharp protocol on the stuff I am interested in. I have tested using a Denon AVR 1804 reciever. I have some raw data but it is captured using a different library because I think it is a bit more readable. Here is an example from pressing the power on button:

```
Decoded Sharp: adr:02 data:E1

Received: 

OFF 	ON
53472 usec, 200 usec		
820 usec, 200 usec	0	
1820 usec, 220 usec	1
780 usec, 220 usec	0
800 usec, 180 usec	0
840 usec, 160 usec	0

1860 usec, 220 usec	1	
800 usec, 280 usec	0
740 usec, 200 usec	0
780 usec, 240 usec	0
780 usec, 180 usec	0
1840 usec, 220 usec	1	
1800 usec, 280 usec	1		
1760 usec, 160 usec	1

840 usec, 180 usec	0
840 usec, 280 usec	0

42620 usec, 260 usec
760 usec, 240 usec	0
1780 usec, 200 usec	1
820 usec, 240 usec	0
780 usec, 200 usec	0
800 usec, 220 usec	0

800 usec, 180 usec	0
1840 usec, 200 usec	1
1820 usec, 260 usec	1	
1760 usec, 200 usec	1
1800 usec, 220 usec	1
820 usec, 180 usec	0
820 usec, 200 usec	0
800 usec, 260 usec	0

1780 usec, 180 usec	1
1820 usec, 240 usec	1

42680 usec, 220 usec
780 usec, 260 usec	0
1780 usec, 220 usec	1
780 usec, 200 usec	0
840 usec, 240 usec	0
740 usec, 200 usec	0

1820 usec, 240 usec	1
800 usec, 160 usec	0
840 usec, 220 usec	0
800 usec, 240 usec	0
760 usec, 200 usec	0
1820 usec, 220 usec	1
1820 usec, 160 usec	1
1840 usec, 220 usec	1

780 usec, 280 usec	0
740 usec, 200 usec	0
```

AnalysIR, 2016-11-15:
OK - Thanks for that

Your signal decodes as Denon on AnalysIR. So you should change the naming convention from Sharp to Denon.

One important issue to note, is that this signal will be very difficult to decode reliably using IRremote due to the relatively short nature of the marks combined with the +/- 50uSecs granularity of IRremote. Sending should be OK with IRremote.

@z3t0 ...FYI

buestad, 2016-11-15:
If it is a Denon or a Sharp protocol really does not matter here. The thing is that there is a long pause of  ~42mSec between these three "chunks" of data as you can see in the raw print. It is this pause which splits the message into three messages. The decodeWhatever() function will be called three times.
My question is if it is possible to prevent the "core" of IRremote to split this. I guess that somewhere there is a setting which specifies the length of the pause between messages.

AnalysIR, 2016-11-18:
Not 100% sure, but try increasing the following in IRremoteInt.h to 50000 uSecs

`#define _GAP 5000`

buestad, 2016-11-21:
Thanks! Do you know if it is possible to override this #define inside the arduino .ino file or do I have to change the library setting?

z3t0, 2016-11-21:
@buestad the library itself needs to be modified unfortunately

buestad, 2016-11-23:
I have tried to change the _GAP setting to 50000 and it works like expected for single button press. I now get all three parts of the message (96 raw samples, before it was 32) into one call of decodeSharp() function. 

But the gap between each message (after 96 samples) is the same as the internal gap (~42ms). So if i press and hold the button i get even more raw samples into one function call. looks like the raw buffer overflows at 101 and decodeing fails since the overall size is to large...

AnalysIR, 2016-11-23:
you can increase the value of RAWBUF in the library settings, IRremoteInt.h

`#define RAWBUF 101 // Maximum length of raw duration buffer`

I believe it is an 8 bit value.

_PS: Because IRremote is a general purpose library designed to work with other sketches/libraries & running on an MCU, there will always be trade-offs (Features vs resources). I believe the defaults were set for common use cases_

buestad, 2016-11-24:
Thanks! I think this wraps up what I was searching for.

closed 2016-11-24 by buestad