Arduino-IRremote/Arduino-IRremote #386: PSA : Accepting Feature Requests

z3t0, 2016-12-04:
Hi All,

### Announcement

First of all I apologize for not being active in the community due to my other commitments.

However! I will be spending a whole week during the Christmas break to completely revamp this library. 

I will be accepting any and all requests, regardless of their nature. There is however one caveat. If you want me to add support for a board not currently supported then you may have to spend some money shipping one to me. But ask me first as I do have quite a collection and so might already have access to the board.

### Steps
1. Create an Issue **detailing** your request. Please provide as much detail as possible.
2. Tag me by typing "@z3t0" somewhere in the issue
3. Tag this issue by typing "#[insert issue here]"
4. Await patiently and check below to see once your issue has been added to the list to be worked upon.

### Issues to be worked on:
*if your request is in this list then it has been brought to my attention and I will work on it, if not please ensure you follow the steps above!*

z3t0, 2016-12-04:
@AnalysIR @bengtmartensson @shirriff @joshuajnoble @pcoughlin @csBlueChip 

Just thought I'd tag some of the main contributors! Sorry for the spam 😄

AnalysIR, 2016-12-04:
A Tutorial on how to add a new Board, might be very useful, with one section devoted to Board suppliers/makers.

Similarly, a Tutorial on adding a new protocol....

It seems the above 2 points would lead to an ever increasing number of supported platforms/protocols.

Of course, one the the biggest barriers to getting 'code' contributions is the learning curve for making contributions to GitHub (PRs etc). I have made a few, but I know I couldn't explain the process properly. Maybe this already exists elsewhere and just needs to be linked?.

A statement on the Goals of the libary, might also help. What is within the scope of the Library & what is not. (For example, I believe it has served well for 8 bit MCU with limited RAM, but may not be the ideal approach for the ever increasing number of more powerful MCUs with lots more resources(Flash/RAM).

Finally, I noticed the recent update with a boarddefs.h file. You might consider putting this into a sub directory with one file per 'board/MCU' supported, along with conditional compiles to include the correct board file. I believe this will make it easier for newcomers to grasp the details vs one increasingly large file.

z3t0, 2016-12-04:
@AnalysIR Thanks I've added all of your suggestions to the list

z3t0, 2017-04-10:
#400

closed 2020-10-17 by ArminJo